openrazer/openrazer #1701: OpenRazer not detecting Viper 8K

duartec3000, 2021-12-09:
### Distribution
```
➜  ~ neofetch
                   -`                    g@abox 
                  .o+`                   ------ 
                 `ooo/                   OS: Arch Linux x86_64 
                `+oooo:                  Host: MS-7C02 1.0 
               `+oooooo:                 Kernel: 5.15.5-zen1-1-zen 
               -+oooooo+:                Uptime: 4 hours, 55 mins 
             `/:-:++oooo+:               Packages: 1220 (pacman), 19 (flatpak) 
            `/++++/+++++++:              Shell: zsh 5.8 
           `/++++++++++++++:             Resolution: 1920x1080 
          `/+++ooooooooooooo/`           DE: GNOME 41.1 
         ./ooosssso++osssssso+`          WM: Mutter 
        .oossssso-````/ossssss+`         WM Theme: Adwaita 
       -osssssso.      :ssssssso.        Theme: Adwaita-dark [GTK2/3] 
      :osssssss/        osssso+++.       Icons: Adwaita [GTK2/3] 
     /ossssssss/        +ssssooo/-       Terminal: gnome-terminal 
   `/ossssso+/:-        -:/+osssso+-     CPU: AMD Ryzen 5 2600X (12) @ 3.600GHz 
  `+sso+:-`                 `.-/+oso:    GPU: AMD ATI Radeon RX Vega 56/64 
 `++:.                           `-/+/   Memory: 3365MiB / 16013MiB 
 .`                                 `/
```

### Support

- [ x] Have you tried rebooting?
- [x ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I'm using **openrazer-meta-git** and **polychromatic-git** but for the sake of me I can't get them to detect my **Razer Viper 8k**. I've see this issue closed on the Oct 9 [#1426](https://github.com/openrazer/openrazer/issues/1426) and thought support for this model was fine but maybe it isn't? or needs extra steps?

Discarded problems with my installation because the now refunded DeathAdder V2 worked fine, DPI, polling rate and RGB leds all could be changed.

I did updgrade from openrazer-meta to openrazer-meta-git just by doing:
`yay -S openrazer-meta-git` it seemed to detect the previous installed packaged, removed them and replaced them.

Any help you can provide would be truly appreciated, here are some logs:

`➜  ~ lsusb | grep 1532

Bus 003 Device 002: ID 1532:0091 Razer USA, Ltd Razer Viper 8KHz
`
`➜  ~ cat ~/.local/share/openrazer/logs/razer.log
2021-11-28 03:30:43 | razer                          | INFO     | Initialising Daemon (v3.1.0). Pid: 33093
2021-11-28 03:30:43 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-11-28 03:30:43 | razer                          | INFO     | Found device.0: 0003:1532:0084.0001
2021-11-28 03:30:43 | razer.device0                  | INFO     | Initialising device.0 RazerDeathAdderV2
2021-11-28 03:30:43 | razer                          | INFO     | Serving DBus
2021-12-03 09:37:34 | razer                          | INFO     | Initialising Daemon (v3.1.0). Pid: 5914
2021-12-03 09:37:34 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-12-03 09:37:34 | razer                          | INFO     | Found device.0: 0003:1532:0084.0001
2021-12-03 09:37:34 | razer.device0                  | INFO     | Initialising device.0 RazerDeathAdderV2
2021-12-03 09:37:34 | razer                          | INFO     | Serving DBus
2021-12-09 15:53:09 | razer                          | INFO     | Initialising Daemon (v3.1.0). Pid: 2968
2021-12-09 15:53:09 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-12-09 15:53:09 | razer                          | INFO     | Serving DBus
2021-12-09 16:09:39 | razer                          | INFO     | Stopping daemon on signal 15
2021-12-09 16:09:41 | razer                          | INFO     | Initialising Daemon (v3.1.0). Pid: 8639
2021-12-09 16:09:41 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-12-09 16:09:41 | razer                          | INFO     | Serving DBus
2021-12-09 16:10:37 | razer                          | INFO     | Stopping daemon on signal 15
2021-12-09 16:10:39 | razer                          | INFO     | Initialising Daemon (v3.1.0). Pid: 8717
2021-12-09 16:10:39 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-12-09 16:10:39 | razer                          | INFO     | Serving DBus
2021-12-09 16:13:10 | razer                          | INFO     | Stopping daemon on signal 15
2021-12-09 16:13:48 | razer                          | INFO     | Initialising Daemon (v3.1.0). Pid: 1658
2021-12-09 16:13:48 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-12-09 16:13:48 | razer                          | INFO     | Serving DBus
2021-12-09 20:53:36 | razer                          | INFO     | Stopping daemon on signal 15
2021-12-09 20:53:38 | razer                          | INFO     | Initialising Daemon (v3.1.0). Pid: 11758
2021-12-09 20:53:38 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-12-09 20:53:38 | razer                          | INFO     | Serving DBus
`
`➜  ~ dkms status                                
openrazer-driver/3.1.0, 5.15.5-zen1-1-zen, x86_64: installed
`

lah7, 2021-12-09:
Since you're using Polychromatic, what's the output of the **Troubleshoot** button? This might identity which part of the software stack (driver, daemon, other) that's not working.

Did you restart after installing the new driver? Not sure if upgrading the driver (`razermouse`) unloads/loads the new version, but if not, you should be able to just unplug and re-plug the mouse, and then hard refresh the frontend (CTRL+F5) for the new driver (with support for this device) to load.

duartec3000, 2021-12-09:
> Since you're using Polychromatic, what's the output of the **Troubleshoot** button? This might identity which part of the software stack (driver, daemon, other) that's not working.
>
> Did you restart [...]

Thank you so much for your suggestions it really helped me troublelshoot further, the problem here was that while upgrading openrazer-meta to the git version it didn't upgrade all the packages so the dkms module was still the stable one.

I removed all razer related packages from my system > restarted > installed openrazer-meta-git and polychromatic-git with yay > restarted > All good, super happy!

It is even showing the 8000 Hz option to select although I'm not going to because most games I play behave weird with it.

thanks again @lah7 and Big Up to the whole team of OpenRazer you guys kick ass!

closed 2021-12-09 by duartec3000