openrazer/openrazer #1702: Missing Effects-Support on Razer Book

Khraut, 2021-12-10:
### Distribution
Manjaro Linux; Build-ID: rolling

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I have the Razer Book 13 running on Manjaro. However the per-Key-RGB Keyboard is not recognized. 

[image]

The model is shown correct, but as you can see, the Custom Effects are registered as not supported. 

This is the output of `lsusb | grep 1532`:

`Bus 003 Device 004: ID 1532:026a Razer USA, Ltd Razer Blade`

It says Razer Blade, but that ist not correct. Maybe the wrong Id causes the issue with the missing effects support?

z3ntu, 2021-12-12:
@Khraut Thanks for reporting. Please try this branch and see if it works there https://github.com/z3ntu/openrazer/tree/razer_book_2020_custom

As noted in the commit itself, I don't know the matrix dimensions for the device. You can try and see which fields are used or you can make a USB packet capture from Razer Synapse (Windows), see wiki for the details.

Khraut, 2021-12-13:
Thank you for your response!

I would like to try this branch, but i don`t have the needed knowledge of the commands needed for building this myself. I used pacman for the official one. Would you like tho give a short instruction of how i can build it myself?

z3ntu, 2022-01-15:
It's documented in the wiki here: https://github.com/openrazer/openrazer/wiki/Building-a-package#arch-linux

closed 2022-07-19 by github-actions[bot]