openrazer/openrazer #1706: Support for Viper 8k

gitrazorhub, 2021-12-15:
### Device Info

* Name: <See under device>
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```

```
Please post the output of
    ls -lah /dev/input/by-id/
```

### Packet Captures
[image]

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

gitrazorhub, 2021-12-15:
Will we be able to reprogram the side buttons?

z3ntu, 2022-01-15:
This device is already supported since OpenRazer v3.2.0. Side buttons can be remapped with e.g. https://github.com/sezanzeb/input-remapper

closed 2022-01-15 by z3ntu