openrazer/openrazer #1707: REL_WHEEL events from keyboard (Black widow v3 pro) replacing volume control to vertial scroll

vazir, 2021-12-15:
### Distribution

Ubuntu 21.10

Keyboard Black Widow v3 Pro

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

After installling OpenRazer drivers, the reel multimedia (normally working as a volume UP/Down) - got reassigned as mouse vertial scroll. Corresponding events also fired from razer - evtest producing events of EV_REL REL_WHEEL value -1 and 1 and 
REL_WHEEL_HI_RES value -120 and 120

Had to remove drivers to return nurmal volume control behavior

GeoDoX, 2021-12-15:
I’m aware of this (after seeing another issue/pr for another device). Unfortunately I won’t be able to get to it until after the holidays. Anyone else is very welcome to fix this if they’re feeling up to it, from what I understand it’s a relatively easy fix.

vazir, 2021-12-15:
> I’m aware of this (after seeing another issue/pr for another device). Unfortunately I won’t be able to get to it until after the holidays. Anyone else is very welcome to fix this if they’re feeling up to it, from what I understand it’s a relatively easy fix.

If you could roughly point where to look, I might be able to do it. Is it a kernel module issue or just a configuration/mapping?

GeoDoX, 2021-12-15:
@vazir check recent issues/prs in the past week or so

vazir, 2021-12-18:
After a little digging, I found the mouse events appears when the daemon is running, and especially when the "driver mode" is enabled in the keyboards.py, so as a quick dirty hack, I disabled the driver mode for myself (#self.set_device_mode(0x03, 0x00)  # Driver mode), as I only need to be able to make backlit workable (no built-in modes are suitable for daily work, you just need adjustable backlit keys). Of course it's not a solution, but at least some solution for the time being.

GeoDoX, 2021-12-19:
> After a little digging, I found the mouse events appears when the daemon is running, and especially when the "driver mode" is enabled in the keyboards.py, so as a quick dirty hack, I disabled the driver mode for myself (#self.set_device_mode(0x03, 0x00) # Driver mode), as I only need to be able to make backlit workable (no built-in modes are suitable for daily work, one just needs adjustable backlit keys). Of course it's not a solution, but at least some solution for the time being.

https://github.com/openrazer/openrazer/issues/1398#issuecomment-961242268 Was the solution I found, but it seems you’ve already found it. I’ll do more digging into seeing if it’s a complete fix (doesn’t break other things)

vazir, 2021-12-20:
> [#1398 (comment)](https://github.com/openrazer/openrazer/issues/1398#issuecomment-961242268) Was the solution I found, but it seems you’ve already found it. I’ll do more digging into seeing if it’s a complete fix (doesn’t break other things)

Not really? i've looked into the logs and code, and tried to isolate the part, after which the behavior breaks. Seems I was just not the first :)

AviMoto, 2022-03-09:
Same issue on Fedora 35 5.16.12-200.fc35.x86_64
with OpenRazer 3.2

closed 2022-08-08 by github-actions[bot]