openrazer/openrazer #1709: Linux Mint help

EighterMOA, 2021-12-17:
Hello everyone!!
Fairly new to linux but being self taught :shrug: 

I have a question, I am trying to get my Razer tartarus pro to work in Linux mint.  I followed the instructions  in the screenshot below.  I dont really know what to do after this, I don't know how to "run" the program or where to go from here.  Any guidance would be appreciated!!  TIA
[image]

lah7, 2021-12-17:
Hello! OpenRazer on its own is just a driver and daemon for other applications/scripts to interact with. You'll need to install an application ("front end"). Choose one of your liking as listed [near the top of the homepage](https://openrazer.github.io/) or from the [README](https://github.com/openrazer/openrazer#applications).

Since you have a Tartarus keypad, OpenRazer doesn't offer a key rebinding facility, since it's mainly focused on lighting. Key mapping is fairly generic, [key-mapper](https://github.com/sezanzeb/key-mapper) seems to be a popular choice.

EighterMOA, 2021-12-17:
Thanks!!!!  That is all I want is to remap my keys.  I'll try key mapper
and see what I can break :-)

closed 2021-12-17 by lah7

EighterMOA, 2021-12-18:
So i have key mapper installed.  When i push the numerical keys on the
keypad to remap them, it does not register anything.   when I push the
thumb buttons it works fine.  am i missing something?

lah7, 2021-12-18:
Razer's keypads behave like keyboards, so it should emit regular key strokes.

I just realised that you said Tartarus **Pro** (I originally misread as the original Tartarus). Turns out your device isn't supported by OpenRazer yet (#1039) so lighting won't configurable, but it should act as a generic input device.

You could try uninstalling OpenRazer since it's no use for your keypad right now. Pressing CTRL+ALT+T and using the Terminal is one way to do that:

    sudo apt autoremove openrazer-meta openrazer-driver-dkms python3-openrazer 

You could also try asking the folks in https://github.com/openrazer/openrazer/issues/1039#issuecomment-979505624 who own this keypad to see if they have any advice.