openrazer/openrazer #1714: Ripple thread overrides custom effects

lah7, 2021-12-23:
> Originally reported in https://github.com/polychromatic/polychromatic/issues/372

If the daemon is playing the "ripple" effect and the user wishes to play a custom effect via a script or application, it ends up competing with the daemon as they are both drawing frames, causing flickers between the two.

The workaround is to apply a hardware effect (such as None, or Static) and then proceed with the custom effect, so the daemon terminates the ripple thread. Currently, the daemon does not persist or have any knowledge about custom API calls.

Is this something that should be fixed in OpenRazer?

### 💡 Ideas

1. Add a conditional check in `draw()` that if ripple is running, terminate thread?
    * Maybe not? Sounds inefficient to me. 🐢 
1. Add a conditional check in the ripple thread to terminate if `draw()` was called?
1. A new function to the API like `fx.stop_ripple()` or `fx.advanced.init()` so the daemon can terminate the ripple thread upon request.
1. Leave it up to the script/app to pick a hardware effect.

What are your thoughts?

SamantazFox, 2021-12-23:
If this is required, I'll gladly provide the USB PCAP of the official Razer app for that keyboard (via a Linux host, a Windows VM and USB passthrough).

lah7, 2021-12-23:
@SamantazFox Thanks for the offer, but it's pretty much specific to the OpenRazer daemon. [There's a ripple thread](https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/misc/ripple_effect.py) in the background that needs to stop, _somehow_. 🤔  Technically, Ripple is a custom software effect.

z3ntu, 2022-01-15:
I would assume that once setCustom is called that the effect is switched to that, like you switch to any effect. So that ripple stops being ripple.

z3ntu, 2023-08-17:
This was discussed in chat a bit, my suggestion: make an internal setCustom function that e.g. ripple is using as `def refresh_keyboard :         self._parent.setCustom()`.

And then make the public setCustom function do the notify that tells everybody custom effect (non-ripple) is on. Then the `daemon/openrazer_daemon/misc/ripple_effect.py` line 217 `def notify` notices this and stops doing the ripple. And this public function should use the internal function to actually write to sysfs.

(hope this summary is clear)

z3ntu, 2023-08-23:
Please check if #2117 resolves this.

lah7, 2023-08-24:
Can confirm that fixes the problem, thanks 👍🏼

commit 636ead8c (2023-08-25): daemon: notify for new effect when calling setCustom/setKeyRow

The RippleManager needs to know when to stop itself and since this
happens via the notify mechanism, we do need to tell the ripple to stop
when someone wants to set a custom effect.

However we need to make sure to not stop when ripple is itself using the
APIs, to make a 'private' version of setKeyRow and setCustom that
doesn't have the notify code, and only add the notify to the public
DBus-accessible function.

Fixes #1714
PR #2117 (2023-08-25): daemon: notify for new effect when calling setCustom/setKeyRow
The RippleManager needs to know when to stop itself and since this happens via the notify mechanism, we do need to tell the ripple to stop when someone wants to set a custom effect.

However we need to make sure to not stop when ripple is itself using the APIs, to make a 'private' version of setKeyRow and setCustom that doesn't have the notify code, and only add the notify to the public DBus-accessible function.

Fixes #1714

---
(note to myself: merge https://github.com/z3ntu/RazerGenie/commits/custom-effect-set-fix after this pat [... cut]
closed 2023-08-25 by z3ntu