openrazer/openrazer #1717: Unresponsive M(1-5) keys after suspend

stephensrmmartin, 2021-12-26:
### Distribution

Arch Linux (12/26/2021)

Razer BlackWidow Ultimate 2013
### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

This issue has existed for a long time, but I have repeatedly forgotten to inquire about it.

After resuming after a machine suspend/sleep, the M1-M5 keys will not work; e.g., they are not read by xev, nor do they active what they are bound to.

This is fixed easily by running systemctl --user restart openrazer-daemon. It has become a bit of a ritual to just open a terminal and type `!?openrazer` to fix it.

Razer log:

```
2021-12-26 14:56:39 | razer                          | INFO     | Initialising Daemon (v3.2.0). Pid: 1028782
2021-12-26 14:56:39 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationV2Chroma ---------------- (1532:0F20)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskEssential ------------------ (1532:0065)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateReceiver ----------- (1532:0088)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateWired -------------- (1532:0086)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV2 ------------------------- (1532:0085)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowTournamentEdition2014 ---- (1532:011C)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3 ----------------------- (1532:024E)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3MiniHyperspeed --------- (1532:0258)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3MiniHyperspeedWireless - (1532:0271)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3ProWired --------------- (1532:025A)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3TK --------------------- (1532:0A24)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlade142021 ------------------------ (1532:0270)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlade15Advanced2020 ---------------- (1532:0253)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlade15Advanced2021 ---------------- (1532:0276)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlade15AdvancedEarly2021 ----------- (1532:026D)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlade17Pro2021 --------------------- (1532:0279)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018 -------------------------- (1532:0233)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Base ---------------------- (1532:023B)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Mercury ------------------- (1532:0240)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Adv ----------------------- (1532:023A)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Base ---------------------- (1532:0246)
2021-12-26 14:56:39 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019StudioEdition ------------- (1532:024D)
[... 156 lines cut ...]
2021-12-26 14:56:43 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't convert event to key name
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/openrazer_daemon/misc/key_event_management.py", line 381, in key_action
[... cut ...]
2021-12-26 14:57:00 | razer.screensaver              | DEBUG    | Received screensaver inactive signal
2021-12-26 14:57:00 | razer.device0                  | INFO     | Resuming RazerBlackWidowUltimate2013
2021-12-26 14:57:00 | razer.device0                  | DEBUG    | DBus call set_brightness
2021-12-26 14:57:08 | razer                          | DEBUG    | State recently changed, writing to disk
2021-12-26 14:57:08 | razer                          | DEBUG    | Writing persistence config

```

Please let me know what else I can provide to help with this.

lah7, 2021-12-27:
Can confirm. It's because the [device has two modes](https://github.com/openrazer/openrazer/wiki/Using-the-keyboard-driver#device_mode---set-device-mode) - a "normal" mode (`0`) and "driver" mode (`3`), the latter enables the M1-M5 & macro recording keys. When the daemon starts, the keyboard is put into "driver" mode. If the BIOS/UEFI is turning off the USB ports, then this probably triggers the power cycle and the device is back to mode `0` when the system wakes up. 

This doesn't always happen to me, possibly because my peripherals are plugged into the monitor, acting as a powered USB hub. Likewise, I've been blind to this too. Whenever it happens, I just quickly run `openrazer-daemon -r`.

You can check the mode with this command:

```bash
cat /sys/bus/hid/drivers/razerkbd/**/device_mode
```

Not sure what's the best approach to fix this, since AFAIK the daemon can't capture the macro keypress, as it's the firmware that causes them to do nothing / flash M repeatedly in mode `0`. A workaround could be to check your BIOS/UEFI settings and disable any USB power saving features.

Maybe a workaround/fix could be to restart the systemd user service after suspend?

* https://unix.stackexchange.com/questions/329445/systemd-stop-service-before-suspend-restart-after-resume
* https://freedesktop.org/software/systemd/man/systemd-suspend.service.html

stephensrmmartin, 2021-12-30:
Hm, I can't seem to get the restart-on-suspend solution to work cleanly.

I tried a few ways; one was making a parameterized system service that runs as my UID, then starts/stops the service via systemctl --user start|stop. But that didn't work at all, for reasons I don't understand.

Another was a restart-after-suspend system service custom-made for openrazer-daemon; this also runs as UID. But systemd does not seem to recognize that it started; so it timesout and kills the process (and daemon) after a few minutes. It used openrazer-daemon -rF as the ExecStart, after suspend.target. This required manually setting the runtime directory as well, since system services don't set the correct env. variables for desktop apps to run.

Another was replacing the openrazer-daemon.service file, with a parameterized system service file; it is enabled by suspend.target, sleep.target, and default; it takes a UID and sets some things up accordingly. But it had a similar error to the above - some env. variable not correct.

Ideally, this would be a systemd-level change. There should be a way to run *user* service files following *system* events. That way we could simply state "openrazer-daemon.service should be run as -rF, and started after each suspend.target or sleep.target is finished". But there's no easy way to do so currently (I think I saw some post that had some somewhat-janky set up that monitored the system dbus for any sleep events, which would then trigger a user-level sleep.target, but that's a messy way to do something that systemd should do).

stephensrmmartin, 2021-12-30:
Found a solution (?):

## /etc/systemd/system/restart-user-service-after-resume.service

```
[Unit]
Description=Restarts yourUserName user service %i after resume
Before=sleep.target
Before=suspend.target
StopWhenUnneeded=yes

[Service]
Type=oneshot
# Change below to be your user name
User=yourUserName
# Change below to have your UID; required for systemctl --user calls to work, otherwise the dbus session is not accessible
Environment=XDG_RUNTIME_DIR=/run/user/1000
RemainAfterExit=yes
ExecStart=/usr/bin/systemctl --user stop %I
ExecStop=/usr/bin/systemctl --user start %I

[Install]
WantedBy=sleep.target
WantedBy=suspend.target

```

```
sudo systemctl enable restart-user-service-on-resume@openrazer-daemon
```

stroyan, 2023-02-09:
This can be handled rather easily in openrazer_daemon.  It is already seeing the resume and setting keyboard lighting.
Adding a "_resume_device" that sets the device mode gets the macro keys working again.
Here is such a change.

This does assume a recent version where the _resume_device function is available to override.
That only went in a couple of months ago, more recently than release 3.5.0.

``diff --git a/daemon/openrazer_daemon/hardware/keyboards.py b/daemon/openrazer_daemon/hardware/keyboards.py
index e34ad82..32b0ab1 100644
--- a/daemon/openrazer_daemon/hardware/keyboards.py
+++ b/daemon/openrazer_daemon/hardware/keyboards.py
@@ -44,6 +44,10 @@ class _MacroKeyboard(_RazerDeviceBrightnessSuspend):
 
         self.key_manager.close()
 
+    def _resume_device(self):
+        self.logger.info('Putting device back into driver mode. Daemon will handle special functionality')
+        self.set_device_mode(0x03, 0x00)  # Driver mode
+
 
 class _RippleKeyboard(_MacroKeyboard):

z3ntu, 2023-02-11:
@stroyan Hm sort of makes sense that the device is basically reset after being turned off during computer suspend. `_resume_device` is really kind of a wrong place to put this since it's only called on lockscreen unlock but I guess the best place we can really put it? Should at least fix this in most cases. If you could add a comment to the function saying something like this, and reference computer suspend then please open a PR with this.

stroyan, 2023-02-11:
I have created a pull request for this.
#2020

PR #2020 (2023-08-20): daemon: restore keyboard driver mode on resume
Suspend may power down a keyboard and stop sending macro keys.

Fixes #1717
closed 2023-08-20 by z3ntu