openrazer/openrazer #1720: Kernel Mod razerkbd: modprobe: ERROR: could not insert 'razerkbd': No such device

Btrc4t, 2021-12-29:
Hello,
I am running Kali Rolling and I installed OpenRazer for Debian Testing
Almost everything is fine except that the kernel mod doesn't seem to be working for me and the troubleshooting section doesn't cover my issue.
Here's what dmesg has to say:
```
$ sudo dmesg | grep razer                                                                                     
[   17.797159] razerkbd: system not supported, no HID or USB support
```
Also modprobe:
```
$ sudo modprobe razerkbd 
modprobe: ERROR: could not insert 'razerkbd': No such device
```

z3ntu, 2021-12-29:
The official Debian package broke but apparently it should be fixed with version 3.2.0+dfsg-3. 

Also please note that we only support the packages from our repository that is linked on the website https://openrazer.github.io/#debian You have the version from the Debian repositories installed.

Btrc4t, 2021-12-29:
> The official Debian package broke but apparently it should be fixed with version 3.2.0+dfsg-3.
> 
> Also please note that we only support the packages from our repository that is linked on the website https://openrazer.github.io/#debian You have the version from the Debian repositories installed.

I've installed openrazer-meta by adding the OpenSuse build service repository for Debian Testing so it is official.
The latest version at this moment for me through this channel seems to be 3.2.0+dfsg-1.

Is there any way to test 3.2.0+dfsg-3 now? Do I have to build it?

z3ntu, 2021-12-29:
I guess our 3.2.0-0 version is considered to be lower than Debian's 3.2.0+dfsg-3. You should be able to grab the -3 deb from Debian sid and install it on your machine until Debian bookworm/testing has been unbroken.

Btrc4t, 2021-12-29:
Attempting to install the .deb package from Debian Testing as well as the one from Debian sid results in:
```
$ sudo dpkg -i openrazer-meta_3.2.0-0_all.deb                                                                                                                                                                                                 1 ⨯
(Reading database ... 423914 files and directories currently installed.)
Preparing to unpack openrazer-meta_3.2.0-0_all.deb ...
Unpacking openrazer-meta (3.2.0-0) over (3.2.0-0) ...
dpkg: dependency problems prevent configuration of openrazer-meta:
 openrazer-meta depends on openrazer-driver-dkms (= 3.2.0-0); however:
  Package openrazer-driver-dkms is not installed.
 openrazer-meta depends on openrazer-daemon (= 3.2.0-0); however:
  Package openrazer-daemon is not installed.
 openrazer-meta depends on python3-openrazer (= 3.2.0-0); however:
  Package python3-openrazer is not installed.

dpkg: error processing package openrazer-meta (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 openrazer-meta
 ```

barti2du, 2021-12-29:
> sudo dpkg -i

This is to be expected: either don't use _dpkg_ or install all relevant packages at once (_dpkg_ even tells their names for You).

z3ntu, 2021-12-29:
You're currently confusing the openrazer repo vs Debian repo. 3.2.0-0 comes from openrazer repo (the one on build.opensuse.org), 3.2.0+dfsg-* comes from Debian repo which has a higher revision number so it gets preferred by apt and you cannot easily "downgrade" to 3.2.0-0. The version 3.2.0+dfsg-1 from Debian repo was broken and is fixed with 3.2.0+dfsg-3 but this version isn't yet in Debian "bookworm" that you're using, so you should be able to manually install this deb from Debian sid and have this fixed.

=> http://ftp.de.debian.org/debian/pool/main/o/openrazer/openrazer-driver-dkms_3.2.0+dfsg-3_all.deb

Btrc4t, 2021-12-29:
> > sudo dpkg -i
> 
> This is to be expected: either don't use _dpkg_ or install all relevant packages at once (_dpkg_ even tells their names for You).

I tried to manually install them but that didn't work

> You're currently confusing the openrazer repo vs Debian repo. 3.2.0-0 comes from openrazer repo (the one on build.opensuse.org), 3.2.0+dfsg-* comes from Debian repo which has a higher revision [...]

Thanks, you were right, I had no idea the package was already present in one of the repositories I have:
```
sudo apt policy openrazer-meta
openrazer-meta:
  Installed: 3.2.0+dfsg-1
  Candidate: 3.2.0+dfsg-1
  Version table:
 *** 3.2.0+dfsg-1 500
        500 http://http.kali.org/kali kali-rolling/main amd64 Packages
        500 http://http.kali.org/kali kali-rolling/main i386 Packages
        100 /var/lib/dpkg/status
     3.2.0-0 500
        500 http://download.opensuse.org/repositories/hardware:/razer/Debian_Unstable  Packages
```

Btrc4t, 2021-12-29:
For whoever may find it useful, I installed the version 3.2.0-0 by creating a new file:
```
sudo nano /etc/apt/preferences.d/preferences
```
With the following contents:
```
Package: openrazer-meta
Pin: origin download.opensuse.org
Pin-Priority: 1001
```
Then simply installed it again with `sudo apt install openrazer-meta`

Now the initial issue has been solved, if you wish to update the troubleshooting guide you may use this ticket, otherwise I or someone else could close it.

z3ntu, 2021-12-29:
Not sure if you noticed but OpenRazer consists of several Debian packages, they should all be in sync version-wise:
* openrazer-daemon
* openrazer-doc
* openrazer-driver-dkms
* openrazer-meta
* python3-openrazer

Btrc4t, 2021-12-29:
> Not sure if you noticed but OpenRazer consists of several Debian packages, they should all be in sync version-wise:
> 
>     * openrazer-daemon
> 
>     * openrazer-doc
> 
>     * openrazer-driver-dkms
> 
>     * openrazer-meta
> 
>     * python3-openrazer

They are in sync with the method I detailed above, the meta package takes care of that

z3ntu, 2021-12-30:
Ok good.

closed 2021-12-30 by z3ntu