openrazer/openrazer #1724: Missing kernel headers for openrazer3.2, header for kernel 5.13.

hajnyckel, 2022-01-01:
### Manjaro Linux, 21.2.0

### Support

Greetings all. 
I am running an issue that might be nothing to be concerned about, but as I am still now to Linux I thought to ask first. 
I got my self a new Razer keyboard and mouse (had Logitech before) and now was trying to install the OpenRazer driver package using the instruction from the ArchRepo but that git link does not work so I searched for a guide and found a YouTube video. The video seems to be working so I followed that guide. First of I installed the "openrazer-daemon-git" and the installation was a success but for this errors:
 `Install DKMS modules                                                                                                                                                     [3/3]
==> ERROR: Missing mnt kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing bin kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing root kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing run kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing usr kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing home kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing srv kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing boot kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing tmp kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing rootfs-pkgs.txt kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing var kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing sbin kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing sys kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing lost+found kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing lib64 kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing etc kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing data kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing proc kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing desktopfs-pkgs.txt kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing dev kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing media kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing lib kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing opt kernel headers for module openrazer-driver/3.2.0.`
After some checking, I found out that I got headers 5.12.3-1 with kernel 5.13.19-2 and thus the reason for the errors, I think. However I cannot find any headers to match my kernel and I need the openrazer 3.2 for my keyboard and mouse. 
So, what do I do now?

z3ntu, 2022-01-01:
Install `linux513-headers`?

hajnyckel, 2022-01-01:
@z3ntu, thank you for your suggestion. Did try it but all I got as a response was `Error: target not found: linux513-headers
`

lah7, 2022-01-01:
Kernel versions 5.12 and 5.13 are [obsolete](https://en.wikipedia.org/wiki/Linux_kernel_version_history), so their packages have disappeared. You'll want 5.10 or 5.15 which is the latest long term kernel (LTS). Typically Arch has a `linux-lts` and `linux-lts-headers` package that follows the latest LTS version, but I'm not sure if [Manjaro has something similar](https://manjaro.org/branch-compare/), as they package older (but still supported) kernels.

This command will update your system, and install the newer kernel:

    sudo pacman -Syu linux-lts linux-lts-headers

If that doesn't exist:

    sudo pacman -Syu linux515 linux515-headers

I believe Manjaro has a graphical interface for managing the kernel, which you could use instead. If you're on NVIDIA, you may need `linux515-nvidia` and similar.

Installing the new kernel should prompt DKMS to build OpenRazer's driver. After rebooting - assuming it's booted into the new kernel - things should begin to work. You can then safely uninstall older kernel packages you have installed:

    pacman -Q | grep linux

hajnyckel, 2022-01-01:
@lah7, thank you for your reply. I had no idea that kernel 5.13 was obsolete and when I checked I had disabled notifications on new kernels in Manjaro so that is why I had not updated it yet. SO updated kernel to 5.15 and then reinstalled `openrazer-daemon-git` and as far as I can see there were no errors, so it all seems to be installed correctly. Also uninstalled the 5.13 kernel and rebooted with no issue. Hope the rest of the installation for openrazer will be trouble free.

Thank you so much!

lah7, 2022-01-01:
You're welcome! Feel free to reach out again if you need help.

💡  Not sure which guide you're following, but if you're looking for an easy way to interact with the driver, you'll likely want to install  `polychromatic` or `razergenie` next.  [(See website for others)](https://openrazer.github.io/)

For stability and to receive update notifications, I'd recommend the non `-git` packages. Uninstalling the `-git` versions and then installing `openrazer-meta` should do the trick. The `-git` packages are pretty much snapshots of the latest development branch.

closed 2022-01-01 by lah7