openrazer/openrazer #1725: Install on Kali - Debian 11

silentwitness9, 2022-01-01:
Hi all - so I've installed this with Polychromatic on my 2020 Blade Stealth running Kali. 
All installed fine, but I keep getting an unrecognised device error when running poloychromatic. 

$ sudo dkms install openrazer-driver/3.2.0 
Module openrazer-driver/3.2.0 already installed on kernel 5.15.0-kali2-amd64 (x86_64).

$ sudo modprobe razerkbd 
modprobe: ERROR: could not insert 'razerkbd': No such device

It can't seem to find the keyboard although it is listed as a supported device. 

Any help greatly appreciated. Thanks

lah7, 2022-01-01:
Hi. According to https://github.com/openrazer/openrazer/issues/1720, `openrazer-driver-dkms` in Debian's repository was/is broken and ends up overriding the one published in [OpenRazer's repository.](https://software.opensuse.org/download.html?project=hardware%3Arazer&package=openrazer-meta)

The solution is to manually install the new (but not yet published) package by Debian - see https://github.com/openrazer/openrazer/issues/1720#issuecomment-1002513728 - or lock OpenRazer's version of the package - https://github.com/openrazer/openrazer/issues/1720#issuecomment-1002522328.

After the faulty package is addressed, things should start working after a reboot.

silentwitness9, 2022-01-02:
Thanks! Working now, cheers for the quick help.

closed 2022-01-02 by silentwitness9