openrazer/openrazer #1726: Implement set_scroll_active and set_logo_active persistence for Razer DeathAdder 3500

braiam, 2022-01-01:
Every time I bootup my pc, I have to turn off the scroll and logo led off, since the device gets hot with normal usage. There seems to be some [facilities for Chroma](https://github.com/openrazer/openrazer/blob/89b2e5808bd20b99ccb871957c4b0ffc51f612a6/daemon/openrazer_daemon/dbus_services/dbus_methods/deathadder_chroma.py#L285) to remember this setting. I'm not sure if a straight up copying the line to mice.py since there seems to be enough facilities in device_base to figure out what to do.

z3ntu, 2022-01-15:
As a workaround you can write a script that uses the OpenRazer Python API to just turn off the lighting, see `examples/` directory in the repo. But I don't think a device should get any hot only because some LEDs are on, sounds a bit like a hardware defect to be honest.

z3ntu, 2022-07-23:
I think now I understand what you mean. Please try this patch to see if it works for you:

```
diff --git a/daemon/openrazer_daemon/hardware/mouse.py b/daemon/openrazer_daemon/hardware/mouse.py
index 7f23c2e9..a842c613 100644
--- a/daemon/openrazer_daemon/hardware/mouse.py
+++ b/daemon/openrazer_daemon/hardware/mouse.py
@@ -1641,6 +1641,7 @@ class RazerDeathAdder3500(__RazerDeviceSpecialBrightnessSuspend):
     USB_VID = 0x1532
     USB_PID = 0x0054
     METHODS = ['get_device_type_mouse',
+               'get_logo_active', 'set_logo_active', 'get_scroll_active', 'set_scroll_active',
                'get_logo_brightness', 'set_logo_brightness', 'set_logo_static', 'set_logo_pulsate', 'set_logo_blinking',
                'get_scroll_brightness', 'set_scroll_brightness', 'set_scroll_static', 'set_scroll_pulsate', 'set_scroll_blinking',
                'max_dpi', 'get_dpi_xy', 'set_dpi_xy', 'get_poll_rate', 'set_poll_rate']

```

braiam, 2022-07-23:
Maybe razergeany isn't doing the right calls, since `~/.config/openrazer/persistence.conf` isn't updated.

z3ntu, 2022-07-23:
Try D-Feet application, select Session Bus at top, search for org.razer, open /org/razer/device/YOUR_SERIAL, then razer.device.lighting.logo, do a call setLogoActive with parameter `False` and see if the device changes.

braiam, 2022-07-23:
It's changing the settings on the device, but it's isn't doing anything on the persistence file.

z3ntu, 2022-07-23:
That sounds like step 1 at least. So before OpenRazer didn't have a way to turn off the logo/scroll LED and now it does, right? I'll check if we're missing any other code.

braiam, 2022-07-23:
No, I could manipulate the status of the led. The daemon just forgets it after a boot up/restart and unconditionally sets it to on.

z3ntu, 2022-07-23:
By turning the brightness to zero?

braiam, 2022-07-23:
No, On/off. The patch basically didn't modify the behavior that wasn't available before. The device only have two state (On/off) for both leds. The feature missing here is persistence. That if I set it off and reboot the machine it should remember that setting.

z3ntu, 2022-07-23:
How did you turn it off before? Without any patch this is how polychromatic looks for me (with the fake driver):

[image]

z3ntu, 2022-07-23:
Also please share `lsusb | grep 1532` I feel like we're talking about different devices here..

braiam, 2022-07-23:
`Bus 001 Device 005: ID 1532:0016 Razer USA, Ltd DeathAdder 3.5G`, also, this is how it looks to me:

[image]

z3ntu, 2022-07-23:
That explains a good chunk of it, I was talking about `1532:0054` which in OpenRazer is the "Razer DeathAdder 3500"

I do recall there being some confusion (also from Razer side) in the naming of these old devices.

z3ntu, 2022-07-23:
Please try this patch:

```diff
diff --git a/daemon/openrazer_daemon/hardware/device_base.py b/daemon/openrazer_daemon/hardware/device_base.py
index 97366356..3f5d1d72 100644
--- a/daemon/openrazer_daemon/hardware/device_base.py
+++ b/daemon/openrazer_daemon/hardware/device_base.py
@@ -211,7 +211,7 @@ class RazerDevice(DBusService):
                 self.add_dbus_method(m[0], m[1], m[2], in_signature=m[3], out_signature=m[4])
 
         for i in self.ZONES[1:]:
-            if 'set_' + i + '_static' in self.METHODS or 'set_' + i + '_static_naga_hex_v2' in self.METHODS or '`set_' + i + 'active' in self.METHODS:
+            if 'set_' + i + '_static' in self.METHODS or 'set_' + i + '_static_naga_hex_v2' in self.METHODS or 'set_' + i + '_active' in self.METHODS:
                 self.zone[i]["present"] = True
                 for m in effect_methods[i]:
                     self.logger.debug("Adding {}.{} method to DBus".format(m[0], m[1]))

```

Now the device seems to show up in persistence.conf

braiam, 2022-07-23:
Yep, now that does it. Also shows a bunch of other config that probably wasn't being saved.

braiam, 2022-07-23:
Well, dpi seems to be consistently not in the file now, while before it was consistently in. Guess that broke somewhere else.

z3ntu, 2022-07-23:
DPI should be fixed by https://github.com/openrazer/openrazer/pull/1848 , can you check?

braiam, 2022-07-23:
Yep, it does appear. If anyone is debugging a similar issue, it may take a while for the daemon to update the persistence file.

z3ntu, 2022-07-23:
So all good with these two patches? Or do you see anything missing or wrong still?

braiam, 2022-07-23:
Yep, it seems to store the current status.

z3ntu, 2022-07-24:
The other patch I'll merge once the comments there are resolved. Thanks for reporting!

commit 46e18195 (2022-07-24): Fix typo in persistence init

This causes some devices like DeathAdder 3.5G not to save to persistence
properly.

Fixes #1726
closed 2022-07-24 by z3ntu