openrazer/openrazer #1727: Support for [Razer Blade 15 Advanced late 2019]

napsterred, 2022-01-02:
Please incorporate device support changes for dev id 24B   of Razer Blade 15 Advanced Late 2019  from the link

https://github.com/napsterred/openrazer.github.io

napsterred, 2022-01-09:
i have altered the code to make it work with the model , pleas incorporate the changes in main open razer program so i can install it from GitHub ,

z3ntu, 2022-01-15:
Can you please open a proper pull request with your changes, and not stuff them into the website repo?
If not possible at all I can spend some time extracting your changes but it's rather tedious.

napsterred, 2022-01-16:
I don't know how to do it, its my first try at working on GitHub for anything, if ya send me instructions I could try.

z3ntu, 2022-01-27:
I'm sure you can find many tutorials online but I unfortunately don't have the time to explain to you in detail.

napsterred, 2022-01-27:
I almost cramped up everything trying to do it, **if you guys could extract the changes** that would be awesome. Even if a specific dev helps me out i could patreon them.

lah7, 2022-01-27:
Done - #1737, as I had some time to spare. I took a copy of your files, deleted the website files, [cloned `openrazer/openrazer`](https://github.com/openrazer/openrazer), pasted them and checked/updated the diffs using the Git client so it can be integrated into main repository.

z3ntu, 2022-04-07:
Closing as #1737 has been created from this, we still have #1041 to track the device in general.

PR #1737 (2024-04-06): Add support for Razer Blade 15 Advanced (Late 2019)
Changes extracted from #1727. Credit goes to @napsterred.

I don't own this device, so can't confirm 100% working, but the changes look good to me.

Closes #1727.
Fixes #1041 (under the assumption the optical keyboard already works or is out of scope for OpenRazer?)
closed 2022-04-07 by z3ntu