openrazer/openrazer #1728: Razer Blackwidow V3(Full Size) Game Mode not working

FirephoenixX02, 2022-01-04:
When I try to enable Gamemode on my Razer Blackwidow V3(Full Size), I get an Error and all System Shortcuts don't work anymore. For example:  <kbd>Alt + Tab</kbd> for switching between Windows etc. My Controller(Frontend) is Polychromatic.

lah7, 2022-01-04:
It's expected for Alt+Tab, Alt+F4 and Win key to be disabled at a firmware level when game mode is active, but I suspect something wasn't implemented correctly for the toggle to work for this keyboard via software.

What does the error say? Could it be the same as https://github.com/openrazer/openrazer/pull/1572#issuecomment-926585972?

FirephoenixX02, 2022-01-04:
It's the same as #1572

wyattearp, 2022-02-23:
FYI, I've tried this on the current head (af1e2b0) and it appears to function. Have you tried testing with a build from the master branch?

FirephoenixX02, 2022-06-23:
I will try it again and if it works i'll close the issue.

FirephoenixX02, 2022-06-24:
The Game Mode itself works now. But i still get an error. This time its a Permission denied error.

wyattearp, 2022-06-29:
Additional details about what permission error you're receiving would be helpful to determine if it's an issue with the Gaming Mode function or something else.

nightsky30, 2022-08-07:
@FirephoenixX02 I too would be interested in the error message you are seeing. I initially had the same issue https://github.com/openrazer/openrazer/pull/1572#issuecomment-926585972 with game mode and the Huntsman V2.

I've not seen your latest error message, but he BlackWidow V3 might benefit from a similar change as was made for the Huntsman V2. If you're interested, would you mind testing toggle of game mode with this branch which includes that minor change? https://github.com/nightsky30/openrazer/commit/60c3b387cc5a9b1c993d3fcdc3cd96db0c751328

FirephoenixX02, 2022-08-12:
> Additional details about what permission error you're receiving would be helpful to determine if it's an issue with the Gaming Mode function or something else.

I will check tomorrow if i still have the error message and if i do in will post it here.

FirephoenixX02, 2022-08-12:
> @FirephoenixX02 I too would be interested in the error message you are seeing. I initially had the same issue https://github.com/openrazer/openrazer/pull/1572#issuecomment-926585972 with game mode [...]

I will check if there is still an error and if there is one i'll try the branch you said i should test.

anyoneyun, 2022-08-17:
Same issue here. Game mode appears to be activated but the led on the keyboard doesn’t light up and game mode can’t be deactivated with the key combo. Polychromatic shows a “permission denied” error when game mode is activated through the “Game mode” button. The led seems to work after I put my system to suspend and then I turn it on again. Also, I’m unable to disable game mode with the key combo and polychromatic shows an error but the latter works: it disables and enables game mode but the led doesn’t light up when activated. I’m using Manjaro, here’s the error shown in polychromatic:
`Traceback (most recent call last):
  File ’/usr/lib/python3.10/site-packages/polychromatic/backends/openrazer.py’, line 1017, in set_device_state
    rdevice.game_mode_led = option_data
  File ’/usr/lib/python3.10/site-packages/openrazer/client/devices/keyboard.py’, line 36, in game_mode_led
    self._dbus_interfaces[’game_mode_led’].setGameMode(True)
  File ’/usr/lib/python3.10/site-packages/dbus/proxies.py’, line 141, in call
    return self._connection.call_blocking(self._named_service,
  File ’/usr/lib/python3.10/site-packages/dbus/connection.py’, line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Python.PermissionError: Traceback (most recent call last):
  File ’/usr/lib/python3.10/site-packages/dbus/service.py’, line 715, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File ’/usr/lib/python3.10/site-packages/openrazer_daemon/dbus_services/dbus_methods/chroma_keyboard.py’, line 104, in set_game_mode
    open(super_file, ’wb’).write(b’\x01’)
PermissionError: [Errno 13] Permission denied: ’/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-10/1-10:1.3/0003:1532:024E.0009/key_super’ `

nightsky30, 2022-08-27:
@anyoneyun Care to try the change in this [branch](https://github.com/nightsky30/openrazer/commit/60c3b387cc5a9b1c993d3fcdc3cd96db0c751328) ?

anyoneyun, 2022-08-27:
Sure, how can I do that?

nightsky30, 2022-08-27:
> Sure, how can I do that?

To clone that branch with the **potential** fix:

```
git clone https://github.com/nightsky30/openrazer -b fix_blackwidow_v3_game_mode
cd openrazer
```

But I'm not sure what distribution you use. How you build and install can vary. You'll have to determine your own personal course of action for the remainder of the steps outlined [here](https://github.com/openrazer/openrazer/wiki/Building-a-package), and of course being sure to reboot after the build/install so that any changes take effect.

anyoneyun, 2022-08-27:
I'm using Manjaro, so I tried to follow the instructions for Arch. I'm a beginner here so I'm very sorry if I'm wasting your time and you don't have to answer to my questions obviously. I actually couldn't follow the instructions in the wiki. There is no PKGBUILD file so I tried cloning the main branch, editing the PKGBUILD file in the source field and put there your fork. No luck though, I installed everything but the error is still there. Thanks for your support and again, I'm very sorry if I'm wasting your time

nightsky30, 2022-08-27:
Nah, you're fine. Don't feel that way.

I'm not very familiar with Arch/Manjaro, and perhaps this is what you tried, but:

`git clone https://aur.archlinux.org/openrazer-git.git`

Then, modify that PKGBUILD file with:

`source=("git+https://github.com/nightsky30/openrazer#branch=fix_blackwidow_v3_game_mode")`

And in the PKGBUILD directory:

`makepkg -sC`

After the build and still in the PKGBUILD directory:

`pacman -U ./*openrazer*.pkg.tar.*`

Reboot

If those instructions aren't working, then I'm not sure what might be missing. Thanks for trying! :smile: Alternatively, if you wanted to test using another distro such as Ubuntu, you could do that with something like VirtualBox and a VM along with the Ubuntu build instructions modified for the repo and branch. Then you could enable USB passthrough for the keyboard with the VM. But that's a bit more work with the setup.

anyoneyun, 2022-08-27:
It works! Thank you so much! Will this be merged in the main branch?

nightsky30, 2022-08-27:
> It works! Thank you so much! Will this be merged in the main branch?

Nice! Thanks for testing it out. I can submit a PR, but can't answer about the merging. I'm just contributing to the cause. :smile:

anyoneyun, 2022-08-27:
Just checking, pacman is offering an update for openrazer. Seems that the AUR package linked in the wiki isn’t up to date, latest version being 3.0.1. Also, when generating the files with the PKGBUILD file, they appear to be version 2.5
Is this branch up to date or am I missing something? Thanks!

nightsky30, 2022-08-27:
My branch is up to date with 3.4.0. If you built with the source change in PKGBUILD, then I think you should be on the latest. A dead giveaway is if your BlackWidow V3 is recognized. If you had built or were running v3.0.1 or v2.5, it wouldn't list your keyboard in RazerGenie/Polychromatic.

anyoneyun, 2022-08-28:
Mhhh. PKGBUILD lists `pkgver=3.0.1.r8.g851c32a5` when I clone 
`https://aur.archlinux.org/openrazer-git.git`
After changing the source and running `makepkg -sC` it generates 2.5 version files:
`openrazer-daemon-git-2.5.0.r337.ged09e48-1-any.pkg.tar.zst`
`openrazer-driver-dkms-git-2.5.0.r337.ged09e48-1-any.pkg.tar.zst`
`openrazer-meta-git-2.5.0.r337.ged09e48-1-any.pkg.tar.zst`
`python-openrazer-git-2.5.0.r337.ged09e48-1-any.pkg.tar.zst`

Also, pkgver in PKGBUILD is changed to 2.5.0.r337.ged09e48. Pacman still shows an update available but the keyboard is recognized and the fix is working. Also, polychromatic shows Openrazer version 3.4.0.

nightsky30, 2022-08-28:
[ed09e48](https://github.com/nightsky30/openrazer/commit/ed09e485089af0a5a69935ba35d2ce62f9b6b1a0) which is the commit value listed in your built package names matches the latest commit in my branch which includes the fix, so I think you're good. I'd go by that and what Polychromatic says regarding v3.4.0. Disregard the 2.5.0. It must be pulling that from the upsidedown :+1:

PR #1899 (2022-10-16): Fix BlackWidow V3 (Full Size, Non-Pro) Game Mode Toggle
Fix BlackWidow V3 (Full Size, Non-Pro) Game Mode Toggle
- Fixes #1728 
- Tested by @anyoneyun - Thanks for testing! :smile:
closed 2022-10-16 by z3ntu