openrazer/openrazer #1729: :(

p371, 2022-01-08:
× openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2022-01-08 11:47:39 CET; 35s ago
       Docs: man:openrazer-daemon(8)
    Process: 3279 ExecStart=/usr/bin/openrazer-daemon -F (code=exited, status=1/FAILURE)
   Main PID: 3279 (code=exited, status=1/FAILURE)
        CPU: 48ms

jan 08 11:47:39 ITmechanic systemd[1123]: Starting Daemon to manage razer devices in userspace...
jan 08 11:47:39 ITmechanic openrazer-daemon[3279]: Traceback (most recent call last):
jan 08 11:47:39 ITmechanic openrazer-daemon[3279]:   File "/usr/bin/openrazer-daemon", line 11, in <module>
jan 08 11:47:39 ITmechanic openrazer-daemon[3279]:     from openrazer_daemon.daemon import RazerDaemon, __version__
jan 08 11:47:39 ITmechanic openrazer-daemon[3279]: ModuleNotFoundError: No module named 'openrazer_daemon'
jan 08 11:47:39 ITmechanic systemd[1123]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
jan 08 11:47:39 ITmechanic systemd[1123]: openrazer-daemon.service: Failed with result 'exit-code'.
jan 08 11:47:39 ITmechanic systemd[1123]: Failed to start Daemon to manage razer devices in userspace.

z3ntu, 2022-01-08:
Is this helpful? https://github.com/openrazer/openrazer/issues/1362

vicmaple, 2022-01-13:
I have the same issue here with RazerGenie. 
I installed both openrazer and razergenie with yay on Manjaro and tried following the installation steps carefully. 
I made sure that I'm correctly assigned to the plugdev group.
I am running Python 3.10. I think every python packages I have are up to date. (`pacman -Qo /usr/lib/python3.8/ `results with nothing.)

Doing `openrazer-daemon -Fv ` results :      
                          
> Traceback (most recent call last):
>   File "/usr/bin/openrazer-daemon", line 11, in <module>
>     from openrazer_daemon.daemon import RazerDaemon, __version__
> ModuleNotFoundError: No module named 'openrazer_daemon'

Looking into /usr/bin/openrazer-daemon, line 11 tries to import a module that doesn't exist at the location. (as far as I know. I'm not familiar with Python syntax. Is it trying to import a module from a folder?) 
`from openrazer_daemon.daemon import RazerDaemon, __version__`

Before posting this, I checked `pacman -Qo /usr/lib/python3.9/ ` , just to make sure, and got :
> /usr/lib/python3.9/ is owned by openrazer-daemon 3.2.0-1
> /usr/lib/python3.9/ is owned by python-openrazer 3.2.0-1
> [...]

vicmaple, 2022-01-13:
Ah! I found a solution. I updated the 3.9 packages to 3.10 with `pacman -Qoq /usr/lib/python3.9/ | yay -S --rebuild -`, but now the cursor won't move :P 

The daemon didn't detect my naga trinity mouse, yet doing `lsusb | grep 1532` reveals that my device is connected...

vicmaple, 2022-01-13:
Probably my final post : I fixed the mouse issue. it seems to be working just fine, and razergenie is detecting it. I forgot to install my kernel headers. Thought it was already installed. Oops!

You can check your kernel version with `mhwd-kernel` . Mine was Linux510, so i took note of it and installed the headers for it.  
( `sudo pacman -Syu linux510-headers` in my case.)

After that i "re-yay'ed" openrazer and razergenie. 

Hopefully it works for OP.

CloudyWhale, 2022-01-29:
O thank god

closed 2022-07-19 by github-actions[bot]