openrazer/openrazer #1730: Support for Laptop Stand Chroma V2

j-maynard, 2022-01-12:
### Device Info

* Name: Laptop Stand Chroma V2
* Product Number: <See under device>
* Model Number: RC21-0168

### Device Information
```
1439   │ Bus 003 Device 005: ID 1532:0f2b Razer USA, Ltd Razer Laptop Stand Chroma V2
1440   │ Device Descriptor:
1441   │   bLength                18
1442   │   bDescriptorType         1
1443   │   bcdUSB               2.00
1444   │   bDeviceClass            0 
1445   │   bDeviceSubClass         0 
1446   │   bDeviceProtocol         0 
1447   │   bMaxPacketSize0        64
1448   │   idVendor           0x1532 Razer USA, Ltd
1449   │   idProduct          0x0f2b 
1450   │   bcdDevice            2.00
1451   │   iManufacturer           1 Razer
1452   │   iProduct                2 Razer Laptop Stand Chroma V2
1453   │   iSerial                 0 
1454   │   bNumConfigurations      1
1455   │   Configuration Descriptor:
1456   │     bLength                 9
1457   │     bDescriptorType         2
1458   │     wTotalLength       0x0054
1459   │     bNumInterfaces          3
1460   │     bConfigurationValue     1
1461   │     iConfiguration          0 
1462   │     bmAttributes         0xa0
1463   │       (Bus Powered)
1464   │       Remote Wakeup
1465   │     MaxPower              500mA
1466   │     Interface Descriptor:
1467   │       bLength                 9
1468   │       bDescriptorType         4
1469   │       bInterfaceNumber        0
1470   │       bAlternateSetting       0
1471   │       bNumEndpoints           1
1472   │       bInterfaceClass         3 Human Interface Device
1473   │       bInterfaceSubClass      1 Boot Interface Subclass
1474   │       bInterfaceProtocol      2 Mouse
1475   │       iInterface              0 
1476   │         HID Device Descriptor:
1477   │           bLength                 9
1478   │           bDescriptorType        33
1479   │           bcdHID               1.11
1480   │           bCountryCode            0 Not supported
1481   │           bNumDescriptors         1
1482   │           bDescriptorType        34 Report
1483   │           wDescriptorLength      94
1484   │          Report Descriptors: 
1485   │            ** UNAVAILABLE **
1486   │       Endpoint Descriptor:
1487   │         bLength                 7
1488   │         bDescriptorType         5
1489   │         bEndpointAddress     0x81  EP 1 IN
1490   │         bmAttributes            3
1491   │           Transfer Type            Interrupt
1492   │           Synch Type               None
1493   │           Usage Type               Data
1494   │         wMaxPacketSize     0x0008  1x 8 bytes
1495   │         bInterval               1
1496   │     Interface Descriptor:
1497   │       bLength                 9
1498   │       bDescriptorType         4
1499   │       bInterfaceNumber        1
1500   │       bAlternateSetting       0
1501   │       bNumEndpoints           1
1502   │       bInterfaceClass         3 Human Interface Device
1503   │       bInterfaceSubClass      0
1504   │       bInterfaceProtocol      1 Keyboard
1505   │       iInterface              0 
1506   │         HID Device Descriptor:
1507   │           bLength                 9
1508   │           bDescriptorType        33
1509   │           bcdHID               1.11
1510   │           bCountryCode            0 Not supported
1511   │           bNumDescriptors         1
1512   │           bDescriptorType        34 Report
1513   │           wDescriptorLength     159
1514   │          Report Descriptors: 
1515   │            ** UNAVAILABLE **
1516   │       Endpoint Descriptor:
1517   │         bLength                 7
1518   │         bDescriptorType         5
1519   │         bEndpointAddress     0x82  EP 2 IN
1520   │         bmAttributes            3
1521   │           Transfer Type            Interrupt
1522   │           Synch Type               None
1523   │           Usage Type               Data
1524   │         wMaxPacketSize     0x0010  1x 16 bytes
1525   │         bInterval               1
1526   │     Interface Descriptor:
1527   │       bLength                 9
1528   │       bDescriptorType         4
1529   │       bInterfaceNumber        2
1530   │       bAlternateSetting       0
1531   │       bNumEndpoints           1
1532   │       bInterfaceClass         3 Human Interface Device
1533   │       bInterfaceSubClass      0 
1534   │       bInterfaceProtocol      1 Keyboard
1535   │       iInterface              0 
1536   │         HID Device Descriptor:
1537   │           bLength                 9
1538   │           bDescriptorType        33
1539   │           bcdHID               1.11
1540   │           bCountryCode            0 Not supported
1541   │           bNumDescriptors         1
1542   │           bDescriptorType        34 Report
1543   │           wDescriptorLength      61
1544   │          Report Descriptors: 
1545   │            ** UNAVAILABLE **
1546   │       Endpoint Descriptor:
1547   │         bLength                 7
1548   │         bDescriptorType         5
1549   │         bEndpointAddress     0x83  EP 3 IN
1550   │         bmAttributes            3
1551   │           Transfer Type            Interrupt
1552   │           Synch Type               None
1553   │           Usage Type               Data
1554   │         wMaxPacketSize     0x0008  1x 8 bytes
1555   │         bInterval               1
```

```
❯ ls -lah /dev/input/by-id/
drwxr-xr-x root root 440 B  Wed Jan 12 23:05:06 2022  .
drwxr-xr-x root root 1.1 KB Wed Jan 12 23:05:06 2022  ..
lrwxrwxrwx root root  10 B  Wed Jan 12 22:59:05 2022  usb-Azurewave_Integrated_RGB_Camera_01.00.00-event-if00 ⇒ ../event11
lrwxrwxrwx root root  10 B  Wed Jan 12 22:59:05 2022  usb-Azurewave_Integrated_RGB_Camera_01.00.00-event-if02 ⇒ ../event31
lrwxrwxrwx root root  11 B  Wed Jan 12 22:59:05 2022  usb-ELAN_Touchscreen-event-if00 ⇒ ../event265
lrwxrwxrwx root root  10 B  Wed Jan 12 22:59:04 2022  usb-Generic_USB_Audio-event-if03 ⇒ ../event28
lrwxrwxrwx root root   9 B  Wed Jan 12 22:59:05 2022  usb-Razer_Inc_Razer_Kiyo_Pro-event-if00 ⇒ ../event9
lrwxrwxrwx root root  10 B  Wed Jan 12 22:59:04 2022  usb-Razer_Razer_BlackWidow_V3_Pro_000000000000-event-if01 ⇒ ../event23
lrwxrwxrwx root root  10 B  Wed Jan 12 22:59:04 2022  usb-Razer_Razer_BlackWidow_V3_Pro_000000000000-event-if03 ⇒ ../event25
lrwxrwxrwx root root  10 B  Wed Jan 12 22:59:04 2022  usb-Razer_Razer_BlackWidow_V3_Pro_000000000000-event-kbd ⇒ ../event20
lrwxrwxrwx root root  10 B  Wed Jan 12 22:59:04 2022  usb-Razer_Razer_BlackWidow_V3_Pro_000000000000-if01-event-kbd ⇒ ../event21
lrwxrwxrwx root root  10 B  Wed Jan 12 22:59:04 2022  usb-Razer_Razer_BlackWidow_V3_Pro_000000000000-if01-event-mouse ⇒ ../event22
lrwxrwxrwx root root   9 B  Wed Jan 12 22:59:04 2022  usb-Razer_Razer_BlackWidow_V3_Pro_000000000000-if01-mouse ⇒ ../mouse5
lrwxrwxrwx root root  10 B  Wed Jan 12 22:59:04 2022  usb-Razer_Razer_BlackWidow_V3_Pro_000000000000-if02-event-mouse ⇒ ../event24
lrwxrwxrwx root root   9 B  Wed Jan 12 22:59:04 2022  usb-Razer_Razer_BlackWidow_V3_Pro_000000000000-if02-mouse ⇒ ../mouse6
lrwxrwxrwx root root   9 B  Wed Jan 12 22:59:05 2022  usb-Razer_Razer_Blade-event-kbd ⇒ ../event4
lrwxrwxrwx root root  10 B  Wed Jan 12 22:59:05 2022  usb-Razer_Razer_Blade-if01-event-kbd ⇒ ../event30
lrwxrwxrwx root root  11 B  Wed Jan 12 22:59:05 2022  usb-Razer_Razer_Blade-if02-event-mouse ⇒ ../event259
lrwxrwxrwx root root   9 B  Wed Jan 12 22:59:05 2022  usb-Razer_Razer_Blade-if02-mouse ⇒ ../mouse2
lrwxrwxrwx root root  11 B  Wed Jan 12 22:59:32 2022  usb-Razer_Razer_Kraken_Kitty_Chroma_Control-event-if00 ⇒ ../event268
lrwxrwxrwx root root  11 B  Wed Jan 12 22:59:32 2022  usb-Razer_Razer_Kraken_Kitty_Edition_00000000-event-if03 ⇒ ../event269
lrwxrwxrwx root root  10 B  Wed Jan 12 22:59:06 2022  usb-Yubico_YubiKey_OTP+FIDO+CCID-event-kbd ⇒ ../event26
```

### Packet Captures

I'll try to grab these when I have more time.

### Synapse Screenshots
[image]

ashvinnihalani, 2022-03-04:
Beginning to work on this:
Quick Effects: 6 Types
Advanced Type: Not sure, cant get this working in VM
Brightness Toggle: On/Off
Brightness Slider: 0 - 100
Switch Off Lighting: Toggle
Google Sheet where I am tracking Packet Analysis: https://docs.google.com/spreadsheets/d/1-6fZ5axHPpi5ufgy2Ka8N_gY4NQUaZL4cpm6APK3fyg/edit?usp=sharing

ashvinnihalani, 2022-03-04:
Initial Packet Headers:
Start: 0x00 
ID: 0x1f
Reserved: 0x000000
Num Params: 0x32 - Fifty Params this seems high
Reserved: 0x0f

ashvinnihalani, 2022-03-04:
Static Single Color - Full Brightness
Command Type: 0x03
Sub Command Byte: 0x00
Params - 0x0000000e(HEX Value of Color, 3 bytes, repeated 15 times)
[Full Brightness Single Color.zip](https://github.com/openrazer/openrazer/files/8183610/Full.Brightness.Single.Color.zip)

ashvinnihalani, 2022-03-04:
Changing Brightness: (Seems to be a Single Command)
Command Type: 0x04
Sub Command Type: 0x01
Params: 0x00ff - 75%, 0x00bf 50%, 0x003f - 25%, 0x00 - 0%.
Not sure on the params mapping. 
Turning the brightness to zero seems to be the same as turning off the lighting
[Toggling Brightness Static F0A599.zip](https://github.com/openrazer/openrazer/files/8183919/Toggling.Brightness.Static.F0A599.zip)

ashvinnihalani, 2022-03-05:
Breathing Effect:
Same Command as Static, a gradual decrease from initial color to black and increase again. 
[Breathing Red to Green.zip](https://github.com/openrazer/openrazer/files/8191293/Breathing.Red.to.Green.zip)

ashvinnihalani, 2022-03-05:
Fire Effect:
Same Command as Static but with different values of RGB in the params. Looks like each of the 15 RGB values refers to an LED unit in the stand.
[Fire Breathing Effects.zip](https://github.com/openrazer/openrazer/files/8191302/Fire.Breathing.Effects.zip)

ashvinnihalani, 2022-03-05:
Other effects seem like the same use the base rgb command with different values for each led. So basically there are two commands: one to set the RGB color on a per led basis and one for the LED brightness of the entire unit. 
[Other Effects.zip](https://github.com/openrazer/openrazer/files/8191320/Other.Effects.zip)

mafarinacci, 2022-05-06:
you guys need any help with this? I have this device and would like to add support.

ashvinnihalani, 2022-05-15:
Sure I haven't had time to do this

Chr1sNo, 2022-06-17:
Ok, as the commit message suggests I have just pushed code to add this laptop stand. Sadly I have no way to test if the code is working as I do not own the stand myself. Came across this because I was adding the stand [to OpenRGB](https://gitlab.com/CalcProgrammer1/OpenRGB/-/merge_requests/1265) and full disclosure I am not familiar with the Openrazer code base.

FWIW: the OpenRGB code has been confirm as working on Windows so am wondering if anyone game to check that my Openrazer test branch compiles and can control their Laptop Stand?

z3ntu, 2022-06-18:
@Chr1sNo Can you open a PR regardless so it's tracked there and more visible?

You could probably also launch a Linux VM on Windows and do USB passthrough to it to test. Otherwise someone else will have to test it on Linux with the device :)

Chr1sNo, 2022-06-18:
> Can you open a PR regardless so it's tracked there and more visible?

Can do.

> You could probably also launch a Linux VM on Windows and do USB passthrough to it to test.

Apologies, can't do that, I don't have the Stand.

RafaelZasas, 2022-08-08:
I can test this if need be- I have the Stand V2 and a Blade pro (2022) running Ubuntu 22.04 with kernel 5.19. I just need some instructions as to how I can get the driver on my machine from [PR](https://github.com/openrazer/openrazer/pull/1840)

z3ntu, 2023-06-19:
From the community chat:

```
usbhid-dump -m 1532:0f2b -ed
001:008:002:DESCRIPTOR         1687192536.708425
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

001:008:001:DESCRIPTOR         1687192536.709025
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 C0 05 0C 09 01 A1 01 85
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

001:008:000:DESCRIPTOR         1687192536.709669
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0
```

closed 2023-06-19 by z3ntu