openrazer/openrazer #1732: BlackWidow V3 Pro, media buttons needs support

mariasoderstrom, 2022-01-15:
### Distribution
Ubuntu Linux
Ubuntu 21.10 64-Bit
Gnome 40.4.0

### Support
BlackWidow V3 Pro's media buttons are not working after installing polychromatic.

[image]

lah7, 2022-01-15:
Similar to #1552.

An identical model (`024E`) had the same problem: https://github.com/openrazer/openrazer/pull/1572#issuecomment-893598938 but I'm guessing it was only fixed for that specific PID (yours is `025A`).

For now, it should be possible to workaround the issue by stopping the daemon, putting the keyboard back into "generic" mode:

    openrazer-daemon -s

As soon as the daemon runs again (which can be triggered by opening any frontend), the keyboard's back into "driver" mode where the bug resides in the driver's key handling.

GeoDoX, 2022-01-16:
I’ll fix this this afternoon! My apologies for missing it in the pr.

hlriffel, 2022-02-08:
Hello,

Is there any update on this? I'm also having this media keys issue with the same device (`025A`).

samoht9277, 2022-02-10:
Same, I'm still having the same issue on `025A`. Media keys are not working.

wyattearp, 2022-02-12:
I just noticed this issue as well. Looks like the fix *might* be as easy as:

```diff
diff --git a/driver/razerkbd_driver.c b/driver/razerkbd_driver.c
index e1eb551..578871d 100644
--- a/driver/razerkbd_driver.c
+++ b/driver/razerkbd_driver.c
@@ -2614,6 +2614,7 @@ static int razer_event(struct hid_device *hdev, struct hid_field *field, struct
         break;
 
     case USB_DEVICE_ID_RAZER_BLACKWIDOW_V3:
+    case USB_DEVICE_ID_RAZER_BLACKWIDOW_V3_PRO_WIRED:
         translation = find_translation(chroma_keys_5, usage->code);
         break;
```

@GeoDoX is that your understanding as well?

wyattearp, 2022-02-15:
Could anyone test my PR to see if it works for you? It "works for me" but I think there should be a little bit more testing than that :)

My current system info:

OS Details:
```
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal
```

Keyboard firmware version: v2.2

[image]

hlriffel, 2022-02-15:
@wyattearp 

I'm using the changes from your PR and it seems to be working perfectly initially. I'll be using it for a few days to make sure.

Thanks!

wyattearp, 2022-02-17:
Anyone else try this?

GeoDoX, 2022-02-17:
@wyattearp That wasn’t what I was going to do, but if that seems to fix it, go ahead with it.

I was just going to reset the “mode” back to device mode as referenced in other issues and PRs related to media keys not working.

I appreciate someone tackling this as I just haven’t had the time to get to it.

samoht9277, 2022-02-18:
Can confirm that @wyattearp PR is in fact working properly. Media buttons now work as intended, thanks!

    Operating System: Arch Linux
    Kernel: Linux 5.16.10-arch1-1
    Architecture: x86-64

PR #1751 (2022-02-19): fix #1732 - add support for media keys
this should move just the needed items for the blackwidow v3 pro to
support the media keys and fix #1732. it works for me, but it could use some other
eyes to make sure i didn't miss something else.

fix refs #1552 & #1572 for details on tracking down the spots to
address.
closed 2022-02-19 by z3ntu