openrazer/openrazer #1733: Daemon not responding - troubleshooting steps followed

marmenta13, 2022-01-15:
I've followed the troubleshooting steps outlined https://github.com/openrazer/openrazer/wiki/Troubleshooting, but still can't get the daemon to start.

System info:
OS: Linux Mint 20.3
Kernel: 5.4.0-94
Machine: Blade 15 Base Model (Early 2020)

Output of openrazer-daemon -Fv
Starting daemon.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/daemonize.py", line 245, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 140, in run_daemon
    daemon = RazerDaemon(verbose=args.verbose,
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 83, in __init__
    self.read_persistence(persistence_file)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 303, in read_persistence
    self._persistence.read(persistence_file)
  File "/usr/lib/python3.8/configparser.py", line 697, in read
    self._read(fp, filename)
  File "/usr/lib/python3.8/configparser.py", line 1017, in _read
    for lineno, line in enumerate(fp, start=1):
  File "/usr/lib/python3.8/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8d in position 1: invalid start byte

Stopping daemon.

I've attached the daemon and razer_mount logs.

Any help is greatly appreciated.
[razer_mount.log](https://github.com/openrazer/openrazer/files/7875149/razer_mount.log)
[razer.log](https://github.com/openrazer/openrazer/files/7875150/razer.log)

marmenta13, 2022-01-15:
Forgot to mention, running openrazer 3.2.0 stable and polychromatic 0.7.3 stable

z3ntu, 2022-01-15:
Seems that your `~/.config/openrazer/persistence.conf` is a bit corrupted. Could you upload the file here for investigation, and then you can remove it from your computer.

marmenta13, 2022-01-15:
[persistence.conf.gz](https://github.com/openrazer/openrazer/files/7875191/persistence.conf.gz)

Uploading file for review and testing. thanks for referencing the other issue too.

marmenta13, 2022-01-15:
That fixed the issue, I appreciate the quick support.

closed 2022-01-15 by marmenta13

z3ntu, 2022-01-15:
If this happens again please inform us. Great it works for you!

commit 43423b7b (2022-03-26): daemon: Fix crash when persistence.conf is corrupt

In some circumstances - presumably on system shutdown - the file can get
corrupted. Deleting it and starting from scratch is still better than
crashing.

Fixes #1652
Fixes #1733
Fixes #1777
commit 74b5f642 (2022-03-26): daemon: Fix crash when persistence.conf is corrupt

In some circumstances - presumably on system shutdown - the file can get
corrupted. Deleting it and starting from scratch is still better than
crashing.

Fixes #1652
Fixes #1733
Fixes #1777