openrazer/openrazer #1735: Compiling OpenRazer kernel modules on Nvidia Jetson Nano Ubuntu 18.04

sh1sh1n11, 2022-01-16:
### Distribution

`NAME="Ubuntu"
VERSION="18.04.6 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.6 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic`

`Linux ubuntu 4.9.253-tegra #1 SMP PREEMPT Sun Jan 16 03:46:29 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux`

I am using Nvidia Jetson Nano B01 4GB soc amd64 (aarch64) board. Also, I currently run the 4.9.253-tegra kernel.

When I try to install the 'openrazer-driver-dkms' package it gives me the following output.

`sudo apt-get install openrazer-driver-dkms
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  dkms linux-headers-4.15.0-166 linux-headers-4.15.0-166-generic linux-headers-generic
Suggested packages:
  menu
The following NEW packages will be installed:
  dkms linux-headers-4.15.0-166 linux-headers-4.15.0-166-generic linux-headers-generic openrazer-driver-dkms
0 upgraded, 5 newly installed, 0 to remove and 0 not upgraded.
Need to get 12.0 MB of archives.
After this operation, 86.4 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 dkms all 2.3-3ubuntu9.7 [68.1 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 linux-headers-4.15.0-166 all 4.15.0-166.174 [10.9 
MB]
Get:3 http://ppa.launchpad.net/openrazer/stable/ubuntu bionic/main arm64 openrazer-driver-dkms all 3.2.0~ubuntu18.04.1 
[59.9 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 linux-headers-4.15.0-166-generic arm64 4.15.0-166.
174 [882 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 linux-headers-generic arm64 4.15.0.166.155 [2,464 
B]
Fetched 12.0 MB in 4s (2,799 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package dkms.
(Reading database ... 128311 files and directories currently installed.)
Preparing to unpack .../dkms_2.3-3ubuntu9.7_all.deb ...
Unpacking dkms (2.3-3ubuntu9.7) ...
Selecting previously unselected package linux-headers-4.15.0-166.
Preparing to unpack .../linux-headers-4.15.0-166_4.15.0-166.174_all.deb ...
Unpacking linux-headers-4.15.0-166 (4.15.0-166.174) ...
Selecting previously unselected package linux-headers-4.15.0-166-generic.
Preparing to unpack .../linux-headers-4.15.0-166-generic_4.15.0-166.174_arm64.deb ...
Unpacking linux-headers-4.15.0-166-generic (4.15.0-166.174) ...
Selecting previously unselected package linux-headers-generic.
Preparing to unpack .../linux-headers-generic_4.15.0.166.155_arm64.deb ...
Unpacking linux-headers-generic (4.15.0.166.155) ...
Selecting previously unselected package openrazer-driver-dkms.
Preparing to unpack .../openrazer-driver-dkms_3.2.0~ubuntu18.04.1_all.deb ...
Unpacking openrazer-driver-dkms (3.2.0~ubuntu18.04.1) ...
Setting up linux-headers-4.15.0-166 (4.15.0-166.174) ...
Setting up dkms (2.3-3ubuntu9.7) ...
Setting up linux-headers-4.15.0-166-generic (4.15.0-166.174) ...
/etc/kernel/header_postinst.d/dkms:
 * dkms: running auto installation service for kernel 4.15.0-166-generic
   ...done.
Setting up linux-headers-generic (4.15.0.166.155) ...
Setting up openrazer-driver-dkms (3.2.0~ubuntu18.04.1) ...
[... 266 lines cut ...]
[ 5341.512130] Could not change i2c_ddc sclk rate
[ 5341.516612] tegradc tegradc.0: hdmi: scdc scrambling status is reset, trying to reconfigure.
[ 5346.614865] tegra_i2c_handle_xfer_error: 2 callbacks suppressed
[ 5346.614871] tegra-i2c 7000c700.i2c: no acknowledge from address 0x54
[ 5346.627177] i2c i2c-3: clk rate 50000 not supported
[ 5346.632064] Could not change i2c_ddc sclk rate
[ 5346.636540] tegradc tegradc.0: hdmi: scdc scrambling status is reset, trying to reconfigure.
[ 5351.735806] tegra_i2c_handle_xfer_error: 2 callbacks suppressed
[ 5351.735812] tegra-i2c 7000c700.i2c: no acknowledge from address 0x54
[ 5351.748150] i2c i2c-3: clk rate 50000 not supported
[ 5351.753047] Could not change i2c_ddc sclk rate
[ 5351.757531] tegradc tegradc.0: hdmi: scdc scrambling status is reset, trying to reconfigure.
[ 5356.858778] tegra_i2c_handle_xfer_error: 2 callbacks suppressed
[ 5356.858784] tegra-i2c 7000c700.i2c: no acknowledge from address 0x54
[ 5356.871152] i2c i2c-3: clk rate 50000 not supported
[ 5356.876042] Could not change i2c_ddc sclk rate
[ 5356.880517] tegradc tegradc.0: hdmi: scdc scrambling status is reset, trying to reconfigure.
[ 5361.974729] tegra_i2c_handle_xfer_error: 2 callbacks suppressed
[ 5361.974735] tegra-i2c 7000c700.i2c: no acknowledge from address 0x54
[ 5361.987049] i2c i2c-3: clk rate 50000 not supported
[ 5361.991938] Could not change i2c_ddc sclk rate
[ 5361.996414] tegradc tegradc.0: hdmi: scdc scrambling status is reset, trying to reconfigure.
[ 5367.094659] tegra_i2c_handle_xfer_error: 2 callbacks suppressed
[ 5367.094665] tegra-i2c 7000c700.i2c: no acknowledge from address 0x54
`

sh1sh1n11, 2022-01-16:
Any help in resolving this issue would be awesome! Thank you :)

lah7, 2022-01-16:
As a guess, the 4.9 kernel is not compatible with OpenRazer's 3.2.0 codebase, so it cannot be compiled.

[It looks like this device uses a custom kernel.](https://forums.developer.nvidia.com/t/upgrading-kernel-to-5/112113) `4.9.253-tegra` that's running doesn't seem to use a generic kernel like `linux-headers-4.15.0-166-generic` which is what it installed but probably can't use.

Some suggestions:

* Check online and the repositories (e.g. with Synaptic) to see if there is a newer [4.16+](https://github.com/openrazer/openrazer/issues/1519#issuecomment-848725182) or 5.x kernel for your hardware.
* [Try an older version of OpenRazer](https://github.com/openrazer/openrazer/releases)
   * In the `.tar.xz` download, there is a script `./scripts/build_debs.sh` you can use to build packages.
   *  3.0.0 is the minimum that supports both your devices.

sh1sh1n11, 2022-01-16:
Thank you so much @lah7 for such a quick response! Means a lot :) And I hear what you're saying about the kernel being custom.

Here's part of the output of 'dpkg-query -l | grep -i headers'.

```
nvidia-l4t-kernel-headers            4.9.253-tegra-32.6.1-20210726122000  arm64        NVIDIA Linux Tegra Kernel Headers Package
```
According to Nvidia's documentation for Jetson Nano, under the [Preparing to Build External Kernel Modules](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/kernel_custom.html#wwpID0E0LB0HA) section, it is specified that the following kernel directory should be used in order to build external kernel modules.

```
/usr/src/linux-headers-$(uname -r)-ubuntu18.04_aarch64
or
/lib/modules/$(uname -r)/build

I checked, and the above mentioned 4.9.253-tegra kernel indeed sits in the below directory.

/usr/src/linux-headers-4.9.253-tegra-ubuntu18.04_aarch64/kernel-4.9
```
I tried following Nvidia's instructions on [Building External Kernel Modules](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/kernel_custom.html#wwpID0E0LB0HA), and ran similar commands as so. But got the same result as the mentioned above in the make.log output.
```
$ cd <path_to_module_source>
$ make ARCH=arm64 –C <kernel_directory> M=$(pwd)
```
As this is the first time I am building kernel modules, I'm really not sure about my approach. Based on the above mentioned 'make.log' that openrazer generated at the time of installation. I found that the openrazer script dropped four files as below.
```
pwd
/lib/modules/4.15.0-166-generic/updates/dkms
ls -ahl
total 420K
drwxr-xr-x 2 root root 4.0K Jan 16 10:00 .
drwxr-xr-x 3 root root 4.0K Jan 16 10:00 ..
-rw-r--r-- 1 root root  87K Jan 16 10:00 razeraccessory.ko
-rw-r--r-- 1 root root 122K Jan 16 10:00 razerkbd.ko
-rw-r--r-- 1 root root  29K Jan 16 10:00 razerkraken.ko
-rw-r--r-- 1 root root 160K Jan 16 10:00 razermouse.ko
```
Can these files be somehow used to build the kernel modules against the 4.9.253-tegra kernel?

Sadly, I don't think I can make use of any other kernel with this device. But I may be wrong.

Also, are you suggesting that since Nvidia uses a quite old (4.9.253-tegra) version of the linux kernel, it is not supported by the new openrazer-driver-dkms (and other relevant openrazer tools)?

If so, I can give the old versions thing a try also. Please let me know what you think. And thanks much for your help once again :)

z3ntu, 2022-01-16:
You can try reverting commit https://github.com/openrazer/openrazer/commit/dbd0d90aa90bb8a8ab905cfe6d033270cc4cef29 or just disable compilation of the razeraccessory module as that doesn't seem to be compatible with your old kernel.

sh1sh1n11, 2022-01-16:
Thank you @z3ntu for the detailed analysis on this. Much appreciated! :) And how would I go about doing either of those? Reverting the commit or disabling the compilation of the razeraccessory module altogether. I understand if it's something very basic and probably not the right place to ask. Maybe you could just give me a couple of pointers to get started?

z3ntu, 2022-01-16:
You can either manually edit/patch the installed source files in /usr/src/  (e.g. remove razeraccessory from this file https://github.com/openrazer/openrazer/blob/master/driver/Makefile) or patch the razeraccessory_driver.c file there.

Or clone this repo, revert the mentioned commit and follow https://github.com/openrazer/openrazer/wiki/Building-a-package#ubuntu--linux-mint

sh1sh1n11, 2022-01-16:
Thank much @z3ntu, I will give that a try! And feel free to close this issue. Although, I wish that Nvidia Jetson and other such soc devices in the Nvidia lineup are at some point added to the supported list for OpenRazer. Thanks again, and have a great weekend! :)

closed 2022-01-16 by z3ntu

sh1sh1n11, 2022-04-18:
Thanks much @lah7 and @z3ntu. And apologies for the late response. I went ahead and manually patched the file in question. Then ran the compilation once again, and this time everything went through successfully, thus generating the binaries. Everything now runs perfectly! Thank you!