openrazer/openrazer #1736: OpenRazer stopped working with recent update

Merung-Kelsith, 2022-01-18:
### Distribution

Kubuntu 20.04 (fully updated)
OpenRazer Daily PPA (fully updated)
Polychromatic Testing PPA (fully updated)

### Support

- [ ] Have you tried rebooting? Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? Yes

Up until about a week ago, my OpenRazer/Polychromatic installation was working with no issues.  I have multiple Razer devices, and all (except Nari Ultimate and Kiyo) were working flawlessly.  Now, I am getting "Unrecognized Device" for all of them.

I ran the "OpenRazer Troubleshooting" command through Polychromatic, and got the following:
1) Check Secure Boot (EFI) Status - Failed: Secure Boot is enabled, but this has not been an issue before.  Also, following the troubleshooting page, "sudo modprobe razerkbd" returns no output, which from what I am reading means the module is successfully loaded.
2) Device can be accessed using plugdev permissions (Failed):  My user is in the plugdev group, and "/usr/lib/udev/rules.d/99-razer.rules" exists (and from what I can tell looks correct).  It is owned by root:root with permissions of 644.  But it does not look like it is being run.  The results of my "cat ~/.local/share/openrazer/logs/razer.log" is below:

`2022-01-18 08:08:21 | razer                          | INFO     | Initialising Daemon (v3.2.0). Pid: 4494
2022-01-18 08:08:21 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-01-18 08:08:21 | razer                          | INFO     | Found device.0: 0003:1532:0085.0005
2022-01-18 08:08:21 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:08.1/0000:07:00.3/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/0003:1532:0085.0005/device_type, file is not owned by plugdev
2022-01-18 08:08:21 | razer                          | INFO     | Found device.0: 0003:1532:024E.000A
2022-01-18 08:08:21 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:08.1/0000:07:00.3/usb1/1-1/1-1.1/1-1.1.4/1-1.1.4:1.2/0003:1532:024E.000A/device_type, file is not owned by plugdev
2022-01-18 08:08:21 | razer                          | INFO     | Found device.0: 0003:1532:0C04.0003
2022-01-18 08:08:21 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:08.1/0000:07:00.3/usb1/1-1/1-1.2/1-1.2:1.0/0003:1532:0C04.0003/device_type, file is not owned by plugdev
2022-01-18 08:08:21 | razer                          | INFO     | Found device.0: 0003:1532:0F08.000E
2022-01-18 08:08:21 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:08.1/0000:07:00.3/usb1/1-1/1-1.4/1-1.4.4/1-1.4.4:1.0/0003:1532:0F08.000E/device_type, file is not owned by plugdev
2022-01-18 08:08:21 | razer                          | INFO     | Serving DBus`

The devices are owned by root:root still.

Any help would be very much appreciated!

z3ntu, 2022-01-18:
Known issue, you can unplug your device and back in and it should hopefully work. The udev rules have some kind of race condition I believe but never been able to track down what's exactly happening

Merung-Kelsith, 2022-01-18:
Do I have to re-plug them in individually?  I have them all hanging off the same USB hub, and have tried unplugging / replugging that in...

Merung-Kelsith, 2022-01-18:
I just tried removing all of them, and plugging them in individually (restarting OpenRazer and Polychromatic after each device).  Still getting the same error.

Merung-Kelsith, 2022-01-18:
One other thing I notice.  In the udev rules, there is "99-systemd.rules" after "99-razer.rules".  This file has the following lines that *might* affect the Razer products, but I am not sure:

`# Pull in backlight save/restore for all backlight devices and
# keyboard backlights
SUBSYSTEM=="backlight", TAG+="systemd", IMPORT{builtin}="path_id", ENV{SYSTEMD_WANTS}+="systemd-backlight@backlight:$name.service"
SUBSYSTEM=="leds", KERNEL=="*kbd_backlight", TAG+="systemd", IMPORT{builtin}="path_id", ENV{SYSTEMD_WANTS}+="systemd-backlight@leds:$name.service"`

z3ntu, 2022-01-18:
Nah we don't use the linux backlight/led subsystem so this shouldn't be an issue

Merung-Kelsith, 2022-01-18:
Hmmm.  Was worth a shot :)

Tried the same steps plugging the USB hub into a different USB port on my laptop - same issue.

Tried plugging a single device directly into the port on my laptop (no USB hub, USB hub completely disconnected) - same issue.

billf8004, 2022-03-17:
I'm not sure if this is the same issue, but I've been having issues since a recent update as well.  Openrazer-daemon -Fv shows this:
`Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/daemonize.py", line 245, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 121, in run_daemon
    daemon = RazerDaemon(verbose=args.verbose,
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 107, in __init__
    self._load_devices(first_run=True)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 409, in _load_devices
    razer_device = device_class(sys_path, device_number, self._config, testing=self._test_dir is not None, additional_interfaces=sorted(additional_interfaces))
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 489, in __init__
    super().__init__(device_path, device_number, config, testing, additional_interfaces, additional_methods)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 452, in __init__
    super().__init__(device_path, device_number, config, testing, additional_interfaces, additional_methods)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 84, in __init__
    DBusService.__init__(self, self.BUS_PATH, object_path)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/service.py", line 87, in __init__
    super(DBusService, self).__init__(bus_object, object_path)
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 459, in __init__
    validate_object_path(object_path)
ValueError: Invalid object path '/org/razer/device/': contains invalid character ''

Stopping daemon.
`

z3ntu, 2022-03-17:
@billf8004 Does this happen every time or just sometimes? Also assuming you only have one Razer device, please run this and post the output `hexdump -C /sys/bus/hid/drivers/razer*/0*/device_serial` (not sure how this behaves if you have multiple devices connected)

z3ntu, 2022-03-26:
Yeah, that doesn't look like a correct serial number... It should be ASCII and not like these bytes

Did you do anything special to that device since it last worked? Firmware update in Windows or something?

billf8004, 2022-03-28:
Nothing at all. Not even sure if you can on a linux box.

billf8004, 2022-03-28:
FYI, I just installed updates and rebooted (nothing related to the devices, python, or openrazer) and ran the hexdump again and got a different result:

$ sudo hexdump -C /sys/bus/hid/drivers/razer*/0*/device_serial
00000000  01 0a 0a 01 10 8c 0a                              |.......|
00000007

z3ntu, 2022-03-28:
Can you check dmesg if there's any message from razermouse, razerkbd etc? I don't really understand why we're getting garbage data from the device...

closed 2022-08-26 by github-actions[bot]