openrazer/openrazer #1738: Orochi v2 not being detected?

Gaped, 2022-01-29:
### Distribution
Arch Linux

openrazer-meta-basilisk-v2-git, via Paru, I have tried the other two options it gives me as well, that being openrazer-meta and openrazer-meta-git.

### Support

- [ ] Have you tried rebooting? Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? Yes, also.

Polychromatic claims that my orochi v2 is not supported, but I thought it was for a good bit? my first time trying openrazer so I may be doing something incorrectly.. It claims it is in fact unsupported on the stable release but says I have the latest version..

lah7, 2022-01-29:
What does it say when you click "Troubleshoot"? A few common culprits are not being in the `plugdev` group or not having the kernel headers package installed before installing the driver (`linux-headers` or `linux-lts-headers`).

Make sure that `openrazer-meta` is the one that's installed, which should include these packages: `openrazer-daemon openrazer-driver-dkms python-openrazer`

Gaped, 2022-01-29:
It claims that the pid 1532:0094 is not supported in the stable release.. I have an idea. I uninstalled the meta package and replaced with the git meta package but maybe i need to remove those pkgs you mentioned explicitly; ill let you know how that works

Gaped, 2022-01-29:
[image]
It seems this package includes the stable versions of those packages you mentioned. hm.

Gaped, 2022-01-29:
> [image] It seems this package includes the stable versions of those packages you mentioned. hm.

Yup! that was the issue. I explicitly installed all 3 git versions of the packages, now my mouse is detected :) I suggest you guys check out openrazer-meta-git because it was giving me the stable packages, or maybeee im dumb. ty for your time!

lah7, 2022-01-29:
Glad it's working!  👍🏼  

The behaviour of [meta packages](https://wiki.archlinux.org/title/Meta_package_and_package_group) might vary by AUR helper. Installing the `-git` package should have prompted the removal of the non-git ones, so explicitly installing them is a good idea.

closed 2022-01-29 by lah7