openrazer/openrazer #1741: [Lancehead wireless] Changing polling rate or installing causes screentearing and vsync lag

SpxterWohlf, 2022-02-01:
### Distribution

Now: Kubuntu 21.10
& previously
Zorin OS 16 
& Linux mint 21

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

### Details with my Lancehead wireless
Im seeing this issue with any video game but very badly in minecraft, currently, installing openrazer and rebooting causes massive input lag and visual screen tearing. The entire monitor freezes up for about the amount of time of the polling rate. And thats after installing just the dameon, installing a gui like polychromatic makes it 10x worse maybe doubles the issue (in any adjustment). I orginally had this issue when I used Linux mint and then switched to zorin, however after switching to zorin and then installing open razer again, the issue came back. Ive since switched to kubuntu and have been fine for about a month but just installed the software last week and now video games and sometimes even youtube are just unusable. This really sucks because my windows loving friends make fun of my linux issues and I cant solve them :cry:.

I have no idea how to reproduce it and I dont have a powerful enough laptop to screenrecord the issue at the same time.

### Specs
- Using the x11 kde not wayland if that helps, also happens on just the laptop screen too without a external monitor plugged in.
- Ive tried to use it plugged in as well as with the receiver 
- My cpu and integrated gpu:
 - CPU: Intel i5-5300U (4) @ 2.900GHz 
 - GPU: Intel HD Graphics 5500

lah7, 2022-02-01:
Please post the output of `dmesg`, which will help see if the system is reporting any issues:

    sudo dmesg > ~/Desktop/dmesg.log

Some suggestions:

* Set the poll rate to the lowest. Stuttering can happen if your laptop's CPU is bottlenecked. The higher the polling rate, the more work the CPU needs to do.

* Try closing the background processes related to OpenRazer to rule out if the daemon is the problem, or the driver:

      killall openrazer-daemon
      killall polychromatic-tray-applet

If there's no stuttering when the daemon (and related GUI) isn't running, it might be a quirk specific to the Lancehead mouse. Is there a long delay when you change a setting (e.g. set brightness)?

SpxterWohlf, 2022-02-02:
I do have issues when changing the brightness or colors sometimes they update on time or not at all, however my firefly v2 mat updates instantly
Here is my dmesg:
[dmesg.log](https://github.com/openrazer/openrazer/files/7982665/dmesg.log)

After closing both services the issue seemed to go away. 
I reopened just openrazer and the issue came back, and it has to do with interaction with the mouse, as using the touchpad is just buttery smooth

lah7, 2022-02-02:
Looks like the mouse is suffering time out issues - that would explain the stutter:
```
kern  :warn  : [Feb 1 16:41] razermouse: Command timed out. Start Marker: 04 id: 1f Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
kern  :warn  : [Feb 1 16:51] razermouse: Command timed out. Start Marker: 04 id: 1f Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
```

For now, you can disable the OpenRazer daemon from starting and only use the software when you need to change a setting:

    systemctl --user disable openrazer-daemon

In addition, you'll need to disable Polychromatic's tray applet as the daemon is configured to automatically start when an application tries to access it.

SpxterWohlf, 2022-02-02:
Oh awesome. Im not to worried about the colors anyways. Normally the reset when i turn off open razer so no big deal. But thank you!

SpxterWohlf, 2022-02-07:
Is there anything I can do further to help fix this issue?

closed 2022-07-19 by github-actions[bot]