openrazer/openrazer #1742: razer basalisk v3 not popping up

Wolfy2107, 2022-02-04:
the software does not seem to want to pick up my razer basalisk v3, i am zorrin, are there any extras i need to install or anyway i could fix this

lah7, 2022-02-04:
[Razer Basilisk V3](https://github.com/openrazer/openrazer/commit/7b39a54f0aacd2355b8b75a88e4ab2851fa45ef7) was added fairly recently, so it's not in any stable release yet. You may need to upgrade your packages to the "daily" repository instead, and reboot:

    sudo add-apt-repository ppa:openrazer/daily
    sudo apt dist-upgrade

If that doesn't work, we need details of what you've tried, and what software you're using to interface with (Polychromatic, RazerGenie)? Did you see the [troubleshooting guide](https://github.com/openrazer/openrazer/wiki/Troubleshooting) for common problems?

NickTheNeko2002, 2022-02-05:
> 

what about on Manjaro/Arch??
apt is only for Debian based Distros as far as i know

lah7, 2022-02-05:
That's right, the command was for the original poster who presumably is running Zorin OS (Ubuntu-based).

Manjaro/Arch users can install the `-git` versions of the packages from the AUR:

    yay -S openrazer-meta-git openrazer-daemon-git openrazer-driver-dkms-git python-openrazer-git

(If necessary, replace `yay` with your chosen [AUR helper](https://wiki.archlinux.org/title/AUR_helpers))

Wolfy2107, 2022-02-05:
it still wont show up, i am using razergenie, i have check a few trouble shoots, but not all,

NickTheNeko2002, 2022-02-05:
> it still wont show up, i am using razergenie, i have check a few trouble shoots, but not all, it says could be due to misconfig of the pc,t says the open razer daemon didnt detect a device even though it plugged in and working.

first try with RazerCommander if it doesn't appear there you might need to uninstall OpenRazer and try re-installing it but this time the Git version, also it works for me (with RazerGenie V3.2.0)

Wolfy2107, 2022-02-05:
> first try with RazerCommander if it doesn't appear there you might need to uninstall OpenRazer and try re-installing it but this time the Git version, also it works for me (with RazerGenie V3.2.0)

i have 3.2.o, i tried commander says i need flathub when i already have that

NickTheNeko2002, 2022-02-05:
> i have 3.2.o, i tried commander says i need flathub when i already have that

i mean you can try compiling it from source code but that's a bit complicated :confused:

lah7, 2022-02-05:
Let's check which OpenRazer build is installed. Have you upgraded it from [`ppa:openrazer/daily` as described above?](https://github.com/openrazer/openrazer/issues/1742#issuecomment-1030178243)

Wolfy2107, 2022-02-05:
> Let's check which OpenRazer build is installed. Have you upgraded to use [`ppa:openrazer/daily` as described above?](https://github.com/openrazer/openrazer/issues/1742#issuecomment-1030178243)

I updated everything and seems to be working now, thanks guy

NickTheNeko2002, 2022-02-05:
> first try with RazerCommander if it doesn't appear there you might need to uninstall OpenRazer and try re-installing it but this time the Git version, also it works for me (with RazerGenie V3.2.0)

i told them to try Re-installing but i guess they skipped that part

closed 2022-02-05 by lah7

NickTheNeko2002, 2022-03-21:
> That's right, the command was for the original poster who presumably is running Zorin OS (Ubuntu-based).
>
> Manjaro/Arch users can install the `-git` versions of the packages from the AUR:
>
> ```
> yay -S [...]

Hey i'm getting this error when i try to install it with this

**```
(3/3) Install DKMS modules
==> ERROR: Missing mnt kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing bin kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing root kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing run kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing usr kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing home kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing srv kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing boot kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing tmp kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing version kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing var kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing sbin kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing sys kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing lost+found kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing lib64 kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing etc kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing proc kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing dev kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing lib kernel headers for module openrazer-driver/3.2.0.
==> ERROR: Missing opt kernel headers for module openrazer-driver/3.2.0.
```**

lah7, 2022-03-21:
@NickTheNeko2002   Make sure your system is up-to-date and has the correct Linux headers package installed. 
For the 5.16 kernel, it should be `linux-headers` (on Arch), or `linux516-headers` (on Manjaro). Then try to get DKMS to build again.

    sudo pacman -Syu
    sudo pacman -S linux-headers
    sudo dkms install openrazer-driver/3.2.0

(This error appeared before: #1724)

NickTheNeko2002, 2022-09-16:
update
I'm having issues again with openrazer and Basilisk V3
`:: Proceed with installation? [Y/n] 
(4/4) checking keys in keyring                     [######################] 100%
(4/4) checking package integrity                   [######################] 100%
(4/4) loading package files                        [######################] 100%
(4/4) checking for file conflicts                  [######################] 100%
(4/4) checking available disk space                [######################] 100%
:: Running pre-transaction hooks...
(1/1) Remove upgraded DKMS modules
==> dkms remove --no-depmod openrazer-driver/3.4.0 -k 5.15.68-1-lts
egrep: warning: egrep is obsolescent; using grep -E
==> dkms remove --no-depmod openrazer-driver/3.4.0 -k 5.19.7-arch1-1
egrep: warning: egrep is obsolescent; using grep -E
==> dkms remove openrazer-driver/3.4.0
egrep: warning: egrep is obsolescent; using grep -E
:: Processing package changes...
(1/4) reinstalling openrazer-driver-dkms-git       [######################] 100%
(2/4) reinstalling openrazer-daemon-git            [######################] 100%
(3/4) reinstalling python-openrazer-git            [######################] 100%
(4/4) reinstalling openrazer-meta-git              [######################] 100%
:: Running post-transaction hooks...
(1/3) Reloading device manager configuration...
(2/3) Arming ConditionNeedsUpdate...
(3/3) Install DKMS modules
п==> dkms install --no-depmod openrazer-driver/3.4.0 -k 5.15.68-1-lts
о==> depmod 5.15.68-1-lts
==> ERROR: Missing 5.19.8-zen1-1-zen kernel modules tree for module openrazer-driver/3.4.0.
==> ERROR: Missing 5.19.7-arch1-1 kernel modules tree for module openrazer-driver/3.4.0.
==> ERROR: Missing 5.19.8-hardened2-1-hardened kernel modules tree for module openrazer-driver/3.4.0.
[nick@hell ~]$ 
`

lah7, 2022-09-16:
> ==> ERROR: Missing 5.19.8-zen1-1-zen kernel modules tree for module openrazer-driver/3.4.0.
> ==> ERROR: Missing 5.19.7-arch1-1 kernel modules tree for module openrazer-driver/3.4.0.
> ==> ERROR: Missing 5.19.8-hardened2-1-hardened kernel modules tree for module openrazer-driver/3.4.0.

@NickTheNeko2002  Looks like you've got quite a few kernels installed. Which one is currently running?

    uname -a

This next command will list installed packages, you can use the version above to know which kernel package it belongs to, and whether it has an equivalent `-headers` installed, e.g. `linux-lts` and `linux-lts-headers`

    pacman -Q | grep linux

Your output doesn't suggest any problems if you're running the `linux-lts 5.15.68` kernel. Maybe a reboot will help? [Try going through the troubleshooting guide](https://github.com/openrazer/openrazer/wiki/Troubleshooting#2-kernel-modules-are-they-built-properly) and create a new issue if you're still stuck.

PS. Basilisk V3 is in the stable release now (since v3.3.0), you can go back to the normal `openrazer` (non-git) packages if you wish.

NickTheNeko2002, 2022-11-01:
also is there any GUI based Program for Remapping the Buttons?? because my mouse has buttons on the side and scrollwheel basically Mouse6, Mouse7 and Mouse8 and i want to re-map them but so far all i found are terminal solutions and i don't really like messing with command line/terminal tools, i'm scared of bricking my System again(So far i've bricked my OS-es like 20 times now i'm using EndavourOS)

lah7, 2022-11-01:
Try this tool: 

https://github.com/sezanzeb/input-remapper/

NickTheNeko2002, 2022-11-14:
hey are there any plans for adding multiple "Profiles" support in Polychromatic for the Razer Basilisk v3?
currently i can only change the DPI on the "White" LED Profile, i've tried on the "Red", "Green", "Blue", "Purple" and etc LED Profile's but no luck with them i tried to see if the "Sync DPI" button will work but no luck either i also tried to change the color in Prefferences > Customise > "Saved Colours (Tab)" Change Color but it still didn't want to work😟😟😟

lah7, 2022-11-15:
OpenRazer doesn't support onboard profiles at the moment, so apps can't support it. I'm not sure how "Sync DPI" works internally, but I would've figured that would at least store DPI ranges to the device across power cycles. Maybe switching profiles with that profile button causes it to reset?

From what I could find, that feature needs documenting and packet captures - https://github.com/openrazer/openrazer/issues/1259#issuecomment-695811506.

NickTheNeko2002, 2022-11-23:
> OpenRazer doesn't support onboard profiles at the moment, so apps can't support it. I'm not sure how "Sync DPI" works internally, but I would've figured that would at least store DPI ranges to the device across power cycles. Maybe switching profiles with that profile button causes it to reset?
> 
> From what I could find, that feature needs documenting and packet captures - [#1259 (comment)](https://github.com/openrazer/openrazer/issues/1259#issuecomment-695811506).

how can I help out with additional information?, I have the mouse but I lack the knowledge tho i am willing to try and help as much as i can, I am not the best at programming(still a noob at programming) but if it's something easy to do i would love to try and help out😟

NickTheNeko2002, 2022-12-22:
> Maybe switching profiles with that profile button causes it to reset?

it only works on the when i'm on the "White"/Blank "preset" which has no saved data on it when i switch to the green "Preset" LED which has stored button(Check Image 1 to see which button i'm talking about) mapping in it but then it doesn't change the DPI
i tried using "Input Remapper" but it doesn't recognize the side button (Mouse6(Check image #2) which i use most of the time for using and enabling my microphone, it would be nice if we could tweak atleast the button settings in PolyChromatic but it doesn't have such an option sadly

Image 1: 
[image]
Image 2: 
[image]