openrazer/openrazer #1744: DeathAdder 3.5G doesn't persist DPI across reboots

lah7, 2022-02-05:
> Originally reported in https://github.com/polychromatic/polychromatic/issues/404:

The DeathAdder 3.5G (a single DPI X device) keeps resetting its DPI value to 3500 across reboots. 

`~/.config/openrazer/persistence.conf` does not contain any `dpi` entries, so it might be silently failing to save the value (being the odd one with no Y value). Possibly that could be reason it resets because the persistence value defaults to `0`.

Manually writing the desired DPI to file and setting it as read only was confirmed to work:
```
[DeathAdder35G]
poll_rate = 1000
dpi_x = 900
dpi_y = 0
```

z3ntu, 2022-02-19:
With the fake driver this gets created in `persistence.conf`:
```
[DeathAdder35G]
dpi_x = 1800
dpi_y = 1800
poll_rate = 500
```
So looks ok?

codefiles, 2022-07-14:
@z3ntu
That does not look ok. `dpi_y` being anything other than `0` is invalid for a DPI X only device.
[pylib/openrazer/client/devices/mice.py#L87](../blob/ef99b35f1276aa0f979b51e505a8ffea1cb7a092/pylib/openrazer/client/devices/mice.py#L87)

[pylib/openrazer/_fake_driver/razerdeathadder3.5g.cfg#L9](../blob/ef99b35f1276aa0f979b51e505a8ffea1cb7a092/pylib/openrazer/_fake_driver/razerdeathadder3.5g.cfg#L9)
`800:800` is also invalid and should be something like `1800:0`. Is `800` even a valid `dpi_x` for this device?

closed 2022-07-24 by z3ntu

z3ntu, 2022-07-31:
FYI there has been one more fix related to this: https://github.com/openrazer/openrazer/commit/f205fce0794d24618c8c692a9b866c6664a4add7

I'll do a new stable release now but please still check if it works correctly for you.

PR #1848 (2022-07-24): daemon: Write persistence when Y=0 for DPI X only devices
Fixes #1744