openrazer/openrazer #1747: Please provide AKMOD for Fedora to make openrazer work on Silverblue

BlauerHunger, 2022-02-07:
The official way of providing dynamic kernel modules on Fedora is by using akmod. It consists of a src.rpm that builds the kernel module(s), that gets put under `/usr/src/akmods`. Installing and uninstalling the modules based on available kernels is handled automatically by the package manager. 

On rpm-ostree based distros like Fedora Silverblue and Fedora Kinoite, which have a steadily growing user base, dkms doesn't work because it's incompatible with the immutable base system, but akmod works and is officially supported. 

Please make openrazer drivers usable on Fedora Silverblue by providing akmod packages.

z3ntu, 2022-02-19:
Are you interested in working on adding support to the rpm package?

Sounds definitely great if the package would support those distros.

BrianAllred, 2022-04-14:
I can confirm that kyokenn's PR works for Fedora Silverblue 36.

DJKero, 2022-09-09:
Hello, I'm kinda new to linux and I'm daily driving Silverblue, is this going to be resolved by merging kyokenn's PR? If not, could you be so kind to point me to a resource so maybe I can help with this?

rscm, 2023-01-08:
> I can confirm that kyokenn's PR works for Fedora Silverblue 36.

I've build the packages but the last step is to add the user to plugdev group, but the group doesn't exists in the system anymore (/usr/lib/group). And for the looks of it (reading previous posts) that group is hardcoded.

dogo, 2023-03-21:
Any updates on this?

z3ntu, 2023-07-16:
@deisi uaccess doesn't work for sysfs last time I checked. But if you have better/newer info feel free to share (and maybe share a proof of concept)

Xake, 2023-11-20:
The merge request was closed due to Fedora Silverblue was not supported by the buildhosts.  
I would argue that the akmods modules still provide a value also in standard Fedora Workstation.  
https://src.fedoraproject.org/rpms/akmods/blob/rawhide/f/README.secureboot  
This makes it somewhat easier to setup so your akmods-built kernel modules are signed and loaded by a computer with secureboot enabled.

my4ng, 2024-07-02:
I have released [akmod-openrazer](https://github.com/my4ng/akmod-openrazer) repo that builds akmod package with podman/docker.

Totallynotmwa, 2024-07-14:
> I have released [akmod-openrazer](https://github.com/my4ng/akmod-openrazer) repo that builds akmod package with podman/docker.

Conflicting packages error :(

my4ng, 2024-07-14:
> Conflicting packages error :(

You need to remove all other openrazer backend packages, including the python daemon + library. The five packages I've listed should be self-contained.

Totallynotmwa, 2024-07-14:
> > Conflicting packages error :(
> 
> You need to remove all other openrazer backend packages, including the python daemon + library. The five packages I've listed should be self-contained.

Oh alr

svenjacobs, 2024-09-04:
Hello guys, I'm new to the concept of immutable distros. Currently using Fedora Workstation but thinking about switching to Silverblue. What is the current state of this issue? Is the [solution](https://github.com/openrazer/openrazer/issues/1747#issuecomment-2201707153) by @my4ng working? So I have to just build the RPMs and install them via `rpm-ostree`?

my4ng, 2024-10-19:
[akmod-openrazer](https://github.com/my4ng/akmod-openrazer) still works for version 3.9.0 for anyone interested.

my4ng, 2024-11-03:
I have made a [repository](https://my4ng.dev/repos/fedora-my4ng.repo) for Fedora containing latest git version of [akmod-openrazer](https://github.com/my4ng/akmod-openrazer):

1. `sudo dnf config-manager addrepo --from-repofile=https://my4ng.dev/repos/fedora-my4ng.repo`
2. `sudo dnf install akmod-openrazer kmod-openrazer openrazer openrazer-daemon python3-openrazer` 
3. `systemctl --user enable --now openrazer-daemon.service`

my4ng, 2025-03-15:
Shameless plug again: I've updated the [repo](https://github.com/my4ng/fedora-my4ng) package to 3.10.1 (latest git) if you are interested.

komorebinator, 2025-10-22:
@RobotRoss it's provided in uBlue-based distros, so you may consider switching to Bazzite