openrazer/openrazer #1748: Disable Volume Wheel on BlackWidow Elite?

jackz314, 2022-02-08:
### Distribution

Ubuntu 21.10

### Support

I've started to have trouble with the volume wheel of the keyboard recently where it changes the volume by itself randomly, and apparently I'm not alone ([example](https://www.reddit.com/r/razer/comments/bp0h26/huntsman_elite_volume_wheel_becomes_scroll/)). Anyways, I'm just wondering if there's a way to disable the volume wheel in some way through openrazer?

I tried [input-remapper](https://github.com/sezanzeb/input-remapper) and [xmodmap](https://superuser.com/a/775811/1015995), but it appears the volume wheel is different from regular keys and can't be controlled through these ways.

drewstaylor, 2022-05-25:
It's a mechanical issue and Razer continues to refuse to offer any help or support. The level of not caring about a very real issue that makes the product unusable is just unreal.

jackz314, 2022-09-23:
I think this is still a pretty significant issue for people, would someone be able to look into this more?

wyattearp, 2022-11-13:
As a note - I have been able to improve the performance of the volume wheel by using contact cleaner. After cleaning, mute & volume control were 100% accurate.

z3ntu, 2022-11-20:
@wyattearp For BlackWidow V3 this should've been fixed a while ago with #1751 and e.g. Ornata Chroma V2 with #1547. Can you make sure you're on the latest version and also provide `lsusb | grep 1532` ?

wyattearp, 2022-11-20:
@z3ntu - TL;DR - I'd recommend closing.

I can confirm (at least for me) on the latest Ubuntu with the latest openrazor (3.4.0). I saw that 3.5 was just cut like 2 hours before this comment so I just tested that on mine. For reference:

```
wyatt@lazarus ~/Downloads $ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy

wyatt@lazarus ~/Downloads $ lsusb | grep 1532
Bus 002 Device 046: ID 1532:025a Razer USA, Ltd Razer BlackWidow V3 Pro
```

I think you can close this issue (though I'm not the originator of it) unless there's something specific about Ubuntu 21.10.

I also want to clarify the mechanical performance issues I experienced with both the volume wheel as well as the volume mute button were resolved with cleaning (which this issue was the only place I found a reference originally to this issue). For future defect hunters, I used `CRC QD Contact Cleaner, 11 Wt Oz, Industrial Strength, Quick Drying, No Residue, Plastic-Safe Electronics Cleaner, Safe For Sensitive Electronics, Aerosol Spray`;  however, this comment does not endorse any specific product. I'm sure any appropriate contact cleaner that is plastic safe will function equally as well.

z3ntu, 2022-11-20:
Okay, let's close. If somebody sees this issue on other keyboards, please open a separate issue with the info which exact device you're using. And please try recent OpenRazer versions (v3.4.0 or v3.5.0)

closed 2022-11-20 by z3ntu

z3ntu, 2023-09-11:
@borgdylan Weird, the device is in the list of devices where this should be fixed: https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L3474

Anyways, as written in my last comment please open a new issue for this and don't necrobump issues.