openrazer/openrazer #1749: Always notification with 1.2% battery at startup, whatever the real battery charging level is (Viper Ultimate Wireless)

HanM23, 2022-02-11:
### Distribution

Manjaro Qonos 21.2.3 XFCE
Razer Viper Ultimate (Wireless + Dock)
OpenRazer 3.2.0

### Support

- [yes ] Have you tried rebooting?
- [ yes] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hi to all,

I have a small issue with openrazer notification when my computer starts, it always throws a notification showing 1.2% battery level, see the picture below.

[image]

Then, the next notification few minutes later (i put it at 60 seconds for testing purpose, but it could be 1 hour) shows the correct battery percentage remaining as you can see in the picture below.

[image]

Here, it's 24,7%, but the notification would be the same, with the same icon with the red bar at the bottom if the battery level is 80% (for instance).
--> If i restart the daemon within polychromatic, the notification becomes normal, i mean without the icon with the red bar.

The only error found in journalctl is :

`kernel: razermouse: Response doesn't match request. Start Marker: 02 id: ff Num Params: 02 Reserved: 00 Command: 81 Params: 01030000000000000000000000000000 .`

Do you already encountered this issue before ?
Thanks for your help.

z3ntu, 2022-02-19:
Can you `cat` the `charge_level` file repeatedly in `/sys/bus/hid/drivers/razermouse/0*/` and see if the output is consistent?

HanM23, 2022-02-21:
Hi,

Thanks for your answer, currently, my mouse is charged at 98,9 % and `cat /sys/bus/hid/drivers/razermouse/0*/charge_level` gives me 249.

Now, i just restarted my system, still have the notification with 1.2% when i enter into my DE (xfce) and `cat /sys/bus/hid/drivers/razermouse/0*/charge_level` gives me 249 too.

Then, as my notification frequency is set to 1 hour, in 1 hour the notification will pop-up with the right percentage (as explained in my previous message).

Also, i have to say I tried to delay by 60 seconds the start of `openrazer-daemon` by modifying `openrazer-daemon.service`, but it does not change anything --> the first notification right after started my system always gives me 1.2%

HanM23, 2022-02-21:
When i restart the daemon, the notification always gives me the right battery percentage.
It's only happening after a reboot.

z3ntu, 2022-02-21:
What happens when you unplug the device and plug it back in?

HanM23, 2022-02-21:
I restarted but it does not change anything.

I do not know if it can be useful but below is what i get in journactl related to my mouse, when i start my system.

```
journalctl -r | grep razer
févr. 21 20:13:01 xxxxxxxx kernel: razermouse: Response doesn't match request. Start Marker: 02 id: ff Num Params: 02 Reserved: 00 Command: 81 Params: 01030000000000000000000000000000 .
févr. 21 20:13:01 xxxxxxxx openrazer-daemon[1996]: 2022-02-21 20:13:01 | razer                          | INFO     | Serving DBus
févr. 21 20:13:00 xxxxxxxx openrazer-daemon[1996]: 2022-02-21 20:13:00 | razer.device1                  | INFO     | Initialising device.1 RazerMouseDock
févr. 21 20:13:00 xxxxxxxx openrazer-daemon[1996]: 2022-02-21 20:13:00 | razer                          | INFO     | Found device.1: 0003:1532:007E.0009
févr. 21 20:13:00 xxxxxxxx openrazer-daemon[1996]: 2022-02-21 20:13:00 | razer.device0                  | INFO     | Initialising device.0 RazerViperUltimateWireless
févr. 21 20:13:00 xxxxxxxx openrazer-daemon[1996]: 2022-02-21 20:13:00 | razer                          | INFO     | Found device.0: 0003:1532:007B.0006
févr. 21 20:13:00 xxxxxxxx openrazer-daemon[1996]: 2022-02-21 20:13:00 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
févr. 21 20:13:00 xxxxxxxx openrazer-daemon[1996]: 2022-02-21 20:13:00 | razer                          | INFO     | Initialising Daemon (v3.2.0). Pid: 1996
févr. 21 20:13:00 xxxxxxxx systemd[1519]: Started Daemon to manage razer devices in userspace.
févr. 21 20:13:00 xxxxxxxx dbus-daemon[1542]: [session uid=1000 pid=1542] Successfully activated service 'org.razer'
févr. 21 20:13:00 xxxxxxxx openrazer-daemon[1996]: Starting daemon.
févr. 21 20:13:00 xxxxxxxx systemd[1519]: Starting Daemon to manage razer devices in userspace...
févr. 21 20:13:00 xxxxxxxx dbus-daemon[1542]: [session uid=1000 pid=1542] Activating via systemd: service name='org.razer' unit='openrazer-daemon.service' requested by ':1.67' (uid=1000 pid=1953 comm="polychromatic-tray-applet                         ")
févr. 21 20:12:53 xxxxxxxx razer_mount[1412]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:007B.0008/
févr. 21 20:12:53 xxxxxxxx razer_mount[1405]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:007B.0006/
févr. 21 20:12:53 xxxxxxxx razer_mount[1400]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:007B.0008/
févr. 21 20:12:53 xxxxxxxx razer_mount[1404]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:007B.0007/
févr. 21 20:12:53 xxxxxxxx razer_mount[1399]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:007B.0006/
févr. 21 20:12:53 xxxxxxxx razer_mount[1401]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:007B.0007/
févr. 21 20:12:53 xxxxxxxx razer_mount[1397]: Device_ID 0003:1532:007B.0008
févr. 21 20:12:53 xxxxxxxx razer_mount[1396]: Device_ID 0003:1532:007B.0007
févr. 21 20:12:53 xxxxxxxx razer_mount[1393]: Device_ID 0003:1532:007B.0006
févr. 21 20:12:53 xxxxxxxx razer_mount[1391]: Driver razermouse
févr. 21 20:12:53 xxxxxxxx razer_mount[1389]: Driver razermouse
févr. 21 20:12:53 xxxxxxxx razer_mount[1390]: Driver razermouse
févr. 21 20:12:53 xxxxxxxx razer_mount[1382]: Device_ID 2-9.1:1.2
févr. 21 20:12:53 xxxxxxxx razer_mount[1380]: Device_ID 2-9.1:1.0
févr. 21 20:12:53 xxxxxxxx razer_mount[1378]: Driver razermouse
févr. 21 20:12:53 xxxxxxxx razer_mount[1379]: Device_ID 2-9.1:1.1
févr. 21 20:12:53 xxxxxxxx razer_mount[1377]: Driver razermouse
févr. 21 20:12:53 xxxxxxxx razer_mount[1376]: Driver razermouse
févr. 21 20:12:53 xxxxxxxx razer_mount[1371]: Modprobed razermouse
févr. 21 20:12:53 xxxxxxxx razer_mount[1366]: Changed group /sys/bus/hid/drivers/razeraccessory/0003:1532:007E.000A/
févr. 21 20:12:53 xxxxxxxx razer_mount[1365]: Changed group /sys/bus/hid/drivers/razeraccessory/0003:1532:007E.0009/
févr. 21 20:12:53 xxxxxxxx razer_mount[1364]: Changed group /sys/bus/hid/drivers/razeraccessory/0003:1532:007E.000B/
févr. 21 20:12:53 xxxxxxxx razer_mount[1360]: Changing group /sys/bus/hid/drivers/razeraccessory/0003:1532:007E.000A/
févr. 21 20:12:53 xxxxxxxx razer_mount[1357]: Device_ID 0003:1532:007E.000A
févr. 21 20:12:53 xxxxxxxx razer_mount[1359]: Changing group /sys/bus/hid/drivers/razeraccessory/0003:1532:007E.0009/
févr. 21 20:12:53 xxxxxxxx razer_mount[1358]: Changing group /sys/bus/hid/drivers/razeraccessory/0003:1532:007E.000B/
févr. 21 20:12:53 xxxxxxxx razer_mount[1356]: Device_ID 0003:1532:007E.000B
févr. 21 20:12:53 xxxxxxxx razer_mount[1355]: Device_ID 0003:1532:007E.0009
févr. 21 20:12:53 xxxxxxxx razer_mount[1354]: Driver razeraccessory
févr. 21 20:12:53 xxxxxxxx razer_mount[1352]: Driver razeraccessory
févr. 21 20:12:53 xxxxxxxx razer_mount[1353]: Driver razeraccessory
févr. 21 20:12:52 xxxxxxxx razer_mount[1346]: Device_ID 2-9.2:1.0
févr. 21 20:12:52 xxxxxxxx razer_mount[1345]: Device_ID 2-9.2:1.1
févr. 21 20:12:52 xxxxxxxx razer_mount[1344]: Device_ID 2-9.2:1.2
févr. 21 20:12:52 xxxxxxxx razer_mount[1340]: Driver razeraccessory
févr. 21 20:12:52 xxxxxxxx razer_mount[1338]: Driver razeraccessory
févr. 21 20:12:52 xxxxxxxx razer_mount[1339]: Driver razeraccessory
févr. 21 20:12:52 xxxxxxxx kernel: razeraccessory 0003:1532:007E.000B: input,hidraw10: USB HID v1.11 Keyboard [RAZER Razer Mouse Dock] on usb-0000:00:14.0-9.2/input2
févr. 21 20:12:52 xxxxxxxx razer_mount[1332]: Modprobed razeraccessory
[... cut ...]
févr. 21 20:12:52 xxxxxxxx kernel: razermouse 0003:1532:007B.0006: input,hidraw5: USB HID v1.11 Mouse [Razer Razer Viper Ultimate Dongle] on usb-0000:00:14.0-9.1/input0
févr. 21 20:12:52 xxxxxxxx razer_mount[1312]: Modprobing razeraccessory
févr. 21 20:12:52 xxxxxxxx razer_mount[1310]: Device_ID 2-9.2
févr. 21 20:12:52 xxxxxxxx razer_mount[1309]: Driver razeraccessory
févr. 21 20:12:52 xxxxxxxx razer_mount[1301]: Modprobing razermouse
févr. 21 20:12:52 xxxxxxxx razer_mount[1300]: Device_ID 2-9.1
févr. 21 20:12:52 xxxxxxxx razer_mount[1299]: Driver razermouse
```

z3ntu, 2022-03-26:
Try stopping openrazer-daemon, then unplug the device, plug it back in and run `cat /sys/bus/hid/drivers/razermouse/0*/charge_level` (make sure to not have any frontend like polychromatic or RazerGenie open, otherwise that might start the daemon again)

HanM23, 2022-04-10:
Hi,

I rebooted my system, had the notification with 1% level battery right at start,
I stopped the daemon with `systemctl --user stop openrazer-daemon`

Then, 

```
cat /sys/bus/hid/drivers/razermouse/0*/charge_level
244
```
I also updated every openrazer packages to 3.30 version :(

sannagy90, 2022-04-23:
I'm running into the same issue on fedora 36 gnome. In my case the reported battery level on startup is 2% and `cat /sys/bus/hid/drivers/razermouse/0*/charge_level` gives me the correct battery level.

9numbernine9, 2022-04-28:
I'm running into this issue as well (Viper Ultimate Wireless, OpenRazer `3.3.0`) on Arch Linux, though for whatever reason my notification says 3% when I start up instead of 1.2%. 

I turned the logging level up to `DEBUG` and the logs look like this:
<details>
  <summary>Click for DEBUG logs (fresh boot)</summary>

```
2022-04-27 23:02:57 | razer                          | INFO     | Initialising Daemon (v3.3.0). Pid: 1001
2022-04-27 23:02:57 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationV2Chroma ---------------- (1532:0F20)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskEssential ------------------ (1532:0065)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateReceiver ----------- (1532:0088)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateWired -------------- (1532:0086)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV2 ------------------------- (1532:0085)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3 ------------------------- (1532:0099)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowTournamentEdition2014 ---- (1532:011C)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3 ----------------------- (1532:024E)
2022-04-27 23:02:57 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowV3MiniHyperspeed --------- (1532:0258)
[... cut ...]
[Install]
WantedBy=default.target
```

And then looking at the `charge_level` before the daemon starts:

```
% cat /sys/bus/hid/drivers/razermouse/0003:1532:007B.0004/charge_level 
244
```

... Which looks like a reasonably good approximation of 97%, so I don't think it's a race between driver and daemon. I'm not a Python expert and haven't looked too closely at the code, but is there an initial value used in the calculation of battery percentage that _doesn't_ come from the the `charge_level` ?

z3ntu, 2022-07-23:
Can you please try adding this patch and sending debug logs again when this issue occurs?
```diff
diff --git a/daemon/openrazer_daemon/dbus_services/dbus_methods/mamba.py b/daemon/openrazer_daemon/dbus_services/dbus_methods/mamba.py
index 92e42b6e..08485721 100644
--- a/daemon/openrazer_daemon/dbus_services/dbus_methods/mamba.py
+++ b/daemon/openrazer_daemon/dbus_services/dbus_methods/mamba.py
@@ -15,11 +15,15 @@ def get_battery(self):
     driver_path = self.get_driver_path('charge_level')
 
     with open(driver_path, 'r') as driver_file:
-        battery_255 = float(driver_file.read().strip())
+        value = driver_file.read().strip()
+        self.logger.debug("value=" + str(value))
+        battery_255 = float(value)
+        self.logger.debug("battery_255=" + str(battery_255))
         if battery_255 < 0:
             return -1.0
 
         battery_100 = (battery_255 / 255) * 100
+        self.logger.debug("battery_100=" + battery_100)
         return battery_100
 
 

```

9numbernine9, 2022-07-23:
[Here you go!](https://gist.github.com/9numbernine9/ed5fa8d0b3f20a537c8cdd1b1e5cf58f)
Had to restart a few times before it happened, but it did eventually.

z3ntu, 2022-07-23:
@9numbernine9 Thanks! Seems really the value `7` is coming from the kernel driver.
Now can you add this patch to the kernel driver and rebuild/reload your modules and check dmesg when this happens?
```diff
diff --git a/driver/razermouse_driver.c b/driver/razermouse_driver.c
index 4f4ccf18..51afaf2a 100644
--- a/driver/razermouse_driver.c
+++ b/driver/razermouse_driver.c
@@ -1019,6 +1019,7 @@ static ssize_t razer_attr_read_charge_level(struct device *dev, struct device_at
 
     response_report = razer_send_payload(usb_dev, &report);
 
+    printk(KERN_WARNING "razermouse FINDME: status=%d arg0=%x arg1=%x\n", response_report.status, response_report.arguments[0], response_report.arguments[1]);
     return sprintf(buf, "%d\n", response_report.arguments[1]);
 }
 

```
If you need any help let me know.

9numbernine9, 2022-07-23:
Rebooted and saw these:
```
[   10.738486] razermouse FINDME: status=2 arg0=0 arg1=7
[   11.560488] razermouse FINDME: status=2 arg0=0 arg1=f9
```

z3ntu, 2022-07-23:
Hm so we send "get battery level" command and get back "get dpi" response (class 0x04, id 0x85), and DPI first byte is 0x07 for you and because the driver code is so great (hint: it's not) it's just interpreting that as charge level. I guess mystery solved!

In general this should get solved once I get around to do my planned changes in the driver code. But I guess it'll still take a while to do this, and I guess you want a fix/workaround earlier? :)

9numbernine9, 2022-07-23:
@z3ntu I would love a fix for workaround! With that said you're under no obligation to provide one in a timely fashion, and I suspect the fix might be non-trivial (e.g. validate commands and responses match and (possibly) retry if they don't?)!

z3ntu, 2022-07-23:
Exactly. Do this and if the command really fails then don't just return a number to user space (mostly 0) but use some -EFOOBAR response so user space actually knows it fails. And the problem is this has never been done so probably the daemon will die quite quickly when this happens, so lots of testing etc needed :)

9numbernine9, 2022-07-23:
Sounds reasonable. Whenever the patch is ready I will do my best to help test it if you need ~~victims~~ volunteers.

z3ntu, 2022-07-23:
Can you try this?
```diff
diff --git a/driver/razermouse_driver.c b/driver/razermouse_driver.c
index 77a6ae6c..df06ca85 100644
--- a/driver/razermouse_driver.c
+++ b/driver/razermouse_driver.c
@@ -1035,6 +1035,15 @@ static ssize_t razer_attr_read_charge_level(struct device *dev, struct device_at
 
     response_report = razer_send_payload(usb_dev, &report);
 
+    // On Viper Ultimate Wireless we sometimes get mismatching response leading
+    // to erroneous battery notification.
+    // TODO Get rid of this ugly workaround by fixing this behavior generally
+    if(response_report.remaining_packets != report.remaining_packets ||
+           response_report.command_class != report.command_class ||
+           response_report.command_id.id != report.command_id.id) {
+        response_report = razer_send_payload(usb_dev, &report);
+    }
+
     return sprintf(buf, "%d\n", response_report.arguments[1]);
 }
 

```
Should/could fix this, by basically retrying once for this specific error. I'll include this if it really fixes this error

9numbernine9, 2022-07-23:
I tweaked the patch slightly to add a `printk` before the value of `response_report` gets set inside the `if` statement, but I don't think the fix worked:
```
[    3.872678] razermouse 0003:1532:007B.0004: input,hidraw3: USB HID v1.11 Mouse [Razer Razer Viper Ultimate Dongle] on usb-0000:2a:00.1-2.1.1/input0
[    4.052528] razermouse 0003:1532:007B.0005: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer Viper Ultimate Dongle] on usb-0000:2a:00.1-2.1.1/input1
[    4.369240] razermouse 0003:1532:007B.0006: input,hidraw8: USB HID v1.11 Keyboard [Razer Razer Viper Ultimate Dongle] on usb-0000:2a:00.1-2.1.1/input2
[    9.848636] razermouse: Response doesn't match request. Start Marker: 02 id: ff Num Params: 07 Reserved: 04 Command: 85 Params: 00070807080000000000000000000000 .
[    9.848641] razermouse FINDME: Applied the fix!
[    9.895644] razermouse: Response doesn't match request. Start Marker: 02 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00f90000000000000000000000000000 .
[    9.912638] razermouse: Response doesn't match request. Start Marker: 02 id: ff Num Params: 07 Reserved: 04 Command: 85 Params: 00070807080000000000000000000000 .
[    9.912643] razermouse FINDME: status=2 arg0=0 arg1=7
[   10.472647] razermouse FINDME: status=2 arg0=0 arg1=f9
```
Still received a "3% Warning" notification on the desktop. :-(

z3ntu, 2022-07-23:
Lol now we're getting a different response back. Fun!

Does this "Response doesn't match request" error also appear in dmesg during normal usage, e.g. changing settings in polychromatic? Or just during setup or whatever when the device is busy?

z3ntu, 2022-07-23:
Also in theory the transaction ID used is supposed to be an incrementing integer (or at least part of that 8-bit int, I think some bits are reserved for a different purpose in some cases), so that the host can match up request and get correct response back and not just whatever comes next - which is what happens currently. Might be worse on wireless because of the enormous delay (compared to wired use case)

9numbernine9, 2022-07-23:
> Lol now we're getting a different response back. Fun!
> 
> Does this "Response doesn't match request" error also appear in dmesg during normal usage, e.g. changing settings in polychromatic? Or just during setup or whatever when the device is busy?

I have _only_ ever seen this occur during initial boot-up when the `openrazer-daemon` checks the battery status for the first time; subsequently restarting the daemon always shows the correct battery percentage in the logs afterward. Playing around in Polychromatic doesn't seem to trigger that log entry either - I just see `razermouse FINDME: status=2 arg0=0 arg1=f7` whenever I tweak the mouse settings in Polychromatic.

z3ntu, 2022-07-23:
That leads me to believe that when we throw a bunch of requests at the device at the same time (= close enough to each other) at daemon startup then the driver/device gets confused which response it should get, and this happens..

z3ntu, 2022-07-23:
Can you add a `usleep_range(600000, 600100);` (or even longer wait time) before it sends the packet again? Probably just waiting for a bit until it calms down might be enough.

9numbernine9, 2022-07-23:
It's a hack, but I think it works. :sweat_smile: 

My source now looks like this:
```
    response_report = razer_send_payload(usb_dev, &report);

    // On Viper Ultimate Wireless we sometimes get mismatching response leading
    // to erroneous battery notification.
    // TODO Get rid of this ugly workaround by fixing this behavior generally
    if(response_report.remaining_packets != report.remaining_packets ||
           response_report.command_class != report.command_class ||
           response_report.command_id.id != report.command_id.id) {
	printk(KERN_WARNING "razermouse FINDME: Sleeping before re-issuing command...");
	usleep_range(600000, 600100);
        response_report = razer_send_payload(usb_dev, &report);
	printk(KERN_WARNING "razermouse FINDME: Command re-issued!");
    }

    printk(KERN_WARNING "razermouse FINDME: status=%d arg0=%x arg1=%x\n", response_report.status, response_report.arguments[0], response_report.arguments[1]);
    return sprintf(buf, "%d\n", response_report.arguments[1]);
```

`dmesg` output looks more like what we would hope to see:
```
[    3.766000] razermouse 0003:1532:007B.0004: input,hidraw3: USB HID v1.11 Mouse [Razer Razer Viper Ultimate Dongle] on usb-0000:2a:00.1-2.1.1/input0
[    3.919179] razermouse 0003:1532:007B.0005: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer Viper Ultimate Dongle] on usb-0000:2a:00.1-2.1.1/input1
[    4.245853] razermouse 0003:1532:007B.0006: input,hidraw5: USB HID v1.11 Keyboard [Razer Razer Viper Ultimate Dongle] on usb-0000:2a:00.1-2.1.1/input2
[   10.089962] razermouse: Response doesn't match request. Start Marker: 01 id: ff Num Params: 07 Reserved: 04 Command: 85 Params: 00000000000000000000000000000000 .
[   10.089967] razermouse FINDME: Sleeping before re-issuing command...
[   10.275962] razermouse FINDME: status=2 arg0=0 arg1=f7
[   10.753959] razermouse FINDME: Command re-issued!
[   10.753961] razermouse FINDME: status=2 arg0=0 arg1=f7
```
And the daemon seems happy too:
```
Jul 23 16:51:25 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:25 | razer.device0                  | DEBUG    | DBus call get_battery
Jul 23 16:51:25 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:25 | razer.device0                  | DEBUG    | value=247
Jul 23 16:51:25 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:25 | razer.device0                  | DEBUG    | battery_255=247.0
Jul 23 16:51:25 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:25 | razer.device0                  | DEBUG    | battery_100=96.86274509803921
Jul 23 16:51:25 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:25 | razer.device0                  | DEBUG    | DBus call is_charging
Jul 23 16:51:25 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:25 | razer.device0                  | DEBUG    | DBus call get_logo_brightness
Jul 23 16:51:25 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:25 | razer.device0                  | DEBUG    | DBus call get_current_logo_effect
Jul 23 16:51:25 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:25 | razer.device0                  | DEBUG    | DBus call get_current_logo_effect_colors
Jul 23 16:51:25 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:25 | razer.device0                  | DEBUG    | DBus call get_current_logo_effect
Jul 23 16:51:25 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:25 | razer.device0                  | DEBUG    | DBus call get_current_logo_wave_dir
Jul 23 16:51:25 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:25 | razer.device0                  | DEBUG    | DBus call get_current_logo_effect_speed
Jul 23 16:51:26 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:26 | razer.device0                  | DEBUG    | value=247
Jul 23 16:51:26 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:26 | razer.device0                  | DEBUG    | battery_255=247.0
Jul 23 16:51:26 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:26 | razer.device0                  | DEBUG    | battery_100=96.86274509803921
Jul 23 16:51:26 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:26 | razer.device0.batterynotifier  | DEBUG    | Razer Viper Ultimate Wireless Battery at 97%
Jul 23 16:51:35 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:35 | razer                          | DEBUG    | State recently changed, writing to disk
Jul 23 16:51:35 mycomputer openrazer-daemon[967]: 2022-07-23 16:51:35 | razer                          | DEBUG    | Writing persistence config
```
Best of all: no incorrect desktop notification. :smile:

z3ntu, 2022-07-23:
Keep using this for a bit longer - maybe one or two weeks, and let me know if it really really works ;D Can you remember pinging me after that and tell me the results? Otherwise I'll set a reminder to ask

9numbernine9, 2022-07-23:
I'll let you know in two weeks or so but feel free to poke me if you want an update sooner than that. :-)

9numbernine9, 2022-08-06:
@z3ntu Two week update: still works great! No incorrect notifications about my battery level, and no other observed incorrect behaviour. 😄 Note that I haven't applied the patches in this issue to the recently released `3.4.0` release.

closed 2022-11-25 by HanM23