openrazer/openrazer #1756: Razer Naga Trinity not recognized although it is supported

jaamivstheworld, 2022-02-17:
### Distribution
Artix Linux (KDE + OpenRC)

### Support

- [yes] Have you tried rebooting?
- [yes] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Although my Razer Naga Trinity is on the supported list, Polychromatic does not recognize it. I ran the troubleshooter and this is the output I got: 
[image]

I'm a very big noob and do not understand what I am supposed to do.

lah7, 2022-02-18:
Looks like the driver isn't installed, because the kernel headers were missing (they're needed to build the driver). Try installing them:

    sudo pacman -S linux-headers

(Or use `linux-lts-headers` if you use the long term release kernel)

Then build the driver, and reboot:

    sudo dkms install openrazer-driver/3.2.0

This is what the "Kernel Modules" is referring to in [the troubleshooting guide](https://github.com/openrazer/openrazer/wiki/Troubleshooting).

jaamivstheworld, 2022-02-18:
I already have linux-headers installed

lah7, 2022-02-18:
What's the output of this command (if anything)?

    sudo modprobe razermouse

jaamivstheworld, 2022-02-19:
Looks like I won't be needing to do that! It just somehow got fixed after I reinstall linux-headers. I don't know why it didn't work before, considering I still had linux-headers before too.

closed 2022-02-19 by lah7