openrazer/openrazer #1757: Support for [device name here]

Rakansami, 2022-02-18:
### Device Info

* Name: <See under device>
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```

```
Please post the output of
    ls -lah /dev/input/by-id/
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

z3ntu, 2022-02-18:
Please fill in the issue template.

z3ntu, 2022-11-20:
Closing since there is no response.

closed 2022-11-20 by z3ntu