openrazer/openrazer #1761: Razer Thunderbolt 4 Dock Chroma is not persistent

psykher, 2022-02-21:
### Distribution

Debian 5.10.92-1
Openrazer 3.2.0
Polychromatic 0.7.3

### Support

- [ ] Have you tried rebooting? Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? Yes

My Razer Thunderbolt 4 Dock Chroma effects are not being persistent after a system restart. Every system reboot resets the effects on this device. Same scenario when unplug & plug

Device Id: 1532:0f21

```
$ lsusb | grep -i 1532
Bus 004 Device 007: ID 1532:0f30 Razer USA, Ltd USB3.1 Hub
Bus 003 Device 003: ID 1532:024a Razer USA, Ltd Blade Stealth (Late 2019)
Bus 003 Device 031: ID 1532:122e Razer USA, Ltd Thunderbolt 4 Dock
Bus 003 Device 030: ID 1532:0f21 Razer USA, Ltd RAZER THUNDERBOLT 4 DOCK CHROMA
Bus 003 Device 028: ID 1532:0f0d Razer USA, Ltd Laptop Stand Chroma
Bus 003 Device 029: ID 1532:007e Razer USA, Ltd RC30-030502 Mouse Dock
Bus 003 Device 026: ID 1532:007b Razer USA, Ltd RC30-0305 Gaming Mouse Dongle [Viper Ultimate (Wireless)]
Bus 003 Device 021: ID 1532:0f2f Razer USA, Ltd USB2.0 Hub
Bus 003 Device 034: ID 1532:0f26 Razer USA, Ltd Razer Charging Pad Chroma
Bus 003 Device 022: ID 1532:0f12 Razer USA, Ltd Razer Raptor 27
```

Devices ownership:

```
$ ls -las /dev/bus/usb
0 crw-rw-r-- 1 root root 189, 390 Feb 20 15:26 007
0 crw-rw-r-- 1 root plugdev 189, 258 Feb 20 14:55 003
0 crw-rw-r-- 1 root root    189, 286 Feb 20 15:26 031
0 crw-rw-r-- 1 root plugdev 189, 285 Feb 20 15:26 030
0 crw-rw-r-- 1 root root    189, 283 Feb 20 15:26 028
0 crw-rw-r-- 1 root plugdev 189, 284 Feb 20 15:26 029
0 crw-rw-r-- 1 root plugdev 189, 281 Feb 20 15:26 026
0 crw-rw-r-- 1 root root    189, 276 Feb 20 15:26 021
0 crw-rw-r-- 1 root plugdev 189, 289 Feb 20 16:18 034
0 crw-rw-r-- 1 root root    189, 277 Feb 20 15:26 022
```

Openrazer logs:

```
$ cat ~/.local/share/openrazer/logs/razer.log
2022-02-20 14:55:37 | razer                          | INFO     | Initialising Daemon (v3.2.0). Pid: 1979
2022-02-20 14:55:37 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-02-20 14:55:37 | razer                          | INFO     | Found device.0: 0003:1532:024A.0004
2022-02-20 14:55:37 | razer.device0                  | INFO     | Initialising device.0 RazerBladeStealthLate2019
2022-02-20 14:55:37 | razer.device0.keymanager       | WARNING  | No event files for KeyWatcher
2022-02-20 14:55:37 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2022-02-20 14:55:37 | razer                          | INFO     | Serving DBus
2022-02-20 14:55:57 | razer                          | INFO     | Found valid device.1: 0003:1532:0F21.0010
2022-02-20 14:55:57 | razer.device1                  | INFO     | Initialising device.1 RazerThunderbolt4DockChroma
2022-02-20 14:55:57 | razer                          | INFO     | Found valid device.2: 0003:1532:007E.000D
2022-02-20 14:55:57 | razer.device2                  | INFO     | Initialising device.2 RazerMouseDock
2022-02-20 14:55:58 | razer                          | INFO     | Found valid device.3: 0003:1532:007B.0006
2022-02-20 14:55:58 | razer.device3                  | INFO     | Initialising device.3 RazerViperUltimateWireless
2022-02-20 14:55:58 | razer                          | INFO     | Found valid device.4: 0003:1532:0F26.0011
2022-02-20 14:55:58 | razer.device4                  | INFO     | Initialising device.4 RazerChargingPadChroma
2022-02-20 15:26:36 | razer                          | WARNING  | Removing 0003:1532:0F26.0011
2022-02-20 15:26:36 | razer                          | WARNING  | Removing 0003:1532:007B.0006
2022-02-20 15:26:36 | razer                          | WARNING  | Removing 0003:1532:007E.000D
2022-02-20 15:26:37 | razer                          | WARNING  | Removing 0003:1532:0F21.0010
2022-02-20 15:26:46 | razer                          | INFO     | Found valid device.1: 0003:1532:0F21.0021
2022-02-20 15:26:46 | razer.device1                  | INFO     | Initialising device.1 RazerThunderbolt4DockChroma
2022-02-20 15:26:46 | razer                          | INFO     | Found valid device.2: 0003:1532:007E.001D
2022-02-20 15:26:46 | razer.device2                  | INFO     | Initialising device.2 RazerMouseDock
2022-02-20 15:26:47 | razer                          | INFO     | Found valid device.3: 0003:1532:007B.0016
2022-02-20 15:26:47 | razer.device3                  | INFO     | Initialising device.3 RazerViperUltimateWireless
2022-02-20 15:26:47 | razer                          | INFO     | Found valid device.4: 0003:1532:0F26.0022
2022-02-20 15:26:47 | razer.device4                  | INFO     | Initialising device.4 RazerChargingPadChroma
2022-02-20 15:29:25 | razer                          | INFO     | Initialising Daemon (v3.2.0). Pid: 11055
2022-02-20 15:29:25 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-02-20 15:29:25 | razer                          | INFO     | Found device.0: 0003:1532:0F26.0022
2022-02-20 15:29:25 | razer.device0                  | INFO     | Initialising device.0 RazerChargingPadChroma
2022-02-20 15:29:25 | razer                          | INFO     | Found device.1: 0003:1532:007B.0016
2022-02-20 15:29:25 | razer.device1                  | INFO     | Initialising device.1 RazerViperUltimateWireless
2022-02-20 15:29:25 | razer                          | INFO     | Found device.2: 0003:1532:007E.001D
2022-02-20 15:29:25 | razer.device2                  | INFO     | Initialising device.2 RazerMouseDock
2022-02-20 15:29:25 | razer                          | INFO     | Found device.3: 0003:1532:0F21.0021
2022-02-20 15:29:25 | razer.device3                  | INFO     | Initialising device.3 RazerThunderbolt4DockChroma
2022-02-20 15:29:25 | razer                          | INFO     | Found device.4: 0003:1532:024A.0004
2022-02-20 15:29:25 | razer.device4                  | INFO     | Initialising device.4 RazerBladeStealthLate2019
2022-02-20 15:29:25 | razer.device4.keymanager       | WARNING  | No event files for KeyWatcher
2022-02-20 15:29:25 | razer.device4                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2022-02-20 15:29:25 | razer                          | INFO     | Serving DBus
2022-02-20 15:39:24 | razer                          | INFO     | Initialising Daemon (v3.2.0). Pid: 15369
2022-02-20 15:39:24 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2022-02-20 15:39:24 | razer                          | INFO     | Found device.0: 0003:1532:0F26.0022
2022-02-20 15:39:24 | razer.device0                  | INFO     | Initialising device.0 RazerChargingPadChroma
2022-02-20 15:39:24 | razer                          | INFO     | Found device.1: 0003:1532:007B.0016
2022-02-20 15:39:24 | razer.device1                  | INFO     | Initialising device.1 RazerViperUltimateWireless
2022-02-20 15:39:24 | razer                          | INFO     | Found device.2: 0003:1532:007E.001D
2022-02-20 15:39:24 | razer.device2                  | INFO     | Initialising device.2 RazerMouseDock
2022-02-20 15:39:24 | razer                          | INFO     | Found device.3: 0003:1532:0F21.0021
2022-02-20 15:39:24 | razer.device3                  | INFO     | Initialising device.3 RazerThunderbolt4DockChroma
2022-02-20 15:39:24 | razer                          | INFO     | Found device.4: 0003:1532:024A.0004
2022-02-20 15:39:24 | razer.device4                  | INFO     | Initialising device.4 RazerBladeStealthLate2019
2022-02-20 15:39:24 | razer.device4.keymanager       | WARNING  | No event files for KeyWatcher
2022-02-20 15:39:24 | razer.device4                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2022-02-20 15:39:24 | razer                          | INFO     | Serving DBus
2022-02-20 16:18:34 | razer                          | WARNING  | Removing 0003:1532:0F26.0022
2022-02-20 16:18:57 | razer                          | INFO     | Found valid device.4: 0003:1532:0F26.0024
2022-02-20 16:18:57 | razer.device4                  | INFO     | Initialising device.4 RazerChargingPadChroma
```

Device mode:

```
$ cat /sys/bus/hid/drivers/razeraccessory/*1532:0F21*/device_mode
3:0
```

Any help would be very appreciated!

lah7, 2022-02-21:
The daemon has a software persistence feature, intended to restart hardware effects on login (and I think on plug too). Newer Razer hardware don't generally remember their last effect. Make sure it is enabled in this file:

    ~/.config/openrazer/razer.conf

```
[Startup]
restore_persistence = True
```

psykher, 2022-02-21:
Thank you so much, it works!

closed 2022-02-21 by lah7