openrazer/openrazer #1765: Support for Razer Basilisk Essential

hanssia, 2022-02-27:
### Device Info

* Name: Razer Basilisk Essential
* Product Number: 1532
* Model Number: 0065

I can see the Basilisk Essential is in the supported device list, but I still get the "unrecognized device", "openrazer hasn't registered this device" error.  Please see the screenshot.

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 003 Device 003: ID 1532:0065 Razer USA, Ltd RZ01-0265, Gaming Mouse [Basilisk Essential]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0065 RZ01-0265, Gaming Mouse [Basilisk Essential]
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Basilisk Essential
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

```
Please post the output of
    ls -lah /dev/input/by-id/
```
total 0
drwxr-xr-x 2 root root 240 Feb 27 12:55 .
drwxr-xr-x 4 root root 680 Feb 27 18:05 ..
lrwxrwxrwx 1 root root   9 Feb 27 12:55 usb-CATEX_TECH._108EC-XRGB_CA2018120001-event-if02 -> ../event3
lrwxrwxrwx 1 root root   9 Feb 27 12:55 usb-CATEX_TECH._108EC-XRGB_CA2018120001-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 Feb 27 12:55 usb-CATEX_TECH._108EC-XRGB_CA2018120001-if02-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Feb 27 12:55 usb-CATEX_TECH._108EC-XRGB_CA2018120001-if02-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 Feb 27 12:55 usb-CATEX_TECH._108EC-XRGB_CA2018120001-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Feb 27 12:55 usb-Razer_Razer_Basilisk_Essential-event-if01 -> ../event9
lrwxrwxrwx 1 root root   9 Feb 27 12:55 usb-Razer_Razer_Basilisk_Essential-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Feb 27 12:55 usb-Razer_Razer_Basilisk_Essential-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root  10 Feb 27 12:55 usb-Razer_Razer_Basilisk_Essential-if02-event-kbd -> ../event10
lrwxrwxrwx 1 root root   9 Feb 27 12:55 usb-Razer_Razer_Basilisk_Essential-mouse -> ../mouse1

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>
[image]

lah7, 2022-02-27:
Since the device is supported, your installation of OpenRazer was not successful.

Please let us know which distro you're using. Click **Troubleshoot** in Polychromatic to see what the problem is, and [follow the troubleshooting wiki page](https://github.com/openrazer/openrazer/wiki/Troubleshooting) on fixing the unrecognized state. Let us know if you get stuck.

Most likely, the driver wasn't built properly or your user is not in the `plugdev` group.

hanssia, 2022-02-27:
> Since the device is supported, your installation of OpenRazer was not successful.
>
> Please let us know which distro you're using. Click **Troubleshoot** in Polychromatic to see what the problem is, [...]
I'm using ArchLinux and installed OpenRazer with yay.

The issue is indeed my installation of OpenRazer was not successful.
A couple of months a go I uninstalled the linux-headers package, so the DKMS modules could not be installed.
After I installed linux-headers, the OpenRazer driver was installed automatically with it.

closed 2022-02-27 by hanssia