openrazer/openrazer #1767: Basilisk V3 is not recognized after reboot (only for specific RGB settings)

joos2k, 2022-03-01:
### Distribution

Arch, fully updated
DKMS ist working fine, no problems so far:

$ sudo dkms status
openrazer-driver/3.2.0, 5.15.25-1-lts, x86_64: installed
openrazer-driver/3.2.0, 5.16.11-arch1-1, x86_64: installed

I used the following packages:
aur/openrazer-meta-git
aur/openrazer-driver-dkms-git
aur/openrazer-daemon-git
aur/python-openrazer-git
aur/polychromatic

### Support

- [x] Have you tried rebooting? <- that causes the problem :-)
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

If I set a static color for the wheel only and turn all other LEDs off, then the mouse is not recognized after reboot. Even if I connect the mouse to another computer - totally desperate I had even tried a Windows box but the result was the same.

Log:
Mär 01 20:18:48 kallisto kernel: usb 1-11: device not accepting address 6, error -71
Mär 01 20:18:48 kallisto kernel: usb 1-11: new full-speed USB device number 7 using xhci_hcd
Mär 01 20:18:48 kallisto kernel: usb 1-11: Device not responding to setup address.
Mär 01 20:18:48 kallisto kernel: usb 1-11: Device not responding to setup address.
Mär 01 20:18:48 kallisto kernel: usb 1-11: device not accepting address 7, error -71
Mär 01 20:18:48 kallisto kernel: usb usb1-port11: unable to enumerate USB device

If I plug it out and in again, then it works fine until I reboot or power cycle the PC.

If I set the default color scheme (spectrum) or if I set a static color for all LEDs (it took me a while to figure that out) then the mouse works fine even after a reboot or PC power cycle.

I have no clue. Maybe it's a firmware bug.
I'm sure my info is way to unspecific, but I don't know where to start with. What could I provide (or do) to further investigate that strange behavior?

HeyPapito, 2022-03-06:
My basilisk v3 is not recognized at all, tried reinstalling the packages as well. My Cynosa Chroma keyboard works fine though so can't be a corrupt installation

joos2k, 2022-03-06:
@HeyPapito If you are on Arch too, you need the -git packages, because the Basilisk V3 was added after the v3.2.0 release.

HeyPapito, 2022-03-06:
@joos2k I am on zorin os (which is pretty much ubuntu) so unsure about how to install git packages

joos2k, 2022-03-06:
@HeyPapito Hmm, have a look here: https://openrazer.github.io/#ubuntu - I think in that case you can use the openrazer/daily PPA. But I don't have any experience with that, so please don't blame me. ;-)

quietvoid, 2022-04-23:
I just encountered this as well.
I am using polychromatic, and setting the effects to None makes the mouse not power back on at boot (both in BIOS and Linux).

So far my workaround is to leave the effect to default (Spectrum) and set the brightness to zero.

LethalManBoob, 2022-08-01:
Same issue here. Putting my mouse on default fixed it.

audoh, 2022-10-17:
Just had this happen to me, can confirm Polychromatic had for some reason set my mouse's firmware effects to None. Setting them to static and then applying my custom effect afterwards fixed the issue.

joos2k, 2022-10-17:
Sometimes (very rare) my Basilisk V3 still refuses to serve after a reboot. There is a new [firmware](https://mysupport.razer.com/app/answers/detail/a_id/6246), maybe this will solve the problem. It hasn't happened again for me since the update, but as already mentioned: The phenomenon only occurs rarely.

tomsquest, 2023-03-09:
Same here after **uninstalling** openrazer and polychromatic : the mouse is not recognized after boot and has to be unplug and plugged again to be recognized.

To uninstall I ran `sudo apt remove openrazer-meta --purge`, then the problem arose at the next boot.

tomsquest, 2023-03-09:
I did a factory reset under windows and it seems that it fixed the issue : no more needed to unplug the mouse after booting.

closed 2023-08-08 by github-actions[bot]