openrazer/openrazer #1768: Device 1532:008F and 1532:0244 not detected.

Kiaos, 2022-03-02:
Device 1532:008F and 1532:0244 not detected. I've reinstalled all the packages I'm using arch linux kde plasma wayland. First time using linux so know im doing something wrong. btw got the device number/letters from polychromatic. razer genie shows nothing at all.

one device is razer tartrus v2 and other is newist razer naga trinity wireless. 

I'm a noob to Linux but I guess installing arch from command line was my right of passage into this experience. Google doesn't help when trying to find answers to openrazer problems so I'm asking here.

lah7, 2022-03-02:
Nothing might be wrong with the installation. The device IDs `008F` and `0244` are not supported in any released build of OpenRazer yet. They are a work in progress or need testing:

*  https://github.com/openrazer/openrazer/pull/1692
*  https://github.com/openrazer/openrazer/pull/1577

You could test one or the other by [building a package](https://github.com/openrazer/openrazer/wiki/Building-a-package#arch-linux) from one of those branches in the meantime.

Kiaos, 2022-03-03:
I would like to test the build but i have no idea how to build the package or what url to use in the instructions to build it. yea the guide is helpful but not sure what i should be linking it to.

LinuxDonald, 2022-03-03:
@Kiaos are you using arch linux? Than I could make packages for you.

Kiaos, 2022-03-04:
> @Kiaos are you using arch linux? Than I could make packages for you.

yes

Kiaos, 2022-03-04:
Just tell me how do i write source=("git+https://<git repo url>#commit=<commit hash>") with 23c9f4fca05f59e8ea745c534999cbc850860e27 commit

LinuxDonald, 2022-03-04:
why commit? you need to use branch or repo

lah7, 2022-03-04:
Underneath the title of a pull request shows where the proposed changes are coming **from**. This indicates the username and branch to build from.

For #1692, it would be:

```diff
-source=("git+https://github.com/openrazer/openrazer")
+source=("git+https://github.com/AliceGrey/openrazer#branch=razer-naga-pro")
```

For #1577:
```diff
-source=("git+https://github.com/openrazer/openrazer")
+source=("git+https://github.com/AlexandreHertzog/openrazer#branch=tartarus_pro")
```

closed 2022-08-02 by github-actions[bot]