openrazer/openrazer #1770: Support for Razer Blade 15 Keyboard

rayzorben, 2022-03-08:
### Device Info

* Name: Razer Blade 15
* Product Number: RZ09-0421NED3
* Model Number: RZ09-0421

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 005: ID 1532:028a Razer USA, Ltd Razer Blade
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x028a 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered) 

```

```
Please post the output of
    ls -lah /dev/input/by-id/

total 0
drwxr-xr-x 2 root root 180 Mar  8 10:38 .
drwxr-xr-x 4 root root 760 Mar  8 10:38 ..
lrwxrwxrwx 1 root root  10 Mar  8 10:38 usb-Azurewave_Integrated_RGB_Camera_01.00.00-event-if00 -> ../event20
lrwxrwxrwx 1 root root  10 Mar  8 10:38 usb-Azurewave_Integrated_RGB_Camera_01.00.00-event-if02 -> ../event21
lrwxrwxrwx 1 root root   9 Mar  8 10:38 usb-Razer_Razer_Blade-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 Mar  8 10:38 usb-Razer_Razer_Blade-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Mar  8 10:38 usb-Razer_Razer_Blade-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Mar  8 10:38 usb-Razer_Razer_Blade-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Mar  8 10:38 usb-Razer_Razer_Blade-if02-mouse -> ../mouse0
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>
Unfortunately I don't have Windows

[image]

brian-mcnamara, 2022-03-09:
Support added in #1766

brian-mcnamara, 2022-03-09:
@rayzorben maybe I can ping you elsewhere since completely different question, but are you having issues with audio channels being reversed? My left channel plays out of the right speaker and vice versa. Curious if its just my setup.

rayzorben, 2022-03-09:
> @rayzorben maybe I can ping you elsewhere since completely different question, but are you having issues with audio channels being reversed? My left channel plays out of the right speaker and vice versa. Curious if its just my setup.

Yep! I am just dealing with it for now, but they are definitely reversed. I thought maybe whoever put the laptop together swapped the cables.

stevebuck, 2022-03-20:
If you need more assurance that it's not the cables, I too am running into the same problem with the same laptop. It is otherwise glorious, just that the built in speakers are inverted.

PR #1766 (2022-04-07): Add 2022 blade 15 support
Getting my new 15" 2022 blade setup with Ubuntu and noticed the USB PID was not added.

This PR adds the PID to the driver and daemon to successfully incorporate open razer with this model.

Fixes #1770
closed 2022-04-07 by z3ntu