openrazer/openrazer #1774: Orochi V2 not recognised

ShabirK21, 2022-03-14:
### Distribution
EndeavourOS

### Support

- [ X] Have you tried rebooting?
- [ X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I installed openrazer-meta-git, added myself to plugdev, started the daemon and rebooted. But my mouse is still not recognized. On polychromatic troubleshoot, it says my PID is not supported, 1532:0094, but on the github page it shows as supported.

z3ntu, 2022-03-26:
Make sure these versions are all the same: `pacman -Q openrazer-meta-git openrazer-driver-dkms-git openrazer-daemon-git python-openrazer-git`

ShabirK21, 2022-03-26:
installed all of the mentioned packages, and rebooted. openrazer-daemon is also running but on razergenie, it says no devices are connected. My mouse is connected via bluetooth.

z3ntu, 2022-03-26:
What's the output of `sudo dkms install openrazer-driver/3.2.0` ?

ShabirK21, 2022-03-26:
Module openrazer-driver/3.2.0 already installed on kernel 5.16.16-zen1-1-zen (x86_64).

z3ntu, 2022-03-26:
`openrazer-daemon -Fv` ?

ShabirK21, 2022-03-26:
Unable to lock on the pidfile.

z3ntu, 2022-03-26:
Can you post the output of `pacman -Q openrazer-meta-git openrazer-driver-dkms-git openrazer-daemon-git python-openrazer-git` ?

ShabirK21, 2022-03-26:
openrazer-meta-git 3.2.0.r11.g43423b7b-1
openrazer-driver-dkms-git 3.2.0.r11.g43423b7b-1
openrazer-daemon-git 3.2.0.r11.g43423b7b-1
python-openrazer-git 3.2.0.r11.g43423b7b-1

ShabirK21, 2022-03-26:
zsh: no matches found: /sys/bus/hid/drivers/razermouse/0*/

for the modprobe, no output

z3ntu, 2022-03-26:
`sudo journalctl -b | grep -i razer_mount` ?

ShabirK21, 2022-03-27:
no output

closed 2022-03-30 by ShabirK21