openrazer/openrazer #1776: Support for Razer Pro Type Ultra -- disabling implicit fnlock for the function keys

albert-kam, 2022-03-15:
### Device Info

* Name: Razer Pro Type Ultra
* Product Number: RZ03-04110100
* Model Number: RZ03-0411

### Device Information
```

Bus 001 Device 007: ID 1532:0277 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0277 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
[... 196 lines cut ...]
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>
To be continued ...

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>
To be continued ...

Linux localdebian 4.19.0-18-amd64 #1 SMP Debian 4.19.208-1 (2021-09-29) x86_64 GNU/Linux
Debian 10 Buster

Problems:
- Pressing function keys such as F1 triggered sound toggling instead of plain F1
- Pressing fn+F1 triggers plain F1 

Expectations:
- Function keys should be the default, and the multimedia functions should be triggered with fn button

What I have tried:
- Getting help from razer customer support --> points out synapse3 --> changed the mapping --> it worked, but the mapping is not persisted to the keyboard itself, so still problematic when attached to a linux box.
- I havent tried xkb, but it seems that fn is at hardware level and cannot be mapped ?
- Bios, numlock keyboard combos, etc, but nothing could udo the default fn-lock
- https://unix.stackexchange.com/questions/693846/razer-pro-type-ultra-keyboard-how-to-use-function-keys-without-holding-down-fn

Please guide me !
Bert

gnos28, 2022-08-30:
Hello, anyone here have the skills to make this keyboard working properly ?
or maybe someone with the skills but without the hardware can guide me through the process of closing this issue ?

best regards

sonovice, 2022-08-30:
One of the problems is that F11 and F12 (light keys) are not even propagated as key events to the computer, so at least those two will always be unremappable until someone finds a way to patch the actual keyboard firmware.

sonovice, 2022-09-09:
On Windows, the Razer Drivers allow to switch the mode.

For Linux, I've come up with this solution: https://gist.github.com/sonovice/8d54608bcf8bc3a76213a21550de3dd8

David-Taub, 2022-09-11:
This solved it for me, on Ubuntu 20.04.
kudos!

pakar, 2023-01-06:
Just wanted to give notice that the latest firmware for the Razer Pro Type Ultra now supports to to toggle Fn-lock by "Fn + Esc".

This seems to be fully undocumented by Razer. Search for "Razer Pro Type Ultra firmware" and you should get a link to razer's support-pages with a windows-based firmware-updater.

It worked for me to update the fw from within a virtual Windows 10 machine (virt-manager/KVM) with USB forwarding of both the wireless-dongle + keyboard. Just remember you need to re-forward the device a few times during the update as the USB keyboard reboots, and DON'T disable the forward. Only the mouse is required during the update, but if something goes wrong a second keyboard is probably good to keep around.
For my keyboard it did two updates, ie after first update it reported that another firmware was available.

Got fed up with Razer after this debacle. Not even their support knew about this issue or could answer the Fn question.

Just wanted to drop this info as i know how frustrating the issue is, and undocumented functions are not fun.

drasill, 2023-01-09:
This is great. Maybe worth closing the issue?

pakar, 2023-01-09:
@drasill Even if the Fn key issue has been resolved it would be nice with support for LED's/Battery level etc.
There are a few other issues that also discuss support for this keyboard. Maybe some admin should review the issues and combine them into a single one and close the rest?

gnos28, 2023-01-10:
I confirm firmware update solve the issue

z3ntu, 2023-01-11:
Apart from fnlock functionality we can just keep this one open until full OpenRazer support is available for this device.

z3ntu, 2023-10-14:
Pcaps are posted in https://github.com/openrazer/openrazer/pull/2136#issuecomment-1734411238

descubriendoelcodigo, 2024-03-02:
Another confirm that firmware update makes fn+esc work under linux(ubuntu 22.04) However it would be cool to have openrazer support for other stuff aswell!