openrazer/openrazer #1780: Daemon won't start, UnicodeDecodeError: 'utf-8' codec can't decode byte 0x88 in position 5: invalid start byte

RP-11, 2022-03-26:
### Distribution

`openrazer-daemon --version
openrazer-daemon 3.2.0
`
### Support

- [ ] Have you tried rebooting? yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

 getting error when daemon is starting

`openrazer-daemon -Fv                                                                                                                                                                                             
Starting daemon.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/daemonize.py", line 245, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 140, in run_daemon
    daemon = RazerDaemon(verbose=args.verbose,
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 83, in __init__
    self.read_persistence(persistence_file)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 303, in read_persistence
    self._persistence.read(persistence_file)
  File "/usr/lib/python3.8/configparser.py", line 697, in read
    self._read(fp, filename)
  File "/usr/lib/python3.8/configparser.py", line 1017, in _read
    for lineno, line in enumerate(fp, start=1):
  File "/usr/lib/python3.8/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x88 in position 5: invalid start byte`

lah7, 2022-03-27:
This issue was very recently fixed: 74b5f642a5619fe7977adccb72c30847696c52f3

For now, please delete this file in your home directory: `~/.config/openrazer/persistence.conf`, then run the daemon again.

RP-11, 2022-03-27:
This worked. Thank you.

closed 2022-03-27 by RP-11