openrazer/openrazer #1782: Support for Razer Basilisk V3

futchas, 2022-03-29:
### Device Info

* Name: Razer Basilisk V3
* Product Number: RZ01-04000100
* Model Number: RZ01-0400

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 007: ID 1532:0099 Razer USA, Ltd Razer Basilisk V3
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0099 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      79
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

```
Please post the output of
    ls -lah /dev/input/by-id/

lrwxrwxrwx 1 root root 10 Mär 29 10:03 /dev/input/by-id/usb-Razer_Razer_Basilisk_V3-event-if01 -> ../event21
lrwxrwxrwx 1 root root  9 Mär 29 10:03 /dev/input/by-id/usb-Razer_Razer_Basilisk_V3-event-mouse -> ../event8
lrwxrwxrwx 1 root root  9 Mär 29 10:03 /dev/input/by-id/usb-Razer_Razer_Basilisk_V3-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root 10 Mär 29 10:03 /dev/input/by-id/usb-Razer_Razer_Basilisk_V3-if02-event-kbd -> ../event23
lrwxrwxrwx 1 root root  9 Mär 29 10:03 /dev/input/by-id/usb-Razer_Razer_Basilisk_V3-mouse -> ../mouse1

```

### Packet Captures

I don't have time for reverse engineering right now
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
No Windows Environment setup available 
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

lah7, 2022-03-29:
Razer Basilisk V3 (`1532:0099`) is already supported (#1703, #1610), ready for the next stable release.

In the meantime, you may install the development/daily/git packages for your distribution.

closed 2022-03-29 by lah7

futchas, 2022-03-29:
ok, when is the next stable release?

futchas, 2022-03-29:
ok I tried it out but it seems I can only control the light stuff through the GUIs but I cannot set/program the buttons. @lah7 did I miss something?

lah7, 2022-03-30:
There's no schedule for stable releases, so I don't really know.

OpenRazer's objective is to support the lighting and Razer-only functionality. For remapping, a lot of users had success with this tool: https://github.com/sezanzeb/input-remapper