openrazer/openrazer #1786: Razer Blade 14 (2021) Logo backlight toggle not working

solinium, 2022-04-05:
### Distribution

Razer Blade 14 (2021) AMD
Fedora Linux 36 (beta)
Package/Version: openrazer-meta-3.2.0-2.22.noarch.rpm

### Support

- [✔️] Have you tried rebooting?
- [✔️] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

The toggle to light up the laptop logo simply does not work. No matter what I do with OpenRazer it will not turn on in linux. The logo lights up in Windows/using Synapse. Checking all 3 logs in troubleshooting shows that there is no change in any logs when the option is toggled. It would be great if there was a fix for this. Additionally, even if there is no fix, if there is a way to toggle the backlight using echo I'd love to know as I'd use that until this was fixed. Thank you.

z3ntu, 2022-04-07:
Please share the output of `lsusb | grep 1532`

solinium, 2022-04-08:
@z3ntu 
catalyst :: ~ » lsusb | grep 1532
Bus 001 Device 003: ID 1532:0270 Razer USA, Ltd Razer Blade

z3ntu, 2022-04-08:
@BeanieBen9990 @johnhout Do you see this also on your devices?

BeanieBen9990, 2022-04-08:
@z3ntu Yeah I haven't been using it much at all, but I'm pretty sure I haven't been able to toggle that on or off, or change modes, IIRC

z3ntu, 2022-04-08:
Ok thanks for the clarification.

I looked at some packet captures posted in #1540 and have this patch for you to try:
```diff
diff --git a/driver/razerkbd_driver.c b/driver/razerkbd_driver.c
index d93d8234..060421cf 100644
--- a/driver/razerkbd_driver.c
+++ b/driver/razerkbd_driver.c
@@ -2056,6 +2056,12 @@ static ssize_t razer_attr_write_set_logo(struct device *dev, struct device_attri
         report = razer_chroma_standard_set_led_effect(VARSTORE, LOGO_LED, state);
     }
 
+    switch(usb_dev->descriptor.idProduct) {
+    case USB_DEVICE_ID_RAZER_BLADE_14_2021:
+        report.transaction_id.id = 0x1F;
+        break;
+    }
+
     razer_send_payload(usb_dev, &report);
 
     return count;

```

Unfortunately the pcaps don't contain an turning the logo off, so it's just a guess. If this doesn't work it would be great if someone could take a usb pcap of turning the logo off with Synapse.

solinium, 2022-04-09:
@z3ntu
Forked, applied the patch, compiled, no luck, still stays off regardless. I'm using the latest polychromatic to toggle the logo backlight, under where it says 'Laptop Lid', there's a 'Brightness' toggle checkbox. If you write a basic guide or link me a page with instructions to take the pcap or even just tell me what program I should use and I can probably figure it out, I'll boot into a windows partition and try to grab that for you.

z3ntu, 2022-04-09:
Would've been too simple if it worked :wink: 

Anyways, here's our wiki page for usb pcaps: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

Let me know if you have any questions.

closed 2022-09-07 by github-actions[bot]

SujayShinoda, 2023-06-19:
Same issue in 2023. Razer support is useless with their battery bloating issues and what not