openrazer/openrazer #1794: Wired Deathadder Essential is not being picked up, Fedora 35

FunkeMonke, 2022-04-09:
As of today I have updated the relavent files from openrazer and tried using RazerGenie but the mouse would not be detected. The app definitely worked because it showed me "No Device Was Connected"

Etaash-mathamsetty, 2022-04-09:
can you dump the output of lsusb?

FunkeMonke, 2022-04-10:
[lsusb_verbose_output.zip](https://github.com/openrazer/openrazer/files/8459053/lsusb_verbose_output.zip)

FunkeMonke, 2022-04-10:
Btw I was using polychromatic and just this is what outputs when I clicked the troubleshoot button. https://imgur.com/a/ecEJLnZ

FunkeMonke, 2022-04-10:
Sorry one last thing, I tried using this for both Razergenie and Polychromatic. Both applications were functioning, and both did not recognize my mouse.

Etaash-mathamsetty, 2022-04-10:
`Bus 001 Device 002: ID 1532:0098 Razer USA, Ltd Razer DeathAdder Essential`
so your mouse is being picked up by linux...

Etaash-mathamsetty, 2022-04-10:
oh I didn't look at that image, you have to modprobe the drivers, so open terminal, and run
`sudo modprobe razermouse`
and post your output (if any)

FunkeMonke, 2022-04-10:
I think I solved it. I ran the "sudo modprobe razermouse" and put the copy pasted the output in google and found a comment on support post  [#98](https://github.com/openrazer/openrazer/issues/98#issuecomment-271167846) 
Both RazerGenie and Polychromatic work now for my Fedora 35.

Just for any future people, what outputed before fixing it was this https://imgur.com/a/HlnYiGT
The solution was to disable secure boot.

Thank you for taking the time to help me @Etaash-mathamsetty

closed 2022-04-10 by lah7

Etaash-mathamsetty, 2022-04-11:
dumb question, but why are your using the breeze theme in gnome lol?