openrazer/openrazer #1796: Viper Ultimate Wireless not supported anymore with 3.3.0 version (PID 1532:007B)

HanM23, 2022-04-10:
### Distribution

Manjaro 21.2.5

### Support

- [ yes] Have you tried rebooting?
- [ yes] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Since the last update to version 3.3, my mouse is apparently not supported anymore, as you can see in the screenshot below.

[image]

Do i have to re-apply the 1710 patch ?

https://github.com/openrazer/openrazer/pull/1710
https://github.com/openrazer/openrazer/issues/1698#issuecomment-996850959

Thanks for your help

lah7, 2022-04-10:
The failed device compatibility check is on Polychromatic's part and [due to these changes](https://github.com/openrazer/openrazer.github.io/commit/8d18609e1eb9646f6bece73aaf2bd3d9d5acdf52), but nothing to be concerned about. It's because the mouse has multiple PIDs depending on mode, but the troubleshooter only reads one of those PIDs from [devices.json](https://openrazer.github.io/api/devices.json).  I'm a little behind on development, but it's on my to-do list,

If you are changing OpenRazer settings via Polychromatic, this might also be the app's fault for writing the old configuration keys for the battery notifier. Also on my to-do list to address.

Take a look at `~/.config/openrazer/razer.conf` in a text editor and check it does not have the old `mouse_battery_notifier` keys. If you see them, delete them, set the correct ones desired then reload the daemon:

```diff
-mouse_battery_notifier = False
-mouse_battery_notifier_freq = 0
battery_notifier = True
battery_notifier_freq = 300
```

HanM23, 2022-04-11:
Thanks, but i do not get it, really i do not know what to think, let me sum up my situation:

1) At first, with the previous version of openrazer, i had no notification at all.

2) Then, with the 1710 patch (and with openrazer 3.2), i get the notifications working, except the first notification at boot which always gives me 1% battery level (and the following notification are OK, they give me the right battery level, but the icon is always the one with the red bars).

3) Today, i updated openrazer to 3.3 version. I only had the notification at boot showing 1% battery level and that's it, then no more notification after that, again.

It's definitely not a `razer.conf` issue as, the keys are ok. I never had issue with those new keys since they exist.

I posted several issues here, mostly related to notifications issues. Checked everything, reinstalled several times and also reinstalled my distro from scratch to make sure my issue is not on my side.

I am pretty sure there are issues which has still not be identified on your side.

I already posted here some weird messages i have in journalctl at boot as :

`kernel: razermouse: Response doesn't match request. Start Marker: 02 id: ff Num Params: 02 Reserved: 00 Command: 81 Params: 01030000000000000000000000000000 .`

Also, i have this :

```
avril 11 01:56:39  kernel: razermouse 0003:1532:007B.0008: input,hidraw10: USB HID v1.11 Keyboard [Razer Razer Viper Ultimate Dongle] on usb-0000:00:14.0-9.1/input2
avril 11 01:56:39  kernel: razeraccessory 0003:1532:007E.000B: input,hidraw7: USB HID v1.11 Keyboard [RAZER Razer Mouse Dock] on usb-0000:00:14.0-9.2/input2
avril 11 01:56:39  kernel: razermouse 0003:1532:007B.0007: input,hidraw9: USB HID v1.11 Keyboard [Razer Razer Viper Ultimate Dongle] on usb-0000:00:14.0-9.1/input1
avril 11 01:56:39  kernel: razeraccessory 0003:1532:007E.000A: input,hidraw6: USB HID v1.11 Keyboard [RAZER Razer Mouse Dock] on usb-0000:00:14.0-9.2/input1
avril 11 01:56:38  kernel: razermouse 0003:1532:007B.0006: input,hidraw8: USB HID v1.11 Mouse [Razer Razer Viper Ultimate Dongle] on usb-0000:00:14.0-9.1/input0
avril 11 01:56:38  kernel: razeraccessory 0003:1532:007E.0009: input,hidraw5: USB HID v1.11 Mouse [RAZER Razer Mouse Dock] on usb-0000:00:14.0-9.2/input0
```
I do not undertstand why I have those 3 times, maybe it's normal i do not know.

Thanks for reading me.
Regards

lah7, 2022-04-12:
Let's keep the multiple issues separate. So:

`-` We know #1710 (#1698) fixed the missing notifications in 3.2.0. There were no other changes in the run up to 3.3.0 for that feature.
`-`  #1749 tracks the wrong percent being reported on every boot. (Weird mouse quirk)
`-` This issue to diagnose why notifications have disappeared with the 3.3.0 update.

Which desktop environment are you using? If it's a KDE desktop, I might have an idea of a possible bug (to do with notifications auto closing then never spawning a new one from that process again). 

The 3 lines from the kernel output looks normal. I don't think it's a Manjaro issue.

HanM23, 2022-04-13:
> Which desktop environment are you using? If it's a KDE desktop, I might have an idea of a possible bug (to do with notifications auto closing then never spawning a new one from that process again).

I use XFCE

alexonpeace, 2022-08-29:
I have the same issue

HanM23, 2022-11-24:
I still have this issue with 3.5.0, although it's different.

Now, i only have notifications when the mouse is in sleep mode. And in this case, all notifications received indicate 0%, which is normal.
I can see those notifications in XFCE -> Settings Manager -> Notifications.

Is there something i can do ? 
Thanks

HanM23, 2022-11-24:
I still have this issue although it's a bit different.
Now, with openrazer 3.5.0, i only get notifications when the mouse is in sleep mode. And of course, those notifications indicate 0% battery level, which is normal.

But, when the mouse is awake, when i am browsing or working, no notifications at all.

I can see those 0% battery level notifications in XFCE -> Settings Manager -> Notifications

lah7, 2022-11-24:
What are the settings for your `~/.config/openrazer/razer.conf`? From the sounds of notifications only showing up at 0%, `battery_notifier_percent` might be set to `0`?

```ini
battery_notifier = False
battery_notifier_freq = 300
battery_notifier_percent = 25
```

HanM23, 2022-11-24:
1. I do not think notifications should be showing something else than 0% when the mouse is in sleep mode, but maybe i am wrong.

2. `~/.config/openrazer/razer.conf`

> [General]
> verbose_logging = False
> 
> [Startup]
> sync_effects_enabled = True
> devices_off_on_screensaver = False
> battery_notifier = True
> battery_notifier_freq = 1800
> restore_persistence = True
> 

3. There is something weird in `~/.config/openrazer/persistence.conf`

> [PM2103H12300243]
> dpi_x = 1800
> dpi_y = 1800
> poll_rate = 1000
> logo_active = True
> logo_brightness = 15
> logo_effect = spectrum
> logo_colors = 0 255 0 0 255 255 0 0 255
> logo_speed = 2
> logo_wave_dir = 1
> 
> [PM2103U07700243]
> backlight_active = True
> backlight_brightness = 25
> backlight_effect = spectrum
> backlight_colors = 0 255 0 0 255 255 0 0 255
> backlight_speed = 1
> backlight_wave_dir = 1
> 
> [UNKWN000000003859]
> dpi_x = 1800
> dpi_y = 1800
> poll_rate = 1000
> logo_active = True
> logo_brightness = 15
> logo_effect = spectrum
> logo_colors = 0 255 0 0 255 255 0 0 255
> logo_speed = 1
> logo_wave_dir = 1

What's  [UNKWN000000003859] ?

When i delete this section, it always reappears

HanM23, 2022-11-24:
My godness, yes, that is better, but why i do not have this setting in my conf file ?
I always installed all updates from AUR.

Thanks.

Hey, could you please look at the first message a wrote in this thread, there is a screenshot where you can see a failed with the red exclamation mark, i still have it ?

Is it important, and how to solve it ?

Many thanks

lah7, 2022-11-24:
> why i do not have this setting in my conf file ?

I think, most likely, Polychromatic was used to edit this file (Tools → OpenRazer → Configure). The current release _(at this time of writing, v0.7.3)_ doesn't know about that new key. The latest code has updated that config editor.

Running the newer `openrazer-daemon` did do a migration from the old keys (`mouse_notifier_*`), but _maybe  didn't_ add the new key. New installs always get [the example razer.conf (comments included)](https://github.com/openrazer/openrazer/blob/master/daemon/resources/razer.conf) so you could move/delete the file if you ever wanted the latest config / defaults.

We'll have to see if adding this back did anything, since the default should've been `33` %:
https://github.com/openrazer/openrazer/blob/55792c2535826b7cae8547a5ba2049332ba415ad/daemon/openrazer_daemon/hardware/mouse.py#L1779

---

> there is a screenshot where you can see a failed with the red exclamation mark, i still have it currently.

That incorrect red exclamation mark has been fixed. Polychromatic hasn't had a new release in ages, but there should be one in the next week or two. If you'd like to test it sooner, feel free to install `polychromatic-git` from AUR.

HanM23, 2022-11-24:
Nice, thank you for your explanations and support !

HanM23, 2022-11-25:
I close this issue because it has been resolved.

closed 2022-11-25 by HanM23