openrazer/openrazer #1800: Audio Buttons - Huntsman V2 Not Working

westfarn, 2022-04-11:
### Device 
Razer Huntsman V2

### Operating System
Ubuntu 20.04 LTS

### Driver Version
openrazer-daemon 3.3.0

### Problem
The dedicated media keys (last, pause/play, skip) do not register with the Operating System anymore. When in gedit, all three buttons now type 'bcd' in the terminal. Looking at other issues, it looks like this could be an issue in the keyboard.py. I was unable to verify if that was a valid fix given that I don't have access to that file on my computer given my companies IT policies. Side note, the volume wheel still works.

Arsimael, 2022-04-11:
I don't want to reopen #1799, but after another reboot: I have the same issue again.

testcenter, 2022-04-21:
same for me.

OS: `pop!_OS 21.10` (Kernel: `5.16.15`)
Huntsman V2 (US Layout)
OpenRazer version: `3.3.0`

kierun, 2022-04-25:
Came here to say the same thing… Sadly, when the keyboard was unsupported, they worked fine for me! :raised_eyebrow: 

OS: Fedora 36 with kernel `5.16.20-200.fc35.x86_64`
Huntsmnan V2 (UK Layout)
OpenRazer version: 

```
openrazer-kernel-modules-dkms-3.3.0-1.1.noarch
openrazer-daemon-3.3.0-1.1.noarch
python3-openrazer-3.3.0-1.1.noarch
openrazer-meta-3.3.0-1.1.noarch
```

patrickscholz, 2022-06-01:
looks like one workaround can be found here https://github.com/openrazer/openrazer/issues/1398#issuecomment-961242268, at least that worked for me as well.

Mai-Lapyst, 2022-06-08:
Tinkered a bit around with the driver and found out that following changes seems to fix the issue:

```
diff --git a/driver/razerkbd_driver.c b/driver/razerkbd_driver.c
index 9bd4212..434e339 100644
--- a/driver/razerkbd_driver.c
+++ b/driver/razerkbd_driver.c
@@ -2682,6 +2682,7 @@ static int razer_event(struct hid_device *hdev, struct hid_field *field, struct
 
     case USB_DEVICE_ID_RAZER_BLACKWIDOW_V3:
     case USB_DEVICE_ID_RAZER_BLACKWIDOW_V3_PRO_WIRED:
+    case USB_DEVICE_ID_RAZER_HUNTSMAN_V2:
         translation = find_translation(chroma_keys_5, usage->code);
         break;
 
@@ -2974,6 +2975,7 @@ static int razer_raw_event(struct hid_device *hdev, struct hid_report *report, u
     case USB_DEVICE_ID_RAZER_BLACKWIDOW_V3:
     case USB_DEVICE_ID_RAZER_ORNATA_V2:
     case USB_DEVICE_ID_RAZER_BLACKWIDOW_V3_PRO_WIRED:
+    case USB_DEVICE_ID_RAZER_HUNTSMAN_V2:
         return razer_raw_event_bitfield(hdev, asc, intf, report, data, size);
     default:
         return razer_raw_event_standard(hdev, asc, intf, report, data, size);
```

kierun, 2022-06-08:
@Mai-Lapyst Could you please open a PR with the changes? Thank you.

kierun, 2022-06-08:
@Mai-Lapyst Does your fix only work for play/pause or for previous and next as well?

Mai-Lapyst, 2022-06-08:
@kierun It fixes all four buttons (play/pause, previous and next, mute)

kierun, 2022-06-09:
> @kierun It fixes all four buttons (play/pause, previous and next, mute)

@Mai-Lapyst Awesome! Thank you so much.

closed 2022-06-18 by z3ntu

z3ntu, 2025-10-21:
@EasySigh Please don't 'revive' years old issues that were about different devices. Check with the daily PPA, maybe #2503 has fixed it for you, not sure what exact device you own. Otherwise please open a new issue, feel free to reference this one so it's cross-linked.

PR #1834 (2022-06-18): Fixing mediabuttons for Huntsman V2
Fixes #1800 by
- using the correct key-mapping for the function keys
- using bitmap raw-events instead of the standard ones