openrazer/openrazer #1803: openrazer-daemon.service - Daemon to manage razer devices in userspace

maraingl-a217, 2022-04-27:
× openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2022-04-27 22:51:51 +04; 25s ago
       Docs: man:openrazer-daemon(8)
    Process: 10331 ExecStart=/usr/bin/openrazer-daemon -F (code=exited, status=1/FAILURE)
   Main PID: 10331 (code=exited, status=1/FAILURE)
        CPU: 86ms

z3ntu, 2022-04-27:
What happens when you run `openrazer-daemon -Fv` in your terminal?

maraingl-a217, 2022-04-27:
> Что происходит, когда вы запускаете `openrazer-daemon -Fv`в своем терминале?

Starting daemon.
2022-04-27 23:04:21 | razer                          | CRITICAL | User is not a member of the plugdev group
2022-04-27 23:04:21 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
Stopping daemon.
Exception ignored in atexit callback: <bound method Daemonize.exit of <daemonize.Daemonize object at 0x7f70c9086dd0>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/daemonize.py", line 72, in exit
    sys.exit(0)
SystemExit: 0

z3ntu, 2022-04-27:
Please do what the message tells you then.

maraingl-a217, 2022-04-27:
> Please do what the message tells you then.

thank you, problem solved

closed 2022-04-27 by maraingl-a217