openrazer/openrazer #1805: Support for [Key Light Chroma]

plusplusjeff, 2022-05-02:
### Device Info

* Name: <Key Light Chroma>
* Product Number: <RZ19-04120100>
* Model Number: <RZ19-0412>

### Device Information
``` 
Please post the output of
    lsusb -d 1532: -v
```

```
Please post the output of
    ls -lah /dev/input/by-id/
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>
[image]

plusplusjeff, 2022-05-02:
I'm very new to GitHub. Hopefully that is everything you need to add support for the Key Light. Would love to be able to control the colors / brightness. Please and thanks!

z3ntu, 2022-06-18:
Please post the output of `lsusb | grep 1532`, also the issue template is there for a reason, not to leave it empty ;)