openrazer/openrazer #1807: Basilisk v3 Settings Change On Dual Boot System

0pLuS0, 2022-05-05:
### Distribution
Slackware 15.0 x86_64

openrazer 3.3.0
polychromatic-0.7.3

### Support

- [ x] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

My system is Dual Boot, Windows 10 and Slackware 15.0.

I'm using polychromatic-0.7.3, I don't know if it plays a part in any of this, or these issues are strictly from Openrazer.

When I shut the computer down inside Windows, and of course I use Synaptics 3 in Windows, then when I start the computer back up and boot into Linux, I see this problem. The mouse keeps cycling through the colors, like it's on the default setting, and I placed polychromatic on Static, with the default Razer Green. It only goes to the Static setting when I log into X.

If I shut down the computer in Linux, and then start the computer, and boot into Linux, there isn't this issue. The mouse options stick as soon as I start up Slackware, half way through the boot up, when Openrazer loads, the mouse goes solid green.

I wish, when I shutdown the computer in Windows, and at some later point in time, when I start the computer, and then boot into Linux, that the mouse settings would stick right when Openrazer is loaded, because many times I just work in the console and I don't log into X, and then the mouse is sitting there cycling through the colors while I'm working in the console.

I personally don't like the colors cycling, it's distracting, and I would love to figure out if it's possible for me to correct, or this is just a limitation of the mouse and sofware, and running Dual Boot, with Synapse in Windows and Openrazer in Linux.

THANKS

lah7, 2022-05-06:
From https://github.com/polychromatic/polychromatic/issues/419:

> It's likely because Razer Synapse handles everything in software, so when that takes over, the hardware is back to a default spectrum upon next power cycle. The hardware effect becomes "software" rather then "static". Some firmware might be smart enough to use the last hardware effect.
>
> The trouble is, even if you set a static effect in Windows, it's actually sending a different command, like "draw a frame with all LEDs green" **(software)** rather then "activate the static effect hardware that's green" **(hardware).** If you're just wanting the same settings on both, the solution might be to uninstall Razer Synapse on Windows (and I mean, make sure it's **fully** removed), then use OpenRazer to set the hardware effect.
>
> Since OpenRazer daemon is a user process, it's only running as soon as you log on (on most distros). Unlike Razer Synapse which I think is a Windows  driver. If you do use Razer Synapse, you could [use the OpenRazer driver sysfs directly](https://github.com/openrazer/openrazer/wiki/Using-the-keyboard-driver) by echoing a file (as root or plugdev) during system boot up.
>
> Equally, OpenRazer doesn't technically know what the hardware is actually doing, which is where the persistence comes in. Until Razer Synapse is removed from the equation, that particular device will reset to spectrum on next power up.

0pLuS0, 2022-05-08:
I guess I'll have to live it then... 

THANKS

0pLuS0, 2023-01-16:
@lah7 Hi,

It would be nice to leave this open, because this has been an ongoing issue for me, with two versions of Openrazer, polychromatic and using Windows 10 and Windows 11 while dual booting with Slackware, and I still can't figure this out.

THANKS

lah7, 2023-01-17:
> I don't see what logging into X has to do with when the openrazer daemon is started, because with the rc.modules.local, the dameon is being loaded before I log into X.

Which user is it running as? I'm guessing `root`.  IIRC, there is weirdness running the daemon as root, because the daemon uses the **session bus** of the user, and root also uses a different config/persistence file (`/root/.config/...`)

It probably works when you log in to X because:

*  `openrazer-daemon` wasn't running as your user.
* Auto starting `polychromatic-tray-applet` is going to summon the daemon (via D-Bus).
* `openrazer-daemon` starts, reads your persistence file and restores effects, as expected.

**Possible solutions:**

* Make it so `rc.modules.local` starts `openrazer-daemon` as your user.
   * _Also double check devices do appear in the log, in case the daemon can't see them under non-X11 for some reason._
* Or use sysfs (as suggested in [#2002](https://github.com/openrazer/openrazer/issues/2002#issuecomment-1384429897) - see below). **No daemon required!**

## sysfs driver
It cuts out the middleman by echo'ing to the driver files directly. It might work better for your Slackware set up.

Put this in a bash script that runs at system boot (after drivers `razerkbd`, `razermouse` have loaded)

    echo -n -e "\xFF\x00\xFF" > /sys/bus/hid/drivers/razerkbd/*:1532:0214.*/matrix_effect_static
    echo -n -e "\xFF\x00\xFF" > /sys/bus/hid/drivers/razermouse/*:1532:0046.*/matrix_effect_static

Adjust as necessary to your hardware PID. This works as `root` or any user in the `plugdev` group (should already be the case)

lah7, 2023-01-17:
> For me, now the problem is, the Keyboard Effects don't work, only the Static using a Huntsman v2 Tenkeyless. 1532:026B

Has Razer Synapse recently (auto-)updated the firmware of the device?

If it just recently started happening, that's the only reason I can think of. Quite possible that what was implemented (#1754) for this device under OpenRazer is no longer compatible with your firmware version now?

0pLuS0, 2023-01-17:
No there have not been any recent updates for Synapse.

I have a Baslisk Essential I bought last month, so I am using that now, instead of the v3 which was defective.

I uninstalled openrazer-daemon and kernel and polychromatic and all the cache and config files too, then rebooted, then reinstalled everything.

Now, only the Ripple & Reactive Effects I can't get to to work when I first booted into Slackware.

But, sometimes when I play around clicking on different Effects, then I click on Static, it doesn't work either, so then I click on the other Effects that work, then go back to Static and it will work.

Seems to be a bug here, where I have to keep playing around with them, in order to get them working.

0pLuS0, 2023-01-19:
After playing with this for several hours, I can't get a lot of the Keyboard Effects to work. Even Static, which has always worked, when I start playing around clicking on some of the other effects, then go back to Static, the keyboard won't light back up.

I beleive I mentioned it before, I uninstalled openrazer and polychromatic, cleaned it all out, then compiled/reinstalled over and still no good.

The keyboard effects all work in Windows, for whatever reason, there's an issue with openrazer support and this keyboard.

```Huntsman v2 Tenkeyless 1532:026B```

THANKS

commit 7f830cb2 (2024-07-02): Add "persistence_dual_boot_quirk" config

It seems a handful of newer devices' have quirky firmware behaviour
after having an encounter with Razer Synapse (for users dual booting
with Windows) when restoring the persisted effect.

Even though we send the request and got no error, the device physically
didn't change its lighting, but it'll work by sending a second time.

Up to now, the user needs to go into a frontend to manually re-apply
their effect. Should the user run into this scenario, they can turn this
option on for a quality of life "now it just works".

Fixes #1807 (Basilisk V3)
Fixes [... cut]
PR #2262 (2024-07-02): Add `persistence_dual_boot_quirk` to razer.conf
It seems a handful of newer devices' have quirky firmware behaviour after having an encounter with Razer Synapse (for users dual booting with Windows) when restoring the persisted effect. Even though we send the request and got no error, the device physically didn't change its lighting, but it'll work by sending a second time.

Up to now, the user needs to go into a frontend to manually re-apply their effect. Should they run into this scenario, they can now turn this option on for a quality of life _"now it just works!"_.

Fixes #1807 (Basilisk V [... cut]
closed 2024-07-02 by z3ntu