openrazer/openrazer #1808: Suggestion: Secondary color choosing ability

Decapitater, 2022-05-07:
Secondary Color is referred to the color that shows up when either CAPS key or Fn Key (specially for mini keyboard ie Razer Huntsman mini) or any other mutimedia bindings ,

it would be cool to have option to change those lights and yes its possible , i have seen them in different color somehow . 

I was told that its something that maybe worth exploring for openrazer.

z3ntu, 2022-06-18:
@lah7 I guess this would be basically software effects?

lah7, 2022-06-18:
Software probably could handle it, but I think it's a firmware "feature". Whether it's configurable or not or simply undocumented, I don't know.

Some Razer Blade's _apparently_ will colour the top FN row green while FN is pressed, regardless of what the software is drawing.

z3ntu, 2022-06-18:
Then packet captures of this are needed :)

Decapitater, 2022-06-27:
@z3ntu i may get ya some will those help ?

z3ntu, 2022-07-23:
@Decapitater This is how you get pcaps: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

What exactly to toggle around in Synapse I don't know. But if it's a firmware feature and it's getting enabled somehow then this will show us.

closed 2022-12-22 by github-actions[bot]