openrazer/openrazer #1809: Support for [Razer PWM PC Fan Controllerdevice name here]

xorax, 2022-05-09:
### Device Info

* Name: Razer PWM PC Fan Controller
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
zoleath@sabnzb:~$ lsusb -d 1532: -v

Bus 002 Device 002: ID 1532:0f3c Razer USA, Ltd Razer PWM PC Fan Controller
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0f3c
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer PWM PC Fan Controller
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
```

```
Please post the output of
    ls -lah /dev/input/by-id/
lrwxrwxrwx 1 root root   9 May  9 08:18 usb-Razer_Razer_PWM_PC_Fan_Controller-event-if01 -> ../event5

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

z3ntu, 2022-06-18:
I assume this thing doesn't have RGB lights? Then packet captures of Razer Synapse changing settings is very much needed https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

Also then share what Razer Synapse on Windows can change, because I'm not familiar with this product at all.

AurianArchive, 2023-02-06:
Hey there. I already tracked down an email for z3 asking if he still needed a packet capture of this product. z3's response was yes, they still needed a packet capture. I don't have a rig set up right now that could do packet captures as required, but I could at least immediately make a video showing what Razer Synapse does with the device on Windows: https://rumble.com/v280u8q-razer-synapse-pwm-overview.html

There was a one liner in the video about maybe it'd just be quicker to buy another PWM and just donate it to OpenRazer... and after realizing I had no idea how Razer was determining temperature for the Fan Speed <-> Temperature ratio... and that replicating that function in Linux might not exactly be the easiest thing to accomplish... that's... what I've done.

[image]

So. I've got a Razer PWM controller for the OpenRazer devs. Just tell me who it can be shipped to.

Freestate2nd, 2025-06-01:
> ### Device Info
> 
>     * Name: Razer PWM PC Fan Controller
> 
>     * Product Number: RZ34-02140700-R3M1
> 
>     * Model Number: [RZ34-02140](https://mysupport.razer.com/app/answers/detail/a_id/5736)
> 
>     * VID:PID 1532:0f3c
> 
> 
> ### Device Information
> ## USB Details
> 
> ~$ lsusb -d 1532: -v
> 
> Bus 003 Device 006: ID 1532:0f3c Razer USA, Ltd Razer PWM PC Fan Controller
> Couldn't open device, some information will be missing
> Device Descriptor:
>   bLength                18
>   bDescriptorType         1
>   bcdUSB               2.00
>   bDeviceClass            0 [unknown]
>   bDeviceSubClass         0 [unknown]
>   bDeviceProtocol         0 
>   bMaxPacketSize0        64
>   idVendor           0x1532 Razer USA, Ltd
>   idProduct          0x0f3c Razer PWM PC Fan Controller
>   bcdDevice            2.00
>   iManufacturer           1 Razer
>   iProduct                2 Razer PWM PC Fan Controller
>   iSerial                 0 
>   bNumConfigurations      1
>   Configuration Descriptor:
>     bLength                 9
>     bDescriptorType         2
>     wTotalLength       0x003b
>     bNumInterfaces          2
>     bConfigurationValue     1
>     iConfiguration          0 
>     bmAttributes         0x80
>       (Bus Powered)
>     MaxPower              500mA
>     Interface Descriptor:
>       bLength                 9
>       bDescriptorType         4
>       bInterfaceNumber        0
>       bAlternateSetting       0
>       bNumEndpoints           1
>       bInterfaceClass         3 Human Interface Device
>       bInterfaceSubClass      0 [unknown]
>       bInterfaceProtocol      0 
>       iInterface              0 
>         HID Device Descriptor:
>           bLength                 9
>           bDescriptorType        33
>           bcdHID               1.11
>           bCountryCode            0 Not supported
>           bNumDescriptors         1
>           bDescriptorType        34 Report
>           wDescriptorLength      22
>           Report Descriptors: 
>             ** UNAVAILABLE **
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x81  EP 1 IN
>         bmAttributes            3
>           Transfer Type            Interrupt
>           Synch Type               None
[... 53 lines cut ...]
>      
> Starting dumping interrupt transfer stream
> with 1 minute timeout.
> 
> 003:006:001:STREAM             1748102979.731951
>  05 40 08 00 00 00 00 01 00 00 01 00 00 00 00 00
> 
> .003:006:000:Interrupt transfer timed out
> 003:006:001:Interrupt transfer timed out
> No more interfaces to dump
> 
> ### Packet captures
> 
> _none so far_
> ### Screenshots
> 
> [image]
> 
> I will try to do a packet capture but no promises, I never did that. This device is useless under linux without your support, there is no any other software/way to make it work please consider adding it.
> 
> Thank you!
> 
> Edit: Here are the control scripts for [LibreHardwareMonitor](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Controller/Razer/RazerFanController.cs) plugin source for [Fan Control](https://github.com/EvanMulawski/FanControl.Razer/blob/main/src/devices/pwm_fan_controller/PwmFanControllerDevice.cs) and plugin source for [SignalRGB](https://gitlab.com/signalrgb/signal-plugins/-/blob/Development/Plugins/Razer/Razer_PWM_Controller.js)

I'm dealing with this device too, capturing is a pain in the ass.

Jastreb, 2025-06-01:
> I'm dealing with this device too, capturing is a pain in the ass.

I know, I removed it from my system since its useless on linux, [asked liquidctl](https://github.com/liquidctl/liquidctl/issues/790) devs to add support for it, they might do it just by using existing scripts, but when idk. Since I use [CooleControl](https://gitlab.com/coolercontrol/coolercontrol/-/tree/main) which relies or works with liquidctl this might be better for me.

With Razer I would have the run the whole thing just to control case fans.

SlimRG, 2025-12-04:
> annel individually in Cooler Control asigning separate fan profiles/cur

Sorry, but that code was unstable

I give this code here for experiments.

```
// SPDX-License-Identifier: GPL-2.0
/*
 * Razer PWM PC Fan Controller (USB HID) -> Linux hwmon driver
 *
 * Goal
 * ----
 * Provide a standard hwmon interface so users can control/read fans via:
 *   - lm-sensors ("sensors")
 *   - fancontrol/pwmconfig
 *   - any tool reading /sys/class/hwmon/hwmon*/{pwm*,fan*_input,...}
 *
 * Device
 * ------
 * Razer PWM PC Fan Controller: VID 0x1532, PID 0x0f3c
 * 8 channels (PWM outputs + tach inputs).
 *
 * Transport
 * ---------
 * The device uses HID Feature reports. We send a 91-byte "packet" via
 * HID_REQ_SET_REPORT and read the response via HID_REQ_GET_REPORT.
 *
 * Concurrency / Ownership
 * -----------------------
 * HID devices can be accessed by multiple parties. If another driver/userspace
 * talks to the device concurrently, you can get flaky results. This driver
 * serializes its own transactions with a mutex, but cannot prevent other
 * processes from sending feature reports. In practice you want this driver to
 * be the only owner (use driver_override/udev rule if needed).
 *
 * Stability feature: periodic "force write"
 * -----------------------------------------
 * Some implementations periodically re-apply current PWM settings to protect
 * against device internal resets or missed writes. We do the same via a
 * delayed_work that runs every 2500 ms and re-sends cached PWM values to all
 * channels that are enabled.
 *
 * Protocol summary
 * ----------------
 * Packet length: 91 bytes.
 * Fields:
 *  [0]  ReportId
 *  [1]  Status
 *  [2]  SequenceNumber (increments by 0x08, never 0x00)
 *  [3]  RemainingCount hi
 *  [4]  RemainingCount lo
 *  [5]  ProtocolType
 *  [6]  DataLength
 *  [7]  CommandClass
 *  [8]  Command
 *  [9..88] Data[80]
 *  [89] CRC (XOR bytes 3..88)
 *  [90] Reserved
 *
 * PWM addressing:
 *  Data[1] = 0x05 + channel (channel 0..7 => 0x05..0x0C)
 *
 * Commands:
 *  Info class 0x00 cmd 0x87: fw version in Data[0..2]
 *  PWM  class 0x0D cmd 0x02: set channel mode, Data[2]=0x04 => manual
 *  PWM  class 0x0D cmd 0x0D: set channel percent, Data[2]=0..100
 *  PWM  class 0x0D cmd 0x81: get channel speed, RPM in Data[4..5] (big-endian)
 */

[... 646 lines cut ...]

	hid_hw_close(hdev);
	hid_hw_stop(hdev);
}

static const struct hid_device_id rz_hid_table[] = {
	{ HID_USB_DEVICE(RZ_VENDOR_ID, RZ_PRODUCT_ID) },
	{ }
};
MODULE_DEVICE_TABLE(hid, rz_hid_table);

static struct hid_driver rz_driver = {
	.name = "razer_pwm_pc_fan",
	.id_table = rz_hid_table,
	.probe = rz_probe,
	.remove = rz_remove,
};

module_hid_driver(rz_driver);

MODULE_AUTHOR("SlimRG-ready out-of-tree example");
MODULE_DESCRIPTION("Razer PWM PC Fan Controller (HID) -> hwmon fan/pwm driver with periodic force-write");
MODULE_LICENSE("GPL");

```

SlimRG, 2025-12-04:
Working scripts:
https://raw.githubusercontent.com/EvanMulawski/FanControl.Razer/refs/heads/main/src/EMRazer.Abstractions/Utils.cs
https://raw.githubusercontent.com/EvanMulawski/FanControl.Razer/refs/heads/main/src/devices/pwm_fan_controller/PwmFanControllerDevice.cs

And
https://raw.githubusercontent.com/LibreHardwareMonitor/LibreHardwareMonitor/5bb71eda8a0959c253599d493f476a76ca6d73a2/LibreHardwareMonitorLib/Hardware/Controller/Razer/RazerFanController.cs

SlimRG, 2025-12-04:
In future I want to rewrite code and try to make it stable
As in code cooler should be shown as standart PWM case cooler in linux, so any app can work with it

Jastreb, 2026-01-06:
> In future I want to rewrite code and try to make it stable As in code cooler should be shown as standart PWM case cooler in linux, so any app can work with it

I see you used chatgpt to write this and its fine as long as you validated it, but while the driver installs, beyond that I am unable to control the device with any software, even fan control, same error as cooler control, unable to set pwm_enable, which I submitted as issue on your project page https://github.com/SlimRG/razer-pwm-pc-fan-hwmon/issues/1

ANeuen, 2026-11-12:
I "vibe coded" a fork of SlimRG/razer-pwm-pc-fan-hwmon to use with CoolerControl to work with Bazzite (Atomic) and Fedora and hwmon. It has been working well and I made sure to test for error states such as unplugging and plugging back in. https://github.com/ANeuen/razer-pwm-pc-fan-hwmon.

Freestate2nd, 2026-11-13:
Hi @z3ntu,

I hope you don't mind me bringing this issue to your attention again.

There has been some encouraging progress recently. In particular, [[ANeuen's fork](https://github.com/ANeuen/razer-pwm-pc-fan-hwmon)](https://github.com/ANeuen/razer-pwm-pc-fan-hwmon) reportedly works with CoolerControl on Bazzite, including handling USB disconnections and reconnections.

Would you or another experienced OpenRazer developer be willing to take a look at the existing code and offer some technical feedback?

It would be wonderful to see these efforts consolidated into a stable, properly maintained Linux driver, potentially suitable for upstream inclusion.

Thank you very much for your time and for all your work on OpenRazer!

z3ntu, 2026-11-14:
Hi @Freestate2nd, unfortunately I really don't have time to look into something like that, there's been much activity in OpenRazer in recent months, right now there's still about 70 pull requests awaiting my attention.