openrazer/openrazer #1810: Support for Razer Huntsman V2 Tenkeyless

codifiedlikenessutility1970, 2022-05-10:
### Device Info

* Name: <Razer Huntsman V2 Tenkeyless>
* Product Number: <RZ03-03940400-R3U1>
* Model Number: <RZ03-0394>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```Bus 007 Device 003: ID 1532:006c Razer USA, Ltd Mamba Elite (Wired)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x006c Mamba Elite (Wired)
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Mamba Elite
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
[... 251 lines cut ...]

```
Please post the output of
    ls -lah /dev/input/by-id/
```total 0
drwxr-xr-x 2 root root 240 May  9 20:45 .
drwxr-xr-x 4 root root 600 May  9 20:45 ..
lrwxrwxrwx 1 root root   9 May  9 20:45 usb-Razer_Razer_Huntsman_V2_Tenkeyless-event-if01 -> ../event4
lrwxrwxrwx 1 root root   9 May  9 20:45 usb-Razer_Razer_Huntsman_V2_Tenkeyless-event-if04 -> ../event8
lrwxrwxrwx 1 root root   9 May  9 20:45 usb-Razer_Razer_Huntsman_V2_Tenkeyless-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 May  9 20:45 usb-Razer_Razer_Huntsman_V2_Tenkeyless-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 May  9 20:45 usb-Razer_Razer_Huntsman_V2_Tenkeyless-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 May  9 20:45 usb-Razer_Razer_Huntsman_V2_Tenkeyless-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 May  9 20:45 usb-Razer_Razer_Mamba_Elite-event-mouse -> ../event9
lrwxrwxrwx 1 root root  10 May  9 20:45 usb-Razer_Razer_Mamba_Elite-if01-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 May  9 20:45 usb-Razer_Razer_Mamba_Elite-if02-event-kbd -> ../event11
lrwxrwxrwx 1 root root   9 May  9 20:45 usb-Razer_Razer_Mamba_Elite-mouse -> ../mouse1

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

ThomChroma, 2022-05-24:
There's a support request already for this one #1681 :+1:

z3ntu, 2022-06-18:
Duplicate of #1681

closed 2022-06-18 by z3ntu