openrazer/openrazer #1812: Support for Razer Wolverine Ultimate

relma2, 2022-05-13:
### Device Info

* Name: Razer Wolverine Ultimate
* Product Number: RZ06-02250
* Model Number: RZ06-02250

### Device Information
```
$ lsusb -d 1532: -v
Bus 001 Device 011: ID 1532:007e Razer USA, Ltd RC30-030502 Mouse Dock
Bus 001 Device 007: ID 1532:0088 Razer USA, Ltd Gaming Mouse [Basilisk Ultimate, Wireless, Receiver]
Bus 001 Device 013: ID 1532:0a14 Razer USA, Ltd Razer Wolverine Ultimate
```

```
$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 520 May 13 14:10 .
drwxr-xr-x 4 root root 840 May 13 14:10 ..
lrwxrwxrwx 1 root root   9 May 13 14:10 usb-HANVON_UGEE_Artist_Pro_16_000000-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 May 13 14:10 usb-HANVON_UGEE_Artist_Pro_16_000000-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 May 13 14:10 usb-HANVON_UGEE_Artist_Pro_16_000000-if01-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 May 13 14:10 usb-HANVON_UGEE_Artist_Pro_16_000000-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 May 13 14:10 usb-HANVON_UGEE_Artist_Pro_16_000000-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 May 13 14:10 usb-Microsoft_Microsoft®_Nano_Transceiver_v2.1-event-if01 -> ../event11
lrwxrwxrwx 1 root root  10 May 13 14:10 usb-Microsoft_Microsoft®_Nano_Transceiver_v2.1-event-if02 -> ../event13
lrwxrwxrwx 1 root root   9 May 13 14:10 usb-Microsoft_Microsoft®_Nano_Transceiver_v2.1-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 May 13 14:10 usb-Microsoft_Microsoft®_Nano_Transceiver_v2.1-if01-event-mouse -> ../event10
lrwxrwxrwx 1 root root   9 May 13 14:10 usb-Microsoft_Microsoft®_Nano_Transceiver_v2.1-if01-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 May 13 14:10 usb-Microsoft_Microsoft®_Nano_Transceiver_v2.1-if02-event-kbd -> ../event12
lrwxrwxrwx 1 root root  10 May 13 14:10 usb-MOSART_Semi._2.4G_Wireless_Mouse-event-if00 -> ../event26
lrwxrwxrwx 1 root root  10 May 13 14:10 usb-MOSART_Semi._2.4G_Wireless_Mouse-event-mouse -> ../event25
lrwxrwxrwx 1 root root   9 May 13 14:10 usb-MOSART_Semi._2.4G_Wireless_Mouse-mouse -> ../mouse4
lrwxrwxrwx 1 root root  10 May 13 14:10 usb-Razer_Razer_Basilisk_Ultimate_Dongle-event-mouse -> ../event14
lrwxrwxrwx 1 root root  10 May 13 14:10 usb-Razer_Razer_Basilisk_Ultimate_Dongle-if01-event-kbd -> ../event15
lrwxrwxrwx 1 root root  10 May 13 14:10 usb-Razer_Razer_Basilisk_Ultimate_Dongle-if02-event-kbd -> ../event16
lrwxrwxrwx 1 root root   9 May 13 14:10 usb-Razer_Razer_Basilisk_Ultimate_Dongle-mouse -> ../mouse3
lrwxrwxrwx 1 root root  10 May 13 14:10 usb-RAZER_Razer_Mouse_Dock-event-mouse -> ../event27
lrwxrwxrwx 1 root root  10 May 13 14:10 usb-RAZER_Razer_Mouse_Dock-if01-event-kbd -> ../event28
lrwxrwxrwx 1 root root  10 May 13 14:10 usb-RAZER_Razer_Mouse_Dock-if02-event-kbd -> ../event29
lrwxrwxrwx 1 root root   9 May 13 14:10 usb-RAZER_Razer_Mouse_Dock-mouse -> ../mouse5
lrwxrwxrwx 1 root root   9 May 13 14:10 usb-Razer_Razer_Wolverine_Ultimate_000055FF40438D69-event-joystick -> ../event6
lrwxrwxrwx 1 root root   6 May 13 14:10 usb-Razer_Razer_Wolverine_Ultimate_000055FF40438D69-joystick -> ../js0
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>
TODO

### Synapse Screenshots
I dont have windows 😞

z3ntu, 2022-06-18:
You can install Windows in a VM btw and do packet captures that way, if you just do USB passthrough of this device to the VM.

Kusoneko, 2022-09-27:
I was gonna make a similar issue, but figured I'd look if it was already asked, as it is I'm gonna add the info I have about the device in here:

### Device Info

* Name: Razer Wolverine Ultimate
* Product Number: 0A14
* Model Number: ???

### Device Information
```
Bus 005 Device 003: ID 1532:0a14 Razer USA, Ltd Razer Wolverine Ultimate
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        71 
  bDeviceProtocol       208 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0a14 
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer Wolverine Ultimate
  iSerial                 3 00004E9C57E0FC22
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0060
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
[... 66 lines cut ...]
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

```
total 0
lrwxrwxrwx 1 root root   9 Sep 25 16:27 usb-Razer_Razer_Wolverine_Ultimate_00004E9C57E0FC22-event-joystick -> ../event6
lrwxrwxrwx 1 root root   6 Sep 25 16:27 usb-Razer_Razer_Wolverine_Ultimate_00004E9C57E0FC22-joystick -> ../js0
```

### Packet Captures && Synapse Screenshots

As much as I wish I could provide either, I actually can't as for some unknown and likely very stupid reason, the Windows Razer Synapse doesn't actually recognize the device, in fact it's supposed to use the Xbox One's Razer Synapse. Windows doesn't recognize any of the macro keys at the back, recognizes it only as a regular xbox controller, and same for Steam atm. I'm essentially asking if there's a way to add support for it at all (not as in the whole controller, the regular xbox controller driver for linux does most of the job, but like windows, doesn't recognize the macro keys, so essentially the question is can openrazer recognize the macro keys and handle them? Can openrazer handle the chroma stuff that you can only manage on Xbox Razer Synapse atm? If not at least have it be recognized for what it is so that polychromatic (or other openrazer front-ends) doesn't show it as an unknown device? Note, I'd like to provide Xbox Synapse screenshots but I actually don't currently have an Xbox and bought the controller specifically for PC gaming stuff. Maybe someday I'll have an actual Xbox on which to get screenshots, but that still doesn't solve the pcap issue however, as I highly doubt on an Xbox you get enough access to everything in general to gain the ability to get some pcap.

z3ntu, 2022-10-15:
Not sure what's then to support from openrazer side? The basic gamepad functionality is/should be provided by other drivers.

Kusoneko, 2022-10-15:
Like I said, at least properly detect it and identify it so polychromatic can tell what it is, and perhaps when I get an xbox I can figure out a way to capture packets (not quite sure how to do that though, would require some sort of device with like 2 usb ports that just captures packets and otherwise does passthrough between the 2 ports, and some way to get the captured packets afterwards) between them to help figure out how to implement chroma functionality instead of being stuck with the default chroma rgb scheme.

z3ntu, 2022-10-15:
So there's a way on xbox to change the rgb lights on the controller? You need to tell me all these details, nothing here is obvious :wink:

Kusoneko, 2022-10-15:
There is also a few extra buttons, which I've tested everything except the quick control panel and they don't work on PC: `2 remappable Multi-Function bumpers, 4 Multi-Function triggers and a Quick Control Panel.` from https://www.razer.com/console-controllers/razer-wolverine-ultimate though we're still lacking a good solution on how to get pcaps from an xbox, I've never done that before and don't currently have one, although I'm planning on it right now for this and the few games that won't work on proton. Would have normally gone with a PS5 to get the exclusives but getting one rn at MSRP is still impossible where I live, as opposed to an Xbox Series X.

relma2, 2023-05-01:
Just wanted to say There is an official Microsoft Store app called [Razer Controller Setup for Xbox](https://apps.microsoft.com/store/detail/razer-controller-setup-for-xbox/9NSXXH05PFRQ?hl=en-us&gl=us) for Windows so hopefully you can get packet captures from that.

relma2, 2023-05-06:
[Wolverine_Ultimate.zip](https://github.com/openrazer/openrazer/files/11412680/Wolverine_Ultimate.zip)

z3ntu, 2023-07-16:
@relma2 My script finds a bunch of packets that would normally match Razer packets but nothing looks familiar, so I'm going to assume it's a completely different protocol. Sorry ^^

bljordan, 2025-04-14:
Is a capture still needed or do we need to figure out reversing the above captures?

z3ntu, 2025-04-19:
@bljordan The latter probably.