openrazer/openrazer #1815: Support for Steam Deck?

Jay2645, 2022-05-17:
### Distribution

SteamOS release rolling (Holo)
VARIANT_ID=steamdeck
BUILD_ID=20220425.2
VERSION_ID=3.1

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

So Steam Deck is built on Arch Linux, but most users are going to be able to see the GUI-based Discover package manager by default. The way the OS is set up, it's somewhat difficult for the vanilla OS to make modifications to the system. [From what I've read online,](https://www.reddit.com/r/SteamDeck/comments/tfb5eb/aur_on_deck/) any system updates will wipe everything outside of the /home folder. [Steam encourages the use of Flatpaks instead](https://twitter.com/thenaughtysquid/status/1500146812921032717) for that reason.

Perhaps I missed something, but with Steam Deck orders starting to go out _en masse_ there will be a sizable contingent of new users trying to get their Razer hardware to work (myself among them, wiiithout Synapse runiiing my "iii" key liiikes to stiicik on my Huntsman V2, III have been typiiing thiiis entiirie message whiiile dealiiing wiiith thiisi). Is it possible to get an official or community-supported version for the Steam Deck? Failing that, at least mentioning [on the download page](https://openrazer.github.io/#download) that the Steam Deck is Arch-based will help new users who aren't necessarily as familiar with Linux as I am.

z3ntu, 2022-05-17:
While I don't know how the OS on the Steam Deck works in detail, if you can't use dkms or at least build out-of-tree kernel modules for the kernel and load them, you won't have any luck with OpenRazer.

Jay2645, 2022-05-17:
I figured as much. I'm also not completely familiar with how the system works yet, I'm going off of what I've read online. It should _work_, it'll just get wiped with every update.

I'm mostly coming from the context of "hey, I want this to work, how do I do that?" and generally wound up being pointed here. From what I can tell, it _should_ work (I'm still in the process of installing; there's just a number of headaches since Valve _really_ doesn't want you in here). And maybe this should be something that the GUI guys wind up handling at the end of the day.

I just thought I'd mention it since I get the feeling I'll be the first of many who wind up in this situation. Feel free to close if you'd like. :)

Jay2645, 2022-05-17:
So, for anyone coming from the future: I have good news and bad news.

Good news is that I got the package installed, and it now works on the Deck (until the next update). I was able to get it working through [this Reddit comment](https://www.reddit.com/r/SteamDeck/comments/t92ozw/for_compiling_c_code/i8qxemd/), which is _probably_ complete overkill but it got everything installed and I stopped getting random errors when running `yay -S openrazer-meta`:

```
sudo steamos-readonly disable && sudo pacman-key --init && sudo pacman-key --populate archlinux && sudo pacman -Syu tmux htop wget git ncdu podman fakeroot yay-git base-devel mlocate cmake gcc holo/linux-headers jupiter/linux-neptune-headers holo/linux-lts-headers $(pacman -Qk 2>/dev/null | grep -ve ' 0 missing' | grep -ie ^libc -e glibc -e gcc -e clang -e headers -e udev -e systemd | awk -F ':' '{print $1}') --overwrite '*'
```

Bad news iisi that even wiiith Polychromatiiic runniinig and the driiivers workiinig through OpenRazer, my Huntsman V2 Analog stiiil has a broken "iii" key. IIIt doesn't happen on Wiiindows (well, iiti does, but only iiif Synapse iisin't runniinig), but that's a separate iiissue from what II wanted to solve here. :)

z3ntu, 2022-05-20:
The i key issue sounds like a case for warranty... But nice that you got it working :) Hopefully this is useful for others that want to do the same!

closed 2022-05-20 by z3ntu