openrazer/openrazer #1816: modprobe: ERROR: could not insert 'razermouse': Key was rejected by service

ourichermath, 2022-05-20:
### Distribution

Operating System: Fedora Linux 35
KDE Plasma Version: 5.24.4
KDE Frameworks Version: 5.91.0
Qt Version: 5.15.2
Kernel Version: 5.17.8-200.fc35.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 12 × AMD Ryzen 5 5600X 6-Core Processor
Memory: 15.6 GiB of RAM
Graphics Processor: AMD Radeon RX 6800 XT

### Support

- [ ] Have you tried rebooting?  Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?  Yes

I'm just installing my new Razer Viper Ultimate (wireless) on Fedora 35 with Secure Boot enabled and when I try various modprobes like:

`sudo modprobe razermouse`

I get the ERROR:  could not insert ...: Key was rejected by service message.  Digging around here, I'd say it's because of a signing issue and my secure boot being on.  I'm not going to disable secure boot and your wiki points me to:

https://github.com/openrazer/openrazer/issues/98

for instructions on signing something to get things working.  But, for the life of me, I can find no actual instructions.  Could you include actual signing instructions in your wiki?  Even a pointer to something current and specific would be good.  I'm just a normal person, so I need something detailed.

On the off-chance this is NOT secure-boot-related, here's some more information:

I installed OpenRazer with:

```
sudo dnf install kernel-devel
sudo dnf config-manager --add-repo https://download.opensuse.org/repositories/hardware:razer/Fedora_35/hardware:razer.repo
sudo dnf install openrazer-meta polychromatic
sudo gpasswd -a $USER plugdev
```
and rebooted.

The mouse is actually recognized by Fedora and works.  But, when I run Polychromatic it shows the mouse (1532:007B) and receiver (1532:007E) with question marks next to them in the sidebar and the message "Unrecognized Device" in the center.  When I run the Troubleshooter, I get plenty of green checks and then red crosses for Kernel Module is Loaded, Check Secure Boot (EFI) Status, and Check Device Compatibility (sorry, the Copy to Clipboard function on that page isn't working).  I saw a reference here that the last one (compatibility) is just a false positive from Polychromatic.  But, the other two and the modprobe error point to Secure Boot/Signing.

`lsusb | grep 1532`

gives me:
```
Bus 001 Device 011: ID 1532:007e Razer USA, Ltd RC30-030502 Mouse Dock
Bus 001 Device 010: ID 1532:007b Razer USA, Ltd RC30-0305 Gaming Mouse Dongle [Viper Ultimate (Wireless)]
```

`sudo dkms install openrazer-driver/3.3.0`

gives me:

`Module openrazer-driver/3.3.0 already installed on kernel 5.17.8-200.fc35.x86_64 (x86_64).`

`uname -r`

gives me:

`5.17.8-200.fc35.x86_64`

`sudo dkms status`

gives me:

`openrazer-driver/3.3.0, 5.17.8-200.fc35.x86_64, x86_64: installed`

`lsusb`

gives me:

```
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 08bb:2704 Texas Instruments PCM2704 16-bit stereo audio DAC
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 2516:014d Cooler Master Co., Ltd. ARES
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 011: ID 1532:007e Razer USA, Ltd RC30-030502 Mouse Dock
Bus 001 Device 010: ID 1532:007b Razer USA, Ltd RC30-0305 Gaming Mouse Dongle [Viper Ultimate (Wireless)]
Bus 001 Device 009: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 004: ID 1b1c:1b39 Corsair Corsair Gaming K65 RGB RAPIDFIRE Keyboard
Bus 001 Device 003: ID 0b05:1939 ASUSTek Computer, Inc. AURA LED Controller
Bus 001 Device 002: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

z3ntu, 2022-05-20:
Sorry, I've never used secure boot so I'm not helpful here. But openrazer kernel modules aren't any different to any other dkms module so you should be able to find something that helps you online. If you find something useful please link it so we can put it on the wiki page though.

ourichermath, 2022-05-20:
Well, the problem is that I'm fairly new to Linux and signing dkms or kernel modules is all greek to me.  So, I have no idea if the things I find are applicable or even currently correct.  I'll look/ask around and see what comes up.  If I find anything, I'll report back.

ourichermath, 2022-05-20:
I've found this:

https://gist.github.com/dasunsrule32/90a5d0f806b5e54cf4d1bceb97835957

which looks promising.  His comment at the end seems to be specific to (a previous version of) OpenRazer.  I'm trying to make some sense of it, but while I'm trying to do that, I thought you might like the link.

wgetgoose, 2022-06-11:
I successfully signed kernel modules using [this project](https://github.com/hf/openrazer-sign) by [@hf](https://github.com/hf). Fedora 36 with a Razer Viper mini.

closed 2022-11-10 by github-actions[bot]